freebsd/drm-kmod #1: [5.2][i915] VAAPI encoding hangs

jbeich, 2020-05-11:
Encoding works fine on e67dcab235c5 (drm_v5.1_4) but not 15e197fa20 (5.2-wip). FFmpeg starts but doesn't progress.

```
$ pciconf -l | grep vga.\*8086
vgapci1@pci0:0:2:0:     class=0x030000 rev=0x06 hdr=0x00 vendor=0x8086 device=0x1912 subvendor=0x1462 subdevice=0x7968

$ pkg install ffmpeg libva-intel-media-driver
$ fetch https://media.xiph.org/video/derf/y4m/pedestrian_area_1080p25.y4m
$ ffmpeg -hide_banner -vaapi_device /dev/dri/renderD128 \
  -i pedestrian_area_1080p25.y4m \
  -vf format=nv12,hwupload -c:v h264_vaapi \
  -global_quality 18 \
  -y pedestrian_area_1080p25.h264.mp4
Input #0, yuv4mpegpipe, from 'pedestrian_area_1080p25.y4m':
  Duration: 00:00:15.00, start: 0.000000, bitrate: 622081 kb/s
    Stream #0:0: Video: rawvideo (I420 / 0x30323449), yuv420p(progressive), 1920x1080, SAR 1:1 DAR 16:9, 25 fps, 25 tbr, 25 tbn, 25 tbc
Stream mapping:
  Stream #0:0 -> #0:0 (rawvideo (native) -> h264 (h264_vaapi))
Press [q] to stop, [?] for help
Output #0, mp4, to 'pedestrian_area_1080p25.h264.mp4':
  Metadata:
    encoder         : Lavf58.29.100
    Stream #0:0: Video: h264 (h264_vaapi) (High) (avc1 / 0x31637661), vaapi_vld, 1920x1080 [SAR 1:1 DAR 16:9], q=-1--1, 25 fps, 12800 tbn, 25 tbc
    Metadata:
      encoder         : Lavc58.54.100 h264_vaapi
[Ctrl+T doesn't work]

$ truss -p $(pgrep ffmpeg)
ioctl(3,0xc010646c { IORW 0x64('d'), 108, 16 },0x7fffffffbf10) ERR#60 'Operation timed out'
ioctl(3,0xc010646c { IORW 0x64('d'), 108, 16 },0x7fffffffbf10) ERR#60 'Operation timed out'
ioctl(3,0xc010646c { IORW 0x64('d'), 108, 16 },0x7fffffffbf10) ERR#60 'Operation timed out'
ioctl(3,0xc010646c { IORW 0x64('d'), 108, 16 },0x7fffffffbf10) ERR#60 'Operation timed out'
ioctl(3,0xc010646c { IORW 0x64('d'), 108, 16 },0x7fffffffbf10) ERR#60 'Operation timed out'
^C

$ procstat -kk $(pgrep ffmpeg)
  PID    TID COMM                TDNAME              KSTACK
 6337 101818 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_timedwait_sig+0x12 linux_add_to_sleepqueue+0x96 linux_schedule_timeout+0x86 i915_request_wait+0x2d0 i915_gem_object_wait_fence+0x93 i915_gem_object_wait+0xae i915_gem_wait_ioctl+0x164 drm_ioctl_kernel+0xfa drm_ioctl+0x2c9 linux_file_ioctl+0x37d kern_ioctl+0x23a sys_ioctl+0x101 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101868 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101869 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101870 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101871 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101872 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101873 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101874 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
 6337 101875 ffmpeg              -                   mi_switch+0xbe sleepq_catch_signals+0x4a4 sleepq_wait_sig+0x9 _sleep+0x1ad umtxq_sleep+0x230 do_wait+0x40a __umtx_op_wait_uint_private+0x82 amd64_syscall+0x10c fast_syscall_common+0x101
```

jbeich, 2020-05-11:
See also [drm_debug=0xFFF output](https://github.com/freebsd/drm-kmod/files/4612301/drm_debug.log).

(This message will be merged into original post.)

evadot, 2020-05-13:
I've started to debug this but the branch isn't clean enough to gisect properly, so I'm regenerating a clean branch that test every commit and I fix them at the same time.
Note that it's working if you don't have any GL app (from the console or without any GL app, I'm using awesome as a window manager and I have the problem when only when I start a GL app before, glxgears is enough).

evadot, 2020-05-13:
Could you test the 5.2-i915-clean branch ?
https://github.com/freebsd/drm-kmod/commit/bae9df9e602230e09d2e000c6ab0723abb85b556#diff-0e4e1619072065a31f0c20475cce94ef
This commit wasn't done correctly in the 5.2-wip branch.
We don't have irq_work_queue and I forgot to patch the code that is needed on FreeBSD.

jbeich, 2020-05-13:
I confirm, VAAPI encoding works fine on `5.2-clean-i915`.

evadot, 2020-05-13:
Thanks for confirming, I have one last bug to fix and then I'll merge this branch into master.

closed 2020-05-13 by evadot