freebsd/drm-kmod #6: I can't load radeonkms (drm-devel-kmod)

pkubaj, 2020-06-04:
I'm on head on powerpc64, r361749. Using 5.2.g20200601, I could successfully load radeonkms, now I'm on 5.2.g20200602.
When executing `kldload radeonkms`:
link_elf: symbol drm_timestamp_precision undefined
KLD radeonkms.ko: depends on drmn - not available or version mismatch
kldload: can't load radeonkms: No such file or directory

evadot, 2020-06-04:
You likely have an old drm.ko in /boot/kernel or /boot/module, nothing changed between those two tags that could cause this.

pkubaj, 2020-06-04:
I rebuilt kernel and drm-devel-kmod. Now I'm getting:
```
link_elf: symbol drm_vblank_offdelay undefined
``` 
I tried to build the previous snapshot (by modifying GH_TAGNAME) but it looks like due to some changes in head, it doesn't build. So it could be that this issue is actually in linuxkpi, not in drm-kmod.

evadot, 2020-06-04:
Previous tagged release won't build due to changes in HEAD in linuxkpi yes.

evadot, 2020-06-09:
Could you try with https://github.com/freebsd/drm-kmod/pull/7 ?
I think that there is something with compiler or arch because I don't need it.

pkubaj, 2020-06-09:
That fixed it, however I'm now hitting another error:
```link_elf: symbol drm_compat_ioctl undefined```

jorgem-seq, 2020-06-10:
In drm_v5.2_3 `drivers/gpu/drm/drm_os_config.h` was removed, it had a lot of macros, for example `#define CONFIG_COMPAT 1`, without it `drm_compat_ioctl` is `NULL`.
Thanks for testing.

evadot, 2020-06-10:
COMPAT is defined here https://github.com/freebsd/drm-kmod/blob/master/kconfig.mk#L30
drm_compat_ioctl is defined here : https://github.com/freebsd/drm-kmod/blob/master/drivers/gpu/drm/drm_ioc32.c#L974
And this file is included is COMPAT is defined as shown here : https://github.com/freebsd/drm-kmod/blob/master/drm/Makefile#L94

pkubaj, 2020-06-11:
Yes, that fix works. Thanks!

evadot, 2020-06-11:
Should be good on the master branch now.
I'll update the ports to 5.3 with those last changes this weekend.

closed 2020-07-08 by evadot