freebsd/drm-kmod #9: "panic: <something>: called with spinlock or critical section held" on KabyLake

wulf7, 2020-06-11:
5.0+ drm-kmods including 5,2_10 often panics on KabyLake GPUs with following backtrace:
```
panic: free: called with spinlock or critical section held
cpuid = 5
time = 1591907458
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe006dd1c2f0
vpanic() at vpanic+0x182/frame 0xfffffe006dd1c340
panic() at panic+0x43/frame 0xfffffe006dd1c3a0
free() at free+0xe4/frame 0xfffffe006dd1c3d0
dma_fence_signal() at dma_fence_signal+0xdb/frame 0xfffffe006dd1c410
reservation_object_add_shared_fence() at reservation_object_add_shared_fence+0x189/frame 0xfffffe006dd1c470
i915_vma_move_to_active() at i915_vma_move_to_active+0x1a0/frame 0xfffffe006dd1c4b0
i915_gem_do_execbuffer() at i915_gem_do_execbuffer+0xdbb/frame 0xfffffe006dd1c6b0
i915_gem_execbuffer2_ioctl() at i915_gem_execbuffer2_ioctl+0x1ad/frame 0xfffffe006dd1c710
drm_ioctl_kernel() at drm_ioctl_kernel+0xf9/frame 0xfffffe006dd1c750
drm_ioctl() at drm_ioctl+0x2c8/frame 0xfffffe006dd1c840
linux_file_ioctl() at linux_file_ioctl+0x326/frame 0xfffffe006dd1c8a0
kern_ioctl() at kern_ioctl+0x1e5/frame 0xfffffe006dd1c900
sys_ioctl() at sys_ioctl+0x127/frame 0xfffffe006dd1c9d0
amd64_syscall() at amd64_syscall+0x133/frame 0xfffffe006dd1caf0
fast_syscall_common() at fast_syscall_common+0x101/frame 0xfffffe006dd1caf0
--- syscall (54, FreeBSD ELF64, sys_ioctl), rip = 0x8005c669a, rsp = 0x7fffffffcbc8, rbp = 0x7fffffffcbf0 ---
KDB: enter: panic
```
Easy way to trigger the panic is to run Xorg with GUC firmware or run vp8enc from multimedia/libva-utils port. Sometimes it happens while running Xorg with standard firmware. See https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=247166 for example.

Exact command asserting the panics changes from time to time. Most often it is locking of sleep-able or nonsleep-able mutex or free(). That command is called through use of dma_fence_is_signaled() routine surrounded by critical section in reservation_object_add_shared_fence() in linuxkpi/gplv2/src/linux_reservation.c

wulf7, 2020-06-11:
To workaround the panic I updated inuxkpi/gplv2/src/linux_reservation.c to 5.3 state and moved dma_fence_is_signaled() out of critical section so reservation_object_add_shared_fence() looks as:

```
void reservation_object_add_shared_fence(struct reservation_object *obj,
                                         struct dma_fence *fence)
{
        struct reservation_object_list *fobj;
        unsigned int i, count;

        dma_fence_get(fence);

        reservation_object_assert_held(obj);

        fobj = reservation_object_get_list(obj);
        count = fobj->shared_count;
#ifdef _linux__
        preempt_disable();
        write_seqcount_begin(&obj->seq);
#endif
        for (i = 0; i < count; ++i) {
                struct dma_fence *old_fence;

                old_fence = rcu_dereference_protected(fobj->shared[i],
                                                      reservation_object_held(obj));
                if (old_fence->context == fence->context ||
                    dma_fence_is_signaled(old_fence)) {
                        dma_fence_put(old_fence);
                        goto replace;
                }
        }

        BUG_ON(fobj->shared_count >= fobj->shared_max);
        count++;

replace:
#ifndef _linux__
        preempt_disable();
        write_seqcount_begin(&obj->seq);
#endif
        RCU_INIT_POINTER(fobj->shared[i], fence);
        /* pointer update must be visible before we extend the shared_count */
        smp_store_mb(fobj->shared_count, count);

        write_seqcount_end(&obj->seq);
        preempt_enable();
}
```
Unfortunately, this file is not under control of VCS, so I provide its content rather than diff.
I do not think that this change is correct but it at least allows system to run without drm-kmod panics

wulf7, 2020-06-11:
Can't reproduce GUC firmware induced panic with drm-devel-kmod port now. Something changed within last month? vp8enc is ok.

evadot, 2020-06-11:
A lot changed within the last month :)
When you say vp8enc is ok it means that it doesn't panic anymore ?

wulf7, 2020-06-12:
vp8enc is ok means it still panics

wulf7, 2020-06-12:
description updated with vp8enc panic backtrace done with kmod compiled with -O0 CFLAG to remove inlining

wulf7, 2020-06-12:
I have got an idea how to fix vp8enc case with introducing of kfree_async() on lkpi layer.
I'll try to implement it.

PR case is different. I was not able reproduce it yet.

commit ef322347 (2021-01-17): Use asynchronous free() call in i915_sw_fence code to fix #9.
commit a83471b4 (2021-01-17): Use asynchronous free() call in i915_sw_fence code to fix #9.
closed 2021-01-17 by evadot