freebsd/drm-kmod #10: panic on attach a NAVI10 card: Unregistered use of FPU in kernel

NorwegianRockCat, 2020-06-21:
I spent some time trying to on the current master branch at commit efc692b with a Radeon 5700XT card (NAVI 10). The code needs an adjustment for the firmware to finish loading. That is, the PSP_SOS has mdelay of 20 in psp_v11_0.c:302 as it waits for something to "finish", but fails in FreeBSD (too quick?) I increased it to 200 and it worked.

Once that finishes, the kernel panics with an unregistered use of the FPU in the kernel.

The backtrace is:

```
Unread portion of the kernel message buffer:
<6>[drm] reserve 0x900000 from 0x8002400000 for PSP TMR
<6>amdgpu: [powerplay] smu driver if version = 0x00000033, smu fw if version = 0x00000035, smu fw version = 0x002a3200 (42.50.0)
<4>amdgpu: [powerplay] SMU driver if version not matched
<6>amdgpu: [powerplay] SMU is initialized successfully!
panic: Unregistered use of FPU in kernel
cpuid = 9
time = 1592673856
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00ea3b8bd0
vpanic() at vpanic+0x182/frame 0xfffffe00ea3b8c20
panic() at panic+0x43/frame 0xfffffe00ea3b8c80
trap() at trap+0x828/frame 0xfffffe00ea3b8d90
calltrap() at calltrap+0x8/frame 0xfffffe00ea3b8d90
--- trap 0x16, rip = 0xffffffff8488afcb, rsp = 0xfffffe00ea3b8e60, rbp = 0xfffffe00ea3b91c0 ---
dcn20_create_resource_pool() at dcn20_create_resource_pool+0x13ab/frame 0xfffffe00ea3b91c0
dc_create_resource_pool() at dc_create_resource_pool+0xc0/frame 0xfffffe00ea3b9220
dc_create() at dc_create+0x2df/frame 0xfffffe00ea3b9280
dm_hw_init() at dm_hw_init+0x27a/frame 0xfffffe00ea3b93b0
amdgpu_device_init() at amdgpu_device_init+0x1d27/frame 0xfffffe00ea3b9480
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0xd0/frame 0xfffffe00ea3b94b0
drm_dev_register() at drm_dev_register+0xc6/frame 0xfffffe00ea3b94e0
amdgpu_pci_probe() at amdgpu_pci_probe+0x17d/frame 0xfffffe00ea3b9520
linux_pci_attach_device() at linux_pci_attach_device+0x569/frame 0xfffffe00ea3b9580
device_attach() at device_attach+0x3dd/frame 0xfffffe00ea3b95d0
bus_generic_driver_added() at bus_generic_driver_added+0xb6/frame 0xfffffe00ea3b9600
devclass_driver_added() at devclass_driver_added+0x39/frame 0xfffffe00ea3b9640
devclass_add_driver() at devclass_add_driver+0x13d/frame 0xfffffe00ea3b9680
_linux_pci_register_driver() at _linux_pci_register_driver+0xdf/frame 0xfffffe00ea3b96b0
amdgpu_evh() at amdgpu_evh+0x92/frame 0xfffffe00ea3b96c0
module_register_init() at module_register_init+0xa4/frame 0xfffffe00ea3b96f0
linker_load_module() at linker_load_module+0xba9/frame 0xfffffe00ea3b9a10
kern_kldload() at kern_kldload+0xb8/frame 0xfffffe00ea3b9a60
sys_kldload() at sys_kldload+0x5b/frame 0xfffffe00ea3b9a90
amd64_syscall() at amd64_syscall+0x119/frame 0xfffffe00ea3b9bb0
fast_syscall_common() at fast_syscall_common+0x101/frame 0xfffffe00ea3b9bb0
--- syscall (304, FreeBSD ELF64, sys_kldload), rip = 0x8002db06a, rsp = 0x7fffffffd538, rbp = 0x7fffffffdab0 ---
KDB: enter: panic
```

I looked at the `dcn20_create_resource_pool()` function and it seems to happen somewhere in the pool's construction. The file has a `kernel_fpu_begin/end` pair in one function, but perhaps FreeBSD is being more strict or something is touching the FPU along the way, or something else is wrong with the functions. There are some places where floating-point variables are being initialized but are not wrapped by these pairs.

I don't know enough about these FPU kernel semantics to fix this. I naively just tried to wrap the `dcn20_create_resource_pool()` with the `kernel_fpu_begin/end`, but that didn't solve the issue (and is likely not reenterant).

I can certainly try out things if there is a need.

Dmesg for the amdgpu load:

```
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] initializing kernel modesetting (NAVI10 0x1002:0x731F 0x1DA2:0xE411 0xC1).
[drm] register mmio base: 0xFCB00000
[drm] register mmio size: 524288
[drm] set register base offset for ATHUB
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[... 80 lines cut ...]
--- trap 0x16, rip = 0xffffffff8488afcb, rsp = 0xfffffe00ea3b8e60, rbp = 0xfffffe00ea3b91c0 ---
dcn20_create_resource_pool() at dcn20_create_resource_pool+0x13ab/frame 0xfffffe00ea3b91c0
dc_create_resource_pool() at dc_create_resource_pool+0xc0/frame 0xfffffe00ea3b9220
dc_create() at dc_create+0x2df/frame 0xfffffe00ea3b9280
dm_hw_init() at dm_hw_init+0x27a/frame 0xfffffe00ea3b93b0
amdgpu_device_init() at amdgpu_device_init+0x1d27/frame 0xfffffe00ea3b9480
amdgpu_driver_load_kms() at amdgpu_driver_load_kms+0xd0/frame 0xfffffe00ea3b94b0
drm_dev_register() at drm_dev_register+0xc6/frame 0xfffffe00ea3b94e0
amdgpu_pci_probe() at amdgpu_pci_probe+0x17d/frame 0xfffffe00ea3b9520
linux_pci_attach_device() at linux_pci_attach_device+0x569/frame 0xfffffe00ea3b9580
device_attach() at device_attach+0x3dd/frame 0xfffffe00ea3b95d0
bus_generic_driver_added() at bus_generic_driver_added+0xb6/frame 0xfffffe00ea3b9600
devclass_driver_added() at devclass_driver_added+0x39/frame 0xfffffe00ea3b9640
devclass_add_driver() at devclass_add_driver+0x13d/frame 0xfffffe00ea3b9680
_linux_pci_register_driver() at _linux_pci_register_driver+0xdf/frame 0xfffffe00ea3b96b0
amdgpu_evh() at amdgpu_evh+0x92/frame 0xfffffe00ea3b96c0
module_register_init() at module_register_init+0xa4/frame 0xfffffe00ea3b96f0
linker_load_module() at linker_load_module+0xba9/frame 0xfffffe00ea3b9a10
kern_kldload() at kern_kldload+0xb8/frame 0xfffffe00ea3b9a60
sys_kldload() at sys_kldload+0x5b/frame 0xfffffe00ea3b9a90
amd64_syscall() at amd64_syscall+0x119/frame 0xfffffe00ea3b9bb0
fast_syscall_common() at fast_syscall_common+0x101/frame 0xfffffe00ea3b9bb0
--- syscall (304, FreeBSD ELF64, sys_kldload), rip = 0x8002db06a, rsp = 0x7fffffffd538, rbp = 0x7fffffffdab0 ---
KDB: enter: panic
```

evadot, 2020-06-25:
I don't see where some fpu instructions are used so it's a bit weird.
Also since a lot function are static they are inlined by the compiler.
I suggest you test recompiling by adding :
__attribute__ ((noinline)) just before every function name for every static function in this file.
i.e.:
`-static bool construct(
+static bool __attribute__ ((noinline)) construct(
`
Then the panic will show the real function using fpu instr and you can then add some kernel_fpu_begin/kernel_fpu_end in it.

NorwegianRockCat, 2020-06-26:
I added the noinline attribute and recompiled. It seems that the FPU is being triggered inside construct(). I added a call to kernel_fpu_begin to the beginning and end of the function. Then, the screen goes blank and the computer freezes (can't ping or ssh into it) and I have to do a hard restart.

I thought it might be some sort of problem with holding the context too long. So, I added a bunch of DRM_INFOs to see if I could limit the section where it is called and it seems that the panic happens *after* the call to `dml_init_instance()` but *before* the call to  `dal_irq_service_dcn20_create()`.

I then wrapped this section of code kernel_fpu_begin/kernel_fpu_end instead, but I got the same result as before (a blank screen and an unresponsive machine). It doesn't panic, so I don't know what is going wrong.

I must admit I'm stumped. Any suggestions? Would you like the panic from after all the inlining?

NorwegianRockCat, 2020-06-27:
I had some more time to narrow this down more. The lines in construct are around 3025 and 3026:

```
  ranges.reader_wm_sets[i].min_fill_clk_mhz = (i > 0) ? (dcn2_0_soc.clock_limits[i - 1].dram_speed_mts / 16) + 1 : 0;
  ranges.reader_wm_sets[i].max_fill_clk_mhz = dcn2_0_soc.clock_limits[i].dram_speed_mts / 16;
```
The element `dram_speed_mts` is a **double**.

I don't know why this is OK in Linux, but it appears, to me at least, that it is a floating point operation.

Regardless the `kernel_fpu_begin` and `kernel_fpu_end` make it through the first iteration of the loop here, but then the screen goes completely blank and the computer locks up during a later iteration requiring a hard restart.

Since the screen goes blank and it doesn't panic, I don't know how to get any more debug information at the moment.

evadot, 2020-06-28:
It could be that it panics but in the middle of switching vt.
You could try blindly typing 'dump' and then after a few seconds 'reboot'.

NorwegianRockCat, 2020-06-29:
I thought I had actually tried that, but obviously I had typed incorrectly. *doh*

It turns out there are several FPU accesses as part of the DCN20 init. Through a process of adding `kernel_fpu_begin/end` pairs and rebooting, I finally got the thing to load. Hooray!

The code currently has a lot of debug information and attributes and what not. I hope to have some time later in the week or over the weekend to clean it up and I can send a pull request.

Now, to the next step, getting X to work.

evadot, 2020-06-30:
> I thought I had actually tried that, but obviously I had typed incorrectly. _doh_
>
> It turns out there are several FPU accesses as part of the DCN20 init. Through a process of adding [...]

Great news !

NorwegianRockCat, 2020-07-05:
I have now [made the pull request](https://github.com/freebsd/drm-kmod/pull/12). I have no idea if I followed the style correctly, but I'm certainly willing to make changes.

closed 2020-07-05 by NorwegianRockCat

zoujiaqing, 2020-07-06:
Now upgrade notice: `kldload: an error occurred while loading module amd gpu. Please cheack dmesg(8) for more details.`

Please upgrade it : https://www.freshports.org/graphics/drm-devel-kmod/

evadot, 2020-07-06:
> Now upgrade notice: `kldload: an error occurred while loading module amd gpu. Please cheack dmesg(8) for more details.`
> 
> Please upgrade it : https://www.freshports.org/graphics/drm-devel-kmod/

Could you give more details ?
Are you using the port or directly the master branch from here ?

zoujiaqing, 2020-07-07:
pkg version:
```text
pkg-devel-kmod-5.3.g20200612
```

FreeBSD version:
```text
FreeBSD 13-CURRENT #2 9963b6e1cfe-c269658: Mon Jun 29 14:20:24 CST 2020 GENERIC amd64
```

NorwegianRockCat, 2020-07-07:
That package is from 12 June. 20200612 == 12 June 2020, and it seems to be out of sync with the kernel version you built.

Regardless, it won't have the changes from my pull request in there yet.

For the moment, if you want the change you'll need to clone and build this repo, or wait until the port gets updated.

zoujiaqing, 2020-07-07:
Thanks @NorwegianRockCat , I install it from source code.
but notice:
```shell
# kldload amggpu
Jul   7 20:11:56 freebsd kernel: pm_runtime_mark_last_busy not implemented -- see your local. kernel hacker
```

I need upgrade kernel?

zoujiaqing, 2020-07-08:
@NorwegianRockCat , I upgrade  kernel from latest freebsd source code. The problem has not been resolved.

NorwegianRockCat, 2020-07-08:
@zoujiaqing  That message is that that system call is not yet implemented. You are running a development version of the OS, so not everything is completed. If you have a local kernel hacker, you can perhaps persuade them to implement it.

As long as you have the console after the module has loaded, then it everything has loaded and you should be ready to go with a the mesa-devel package.

zoujiaqing, 2020-07-09:
Very good! Thank you @NorwegianRockCat ;)