freebsd/drm-kmod #13: no screens found error

lework, 2020-07-13:
I encountered no screens found error in freebsd 13, the following is detailed information, how can I solve this situation?

kernel: 
```
13.0-CURRENT #2 629b165ee71-c269837(master): Fri Jul 10 22:50:17 CST 2020
```

drm-devel-kmod: 

```
$ pkg info drm-devel-kmod 
drm-devel-kmod-5.3.g20200708
Name           : drm-devel-kmod
Version        : 5.3.g20200708
Installed on   : Mon Jul 13 11:14:06 2020 CST
Origin         : graphics/drm-devel-kmod
Architecture   : FreeBSD:13:amd64
Prefix         : /usr/local
Categories     : graphics kld
Licenses       : BSD2CLAUSE, MIT, GPLv2
Maintainer     : x11@FreeBSD.org
WWW            : https://github.com/FreeBSDDesktop/kms-drm
Comment        : DRM modules for the linuxkpi-based KMS components (development version)
Options        :
        DEBUG          : off
Annotations    :
        FreeBSD_version: 1300100
Flat size      : 11.0MiB
Description    :
amdgpu, i915, and radeon DRM modules for the linuxkpi-based KMS components.
Currently corresponding to Linux 5.0 DRM.
This version is the development version and only works on FreeBSD CURRENT.
amdgpu and radeonkms are known to fail with EFI boot.

WWW: https://github.com/FreeBSDDesktop/kms-drm
```

vga：
```
$ pciconf -lv | grep -B4 VGA  
vgapci0@pci0:3:0:0:     class=0x030000 rev=0xc1 hdr=0x00 vendor=0x1002 device=0x15d8 subvendor=0x19e5 subdevice=0x3e18
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Picasso'
    class      = display
    subclass   = VGA
```

/etc/X11/xorg.conf:
```
Section "Device"
        Identifier  "Card0"
        Driver      "amdgpu"
        BusID       "PCI:3:0:0"
        Option "TearFree" "On"
        Option "DRI" "True"
        Option "AccelMethod" "exa"
        Option "MigrationHeuristic" "greedy"
EndSection
```

boot log:

```
Trying to mount root from ufs:/dev/nvd0p10 [rw]...
WARNING: WITNESS option enabled, expect reduced performance.
uhub1: 3 ports with 3 removable, self powered
uhub0: 8 ports with 8 removable, self powered
[drm:drm_core_init] Initialized
[drm] amdgpu kernel modesetting enabled.
ugen1.2: <Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint Device> at usbus1
ugen0.2: <PixArt A4Tech OpticalMouse> at usbus0
ums0 on uhub0
ums0: <PixArt A4Tech OpticalMouse, class 0/0, rev 2.00/1.00, addr 1> on usbus0
ums0: 5 buttons and [XYZ] coordinates ID=0
```

Xorg log:

```
[   372.017] (II) LoadModule: "amdgpu"
[   372.018] (II) Loading /usr/local/lib/xorg/modules/drivers/amdgpu_drv.so
[   372.018] (II) Module amdgpu: vendor="X.Org Foundation"
[   372.018]    compiled for 1.20.8, module version = 19.1.0
[   372.018]    Module class: X.Org Video Driver
[   372.018]    ABI class: X.Org Video Driver, version 24.1
[   372.018] (II) AMDGPU: Driver for AMD Radeon:
        All GPUs supported by the amdgpu kernel driver
[   372.018] (--) Using syscons driver with X support (version 2.0)
[   372.018] (--) using VT number 9

[   372.020] (II) AMDGPU(0): [KMS] drm report modesetting isn't supported.
[   372.020] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[   372.020] (EE) Screen 0 deleted because of no matching config section.
[   372.020] (II) UnloadModule: "amdgpu"
[   372.020] (EE) Device(s) detected, but none match those in the config file.
[   372.020] (EE) 
Fatal server error:
[   372.020] (EE) no screens found(EE) 
[   372.020] (EE) 
```

evadot, 2020-07-13:
[   372.020] (II) AMDGPU(0): [KMS] drm report modesetting isn't supported.
That's a bit weird.
What's the Xorg.log if you don't have any configuration ?

lework, 2020-07-14:
I do not know either...

evadot, 2020-07-24:
So, I have the exact same system, I'm running r363250 on it with the latest package and I don't have this problem.
Please upgrade your packages and test again.
Also do everything work for you if you remove /etc/X11/xorg.conf ?

evadot, 2020-09-14:
Ping ?

lework, 2020-09-14:
I reinstalled the system using version 13 and it worked.

closed 2020-09-14 by lework