freebsd/drm-kmod #14: No console when external HDMI monitor connected

lwhsu, 2020-07-15:
I'm not sure if this is due to the system or drm-kmod, but now after I kldload i915kms.ko in console, the main monitor on my laptop hangs if there is an external HDMI monitor connected, all the following updates only display on the external monitor. If I disconnect the external monitor, the main monitor on the laptop will catch up all the updates. Under X, both monitors work fine.

Is this an intentional behavior? This is different than before and is a bit inconvenient if you want to do presentation.

System information:
`FreeBSD x1c.lwhsu.org 13.0-CURRENT FreeBSD 13.0-CURRENT #19 r363121M: Mon Jul 13 18:41:40 CST 2020     root@x1c.lwhsu.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64`

I've tested this with 5.3.g20200708 and 5.3.g20200710, both have this situation.

Wanted to revert to 5.3.g20200612 but it cannot compile on 13.0-CURRENT r363121 currently.

evadot, 2020-07-24:
I don't have this problem on my skylake system.
I'll test the other ones.
What generation is it ?

lwhsu, 2020-07-24:
It's x1 carbon 6th, Intel Kaby Lake.

lwhsu, 2020-07-24:
After seeing #17, I tried booting with `hw.i915kms.enable_psr=0` in `/boot/loader.conf`.  It restores previous behavior. But manually toggling `hw.i915kms.enable_psr=0` after loading `i915kms.ko` doesn't work.
Also, after entering and exiting X, the behavior is the same as previous, no matter the value of `hw.i915kms.enable_psr` is.

evadot, 2020-07-24:
hw.i915kms.enable_psr is a tunable so you can't change it after booting with sysctl, kenv should work.

lwhsu, 2020-08-08:
BTW, tt is surprisingly that setting `hw.i915kms.enable_psr=0` seems to fix my suspend/resume issue starting from the end of last year.

lwhsu, 2020-08-08:
Should we have that setting by default (at least for Kaby Lake?), or was the setting changed around end of last year?

evadot, 2020-09-14:
I've made the setting default for us because I don't have hardware to reproduce the problem.
Does the latest update to drm-current-kmod or drm-devel-kmod fixes the issue for you ?

lwhsu, 2020-09-14:
Haven't updated to that version. I'll test and report here.

lwhsu, 2021-01-22:
> @lwhsu ping ? Do you still have this issue with latest 13.0-ALPHA1 and latest drm-kmod ?
> Thanks

Sorry that I still cannot update to newer -current because am busy at other projects. Will try out in the weekend.

lwhsu, 2021-01-29:
Tested again with 14.0-CURRENT c256325-g4f009328a247 and drm-devel-kmod-5.4.62.g20210118, looks good to me.

closed 2021-01-29 by lwhsu