freebsd/drm-kmod #19: Kernel panic while transcoding a video with Plex

sisoteuz, 2020-08-09:
Hi,

I upgraded to FreeBSD 13.0-CURRENT r364031 to be able to leverage hardware video transcoding with the integrated Intel chipset on my Celeron G5900 by installing the port drm-devel-kmod.

Whille hardware transcoding has been working fine with various videos, I encountered a kernel crash while a video was being transcoded which occurs all the time on that particular one : 

```
Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0xdf
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff80bdd2b4
stack pointer           = 0x0:0xfffffe00d2be56d0
frame pointer           = 0x0:0xfffffe00d2be56d0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = interrupt enabled, resume, IOPL = 0
current process         = 4611 (Plex Transcoder)
trap number             = 12
panic: page fault
cpuid = 0
time = 1596976796
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00d2be5390
vpanic() at vpanic+0x182/frame 0xfffffe00d2be53e0
panic() at panic+0x43/frame 0xfffffe00d2be5440
trap_fatal() at trap_fatal+0x387/frame 0xfffffe00d2be54a0
trap_pfault() at trap_pfault+0x4f/frame 0xfffffe00d2be54f0
trap() at trap+0x271/frame 0xfffffe00d2be5600
calltrap() at calltrap+0x8/frame 0xfffffe00d2be5600
--- trap 0xc, rip = 0xffffffff80bdd2b4, rsp = 0xfffffe00d2be56d0, rbp = 0xfffffe00d2be56d0 ---
_rw_wowned() at _rw_wowned+0x4/frame 0xfffffe00d2be56d0
vm_page_busy_acquire() at vm_page_busy_acquire+0x141/frame 0xfffffe00d2be5710
remap_io_mapping() at remap_io_mapping+0x120/frame 0xfffffe00d2be5760
i915_gem_fault() at i915_gem_fault+0x25f/frame 0xfffffe00d2be57d0
linux_cdev_pager_populate() at linux_cdev_pager_populate+0x11b/frame 0xfffffe00d2be5840
vm_fault() at vm_fault+0x3d1/frame 0xfffffe00d2be5950
vm_fault_trap() at vm_fault_trap+0x60/frame 0xfffffe00d2be5990
trap_pfault() at trap_pfault+0x19c/frame 0xfffffe00d2be59e0
trap() at trap+0x3f1/frame 0xfffffe00d2be5af0
calltrap() at calltrap+0x8/frame 0xfffffe00d2be5af0
--- trap 0xc, rip = 0x80296659a, rsp = 0x7fffffffbd38, rbp = 0x80fc00000 ---
KDB: enter: panic
```

There is nothing of interest in Plex transcoder or server logs (DEBUG level). 

No crash dump was generated either in `/var/crash`. Not sure what other information to include but happy to run/include whatever is required to investigate this further.

Thank you.

evadot, 2020-09-14:
Do you have an easier way to reproduce ?

evadot, 2021-04-19:
Closing due to no activity.

closed 2021-04-19 by evadot

sisoteuz, 2022-09-01:
I am still having the issue on FreeBSD-13.1 with `drm-fbsd13-kmod-5.4.191.g20220604_1` port on videos that have [ASS ](https://en.wikipedia.org/wiki/SubStation_Alpha)subtitles.

I could finally find a workaround that seems to be consistently working now. I need to compile the drm-fbsd13-kmod with `make -DWITH_DEBUG DEBUG_FLAGS="-O1"`.

So essentially it seems like there is an issue with default `-O2` optimizations which cause the i915 driver to misbehave and cause a kernel crash. Could it be related to some `volatile` behaviour ?

Below some information from the crash dump I get with the driver compiled with `-O2`:

1. **General information**
```
Fatal trap 9: general protection fault while in kernel mode
cpuid = 0; apic id = 00
instruction pointer     = 0x20:0xffffffff80c167c4
stack pointer           = 0x0:0xfffffe0123171a60
frame pointer           = 0x0:0xfffffe0123171a60
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = interrupt enabled, resume, IOPL = 0
current process         = 6423 (Plex Transcoder)
trap number             = 9
panic: general protection fault
```

2. **Backtrace**
```
(kgdb) bt
#0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  doadump (textdump=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:399
#2  0xffffffff80c1b75c in kern_reboot (howto=260) at /usr/src/sys/kern/kern_shutdown.c:487
#3  0xffffffff80c1bbce in vpanic (fmt=0xffffffff811b4fb9 "%s", ap=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:920
#4  0xffffffff80c1b9d3 in panic (fmt=<unavailable>) at /usr/src/sys/kern/kern_shutdown.c:844
#5  0xffffffff810afdf5 in trap_fatal (frame=0xfffffe01231719a0, eva=0) at /usr/src/sys/amd64/amd64/trap.c:944
#6  <signal handler called>
#7  _rw_wowned (c=0x2659c92217d5aa52) at /usr/src/sys/kern/kern_rwlock.c:270
#8  0xffffffff80f66027 in vm_page_busy_acquire (m=0xfffffe00040ff9e8, allocflags=allocflags@entry=16) at /usr/src/sys/vm/vm_page.c:888
#9  0xffffffff82b91940 in remap_io_mapping (vma=vma@entry=0xfffff800253ec100, addr=<optimized out>, pfn=<optimized out>, size=<optimized out>,
    iomap=iomap@entry=0xfffffe00d964c6e0) at /var/ports/usr/ports/graphics/drm-fbsd13-kmod/work/drm-kmod-drm_v5.4.191_1/drivers/gpu/drm/i915/intel_freebsd.c:193
#10 0xffffffff82c2a401 in i915_gem_fault (dummy=<optimized out>, vmf=<optimized out>)
    at /var/ports/usr/ports/graphics/drm-fbsd13-kmod/work/drm-kmod-drm_v5.4.191_1/drivers/gpu/drm/i915/gem/i915_gem_mman.c:366
#11 0xffffffff80e5ea3b in linux_cdev_pager_populate (vm_obj=0xfffff801850a2630, pidx=<optimized out>, fault_type=<optimized out>, max_prot=<optimized out>,
    first=0xfffffe0123171c00, last=0xfffffe0123171c18) at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:691
#12 0xffffffff80f4f9cb in vm_pager_populate (object=0x2659c92217d5aa52, object@entry=0xfffffe0123171b90, pidx=16, fault_type=1024, first=0xfffffe0123171c00,
    last=0xfffffe0123171c18, max_prot=<optimized out>) at /usr/src/sys/vm/vm_pager.h:182
#13 vm_fault_populate (fs=0xfffffe0123171cb0) at /usr/src/sys/vm/vm_fault.c:469
#14 vm_fault_allocate (fs=fs@entry=0xfffffe0123171cb0) at /usr/src/sys/vm/vm_fault.c:1163
#15 0xffffffff80f4e6bf in vm_fault_object (fs=0xfffffe0123171cb0, behindp=0xfffffe0123171ca4, aheadp=0xfffffe0123171cac) at /usr/src/sys/vm/vm_fault.c:1414
#16 vm_fault (map=<optimized out>, map@entry=0xfffffe011f5183f0, vaddr=vaddr@entry=34521219072, fault_type=<optimized out>, fault_type@entry=1 '\001',
    fault_flags=<optimized out>, fault_flags@entry=0, m_hold=<optimized out>, m_hold@entry=0x0) at /usr/src/sys/vm/vm_fault.c:1549
#17 0xffffffff80f4e2cd in vm_fault_trap (map=0xfffffe011f5183f0, vaddr=vaddr@entry=34521219072, fault_type=<optimized out>, fault_flags=fault_flags@entry=0,
    signo=0xfffffe0123171f04, ucode=0xfffffe0123171f00) at /usr/src/sys/vm/vm_fault.c:667
#18 0xffffffff810afff3 in trap_pfault (frame=frame@entry=0xfffffe0123171f40, usermode=true, signo=0xfffffe00d964c6e0, signo@entry=0xfffffe0123171f04, ucode=0x2,
    ucode@entry=0xfffffe0123171f00) at /usr/src/sys/amd64/amd64/trap.c:848
#19 0xffffffff810af61a in trap (frame=0xfffffe0123171f40) at /usr/src/sys/amd64/amd64/trap.c:383
#20 <signal handler called>
#21 0x00000008025837aa in ?? ()
```

3. **Instruction that caused the crash**
```
(kgdb) l *0xffffffff80c167c4
0xffffffff80c167c4 is in _rw_wowned (/usr/src/sys/kern/kern_rwlock.c:270).
265
266     int
267     _rw_wowned(const volatile uintptr_t *c)
268     {
269
270             return (rw_wowner(rwlock2rw(c)) == curthread);
271     }
272
273     void
274     _rw_wlock_cookie(volatile uintptr_t *c, const char *file, int line)
```

Let me know if you need more details from the backtrace.

SebastianChudoba, 2022-09-22:
Yes, please re-open. hw transcoding with plex is still unstable on freebsd 13.1

allanjude, 2023-03-14:
Small reproduction: compile ffmpeg with quicksync (qsv) support:

try to transcode a video file using hevc_qsv

`ffmpeg -y -i bunny_hevc.ts -t 30 -vcodec hevc_qsv -b:v 8000K -acodec aac bunny_hevc.mp4`

```
Reading symbols from /boot/kernel/kernel...
Reading symbols from /usr/lib/debug//boot/kernel/kernel.debug...

Unread portion of the kernel message buffer:

Fatal trap 9: general protection fault while in kernel mode
cpuid = 3; apic id = 03
instruction pointer	= 0x20:0xffffffff80c02a24
stack pointer	        = 0x0:0xfffffe011295c9a0
frame pointer	        = 0x0:0xfffffe011295c9a0
code segment		= base 0x0, limit 0xfffff, type 0x1b
			= DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags	= interrupt enabled, resume, IOPL = 0
current process		= 1137 (ffmpeg)
trap number		= 9
panic: general protection fault
cpuid = 3
time = 1678822386
KDB: stack backtrace:
#0 0xffffffff80c553b5 at kdb_backtrace+0x65
#1 0xffffffff80c07dd1 at vpanic+0x151
#2 0xffffffff80c07c73 at panic+0x43
#3 0xffffffff810c1fa7 at trap_fatal+0x387
#4 0xffffffff810991a8 at calltrap+0x8
#5 0xffffffff80f78e3a at vm_page_busy_acquire+0xfa
#6 0xffffffff80e67421 at lkpi_vmf_insert_pfn_prot_locked+0xb1
#7 0xffffffff82d351f0 at remap_io_mapping+0xb0
#8 0xffffffff82e4f366 at vm_fault_gtt+0x406
#9 0xffffffff80e61c5b at linux_cdev_pager_populate+0x11b
#10 0xffffffff80f6299e at vm_fault_allocate+0x32e
#11 0xffffffff80f6168f at vm_fault+0x2ef
#12 0xffffffff80f6128d at vm_fault_trap+0x6d
#13 0xffffffff810c21a3 at trap_pfault+0x1f3
#14 0xffffffff810c1790 at trap+0x440
#15 0xffffffff810991a8 at calltrap+0x8
Timeout initializing vt_vga
Uptime: 1m33s
Dumping 762 out of 16239 MB:..3%..11%..21%..32%..42%..51%..61%..72%..82%..91%

__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
55		__asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
(kgdb) bt
#0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  dump_savectx () at /usr/src/sys/kern/kern_shutdown.c:392
#2  0xffffffff80c07968 in dumpsys (di=0x0) at /usr/src/sys/x86/include/dump.h:87
#3  doadump (textdump=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:421
#4  kern_reboot (howto=260) at /usr/src/sys/kern/kern_shutdown.c:495
#5  0xffffffff80c07e3e in vpanic (fmt=<optimized out>, ap=ap@entry=0xfffffe011295c850) at /usr/src/sys/kern/kern_shutdown.c:934
#6  0xffffffff80c07c73 in panic (fmt=<unavailable>) at /usr/src/sys/kern/kern_shutdown.c:858
#7  0xffffffff810c1fa7 in trap_fatal (frame=0xfffffe011295c8e0, eva=0) at /usr/src/sys/amd64/amd64/trap.c:942
#8  <signal handler called>
#9  _rw_wowned (c=0xff29bd39fd2b29ef) at /usr/src/sys/kern/kern_rwlock.c:270
#10 0xffffffff80f78e3a in vm_page_busy_acquire (m=0xfffffe0004dffb88, allocflags=allocflags@entry=16) at /usr/src/sys/vm/vm_page.c:888
#11 0xffffffff80e67421 in lkpi_vmf_insert_pfn_prot_locked (vma=0xfffff8018954b800, addr=<optimized out>, pfn=<optimized out>, prot=24)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_page.c:300
#12 0xffffffff82d351f0 in remap_io_mapping () from /boot/modules/i915kms.ko
#13 0xffffffff82e4f366 in vm_fault_gtt () from /boot/modules/i915kms.ko
#14 0xffffffff80e61c5b in linux_cdev_pager_populate (vm_obj=0xc0005, pidx=<optimized out>, fault_type=<optimized out>, max_prot=<optimized out>, first=0xfffffe011295cc00, 
    last=0xfffffe011295cc18) at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:694
#15 0xffffffff80f6299e in vm_pager_populate (object=0xff29bd39fd2b29ef, object@entry=0xfffffe011295cb90, pidx=16, fault_type=1024, first=0xfffffe011295cc00, 
    last=0xfffffe011295cc18, max_prot=<optimized out>) at /usr/src/sys/vm/vm_pager.h:188
#16 vm_fault_populate (fs=0xfffffe011295cca0) at /usr/src/sys/vm/vm_fault.c:469
#17 vm_fault_allocate (fs=fs@entry=0xfffffe011295cca0) at /usr/src/sys/vm/vm_fault.c:1163
#18 0xffffffff80f6168f in vm_fault_object (fs=0xfffffe011295cca0, behindp=0xfffffe011295cc94, aheadp=0xfffffe011295cc9c) at /usr/src/sys/vm/vm_fault.c:1418
#19 vm_fault (map=map@entry=0xfffffe0110c3a3f0, vaddr=vaddr@entry=37572575232, fault_type=fault_type@entry=2 '\002', fault_flags=fault_flags@entry=0, m_hold=m_hold@entry=0x0)
    at /usr/src/sys/vm/vm_fault.c:1553
#20 0xffffffff80f6128d in vm_fault_trap (map=0xfffffe0110c3a3f0, vaddr=vaddr@entry=37572575232, fault_type=<optimized out>, fault_flags=fault_flags@entry=0, 
    signo=0xfffffe011295cf00, ucode=0xfffffe011295cf04) at /usr/src/sys/vm/vm_fault.c:667
#21 0xffffffff810c21a3 in trap_pfault (frame=frame@entry=0xfffffe011295cf40, usermode=true, signo=0xfffffe00dcbf05e8, signo@entry=0xfffffe011295cf00, ucode=0x3004, 
    ucode@entry=0xfffffe011295cf04) at /usr/src/sys/amd64/amd64/trap.c:846
#22 0xffffffff810c1790 in trap (frame=0xfffffe011295cf40) at /usr/src/sys/amd64/amd64/trap.c:385
#23 <signal handler called>
#24 0x0000000827ed265a in ?? ()
Backtrace stopped: Cannot access memory at address 0x8960fb538
```

claudiozz, 2024-11-30:
I think this has been closed incorrectly (the commit above seems unrelated). The issue is still happening

I'm experiencing a kernel panic when using VA-API for hardware-accelerated video transcoding on Freebsd 14.1.

CPU: Intel(R) Core(TM) i7-2600
GPU: Intel HD Graphics 2000

Driver Versions:
libva-intel-driver-2.4.1_3
libva-intel-hybrid-driver-1.0.2_3
libva-intel-media-driver-23.4.3
drm-61-kmod-6.1.92_2
libdrm-2.4.123,1

Kernel Panic (Backtrace):
```
KDB: stack backtrace:
#0 0xffffffff80b7207d at kdb_backtrace+0x5d
#1 0xffffffff80b25631 at vpanic+0x131
#2 0xffffffff80b25453 at panic+0x43
#3 0xffffffff8100dd7b at trap_fatal+0x40b
#4 0xffffffff80fe0d58 at calltrap+0x8
#5 0xffffffff80ec4191 at vm_page_busy_acquire+0xc1
#6 0xffffffff80daa998 at lkpi_vmf_insert_pfn_prot_locked+0x118
#7 0xffffffff82834060 at remap_io_mapping+0xb0
#8 0xffffffff82986487 at vm_fault_gtt+0x4f7
#9 0xffffffff80da4b6f at linux_cdev_pager_populate+0x15f
#10 0xffffffff80eac6ac at vm_fault_allocate+0x39c
#11 0xffffffff80eaaf0d at vm_fault+0x43d
#12 0xffffffff80eaa96d at vm_fault_trap+0x5d
#13 0xffffffff8100df9d at trap_pfault+0x21d
#14 0xffffffff8100d56d at trap+0x51d
#15 0xffffffff80fe0d58 at calltrap+0x8
```

Repro:

fetch  https://mirror.clarkson.edu/blender/demo/movies/BBB/bbb_sunflower_2160p_60fps_normal.mp4
Run the ffmpeg command in the attached [repro.sh](https://github.com/user-attachments/files/17967337/repro.sh.txt)

Full crash dump: [vmcore.6.txt](https://github.com/user-attachments/files/17967338/vmcore.6.txt)

Vainfo, kernel backtrace and pciconf output: [vainfo-pciconf.txt](https://github.com/user-attachments/files/17967339/vainfo-pciconf.txt)