freebsd/drm-kmod #22: drm-devel-kmod 5.4.62.g20200905_1 fails on powerpc64

pkubaj, 2020-09-11:
It fails at building amdgpu, which doesn't work on powerpc64 anyway, so you could actually only build radeonkms on powerpc64:
```
cc  -O2 -pipe -fno-strict-aliasing '-DKBUILD_MODNAME="amdgpu"' '-DLINUXKPI_PARAM_PREFIX=amdgpu_' -DDRM_SYSCTL_PARAM_PREFIX=_amdgpu -DLINUXKPI_VERSION=50000 -DCONFIG_DRM_AMDGPU_CIK -DCONFIG_DRM_AMDGPU_SI -DCONFIG_DRM_AMD_DC -DCONFIG_DRM_AMD_DC_FBC -DCONFIG_DRM_AMD_POWERPLAY -DCONFIG_DRM_I915_ALPHA_SUPPORT -DCONFIG_DRM_I915_FORCE_PROBE='"*"' -DCONFIG_DRM_I915_CAPTURE_ERROR -DCONFIG_DRM_I915_SPIN_REQUEST=5 -DCONFIG_DRM_I915_USERFAULT_AUTOSUSPEND=250 -DCONFIG_DRM_LOAD_EDID_FIRMWARE -DCONFIG_DRM_MIPI_DSI -DCONFIG_DRM_PANEL_ORIENTATION_QUIRKS -DCONFIG_DRM_VMWGFX_FBCON -DCONFIG_DRM_FBDEV_EMULATION -DCONFIG_DRM_FBDEV_OVERALLOC=100 -DCONFIG_DRM_LEGACY -DCONFIG_DRM_VM -DCONFIG_ARCH_HAVE_NMI_SAFE_CMPXCHG -DCONFIG_BACKLIGHT_CLASS_DEVICE -DCONFIG_DMI -DCONFIG_FB -DCONFIG_MTRR -DCONFIG_PCI -DCONFIG_PM -DCONFIG_SMP  -Werror -D_KERNEL -DKLD_MODULE -nostdinc  -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/linuxkpi/gplv2/include -I/usr/src/sys/compat/linuxkpi/common/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/linuxkpi/dummy/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/include/drm -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/include/uapi -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/acp/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/amdgpu -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/amdgpu_dm -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/dc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/dc/clk_mgr/ -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/dc/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/dc/inc/hw -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/modules/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/include/asic_reg -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/powerplay/hwmgr -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/powerplay/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/powerplay/smumgr -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/scheduler -include /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/amd/amdgpu/opt_global.h -I. -I/usr/src/sys
-I/usr/src/sys/contrib/ck/include -fno-common  -fPIC -mlongcall -fno-omit-frame-pointer -fdebug-prefix-map=./machine=/usr/src/sys/powerpc/include     -MD  -MF.depend.amdgpu_dm_pp_smu.o -MTamdgpu_dm_pp_smu.o -mno-altivec -msoft-float -mabi=elfv2 -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error-tautological-compare -Wno-error-empty-body -Wno-error-parentheses-equality -Wno-error-unused-function -Wno-error-pointer-sign -Wno-error-shift-negative-value -Wno-address-of-packed-member -Wno-format-zero-length -Wno-pointer-arith -Wno-format -Wno-cast-qual -Wno-pointer-sign -Wno-missing-prototypes -Wno-enum-conversion -Wno-incompatible-pointer-types -Wno-missing-prototypes -Wno-enum-conversion -Wno-incompatible-pointer-types    -std=iso9899:1999 -c /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_pp_smu.c -o amdgpu_dm_pp_smu.o
/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62/drivers/gpu/drm/amd/display/amdgpu_dm/amdgpu_dm_pp_smu.c:821:21: error: unknown type name 'BOOLEAN'
        struct pp_smu *pp, BOOLEAN pstate_handshake_supported)
                           ^
1 error generated.
```

pkubaj, 2020-09-11:
Removing `#ifdef linux` around `typedef bool BOOLEAN` from `drivers/gpu/drm/amd/display/dc/dm_pp_smu.h` fixes this build issue.

evadot, 2020-09-11:
The typedef caused some problems when it was introduced, I'll try on amd64 if that's not the case anymore and if it's still the case I'll remove amdgpu from the powerpc64 list.
Thanks for the report.

commit 0975f440 (2020-09-14): amdgpu: Fix compile on powerpc64

The BOOLEAN typedef is defined when CONFIG_ACPI is selected which is not
the case for powerpc64.
Only define it when we aren't compiling for i386, amd64 or aarch64.

Fixes: #22
commit 8c1da9f8 (2020-09-14): amdgpu: Fix compile on powerpc64

The BOOLEAN typedef is defined when CONFIG_ACPI is selected which is not
the case for powerpc64.
Only define it when we aren't compiling for i386, amd64 or aarch64.

Fixes: #22
closed 2020-09-14 by evadot