freebsd/drm-kmod #23: Panic on boot with Renoir when testing drm-live-devel-20200906.img

dnebdal, 2020-09-11:
Thinkpad T14 Gen1 AMD
Ryzen 5 PRO 4650U with Radeon Graphics
FreeBSD 13.0-CURRENT 76ceb436904-271329(drm-devel)
(Specifically the image linked in freebsd-current by Emmanuel Vadot on Aug 31).

Issue:
Boots in EFI framebuffer. Prints a few errors, then panics. Until I figure out how to get a crash dump out of this, here is some hand transcription. The boot proceeds normally until "Loading kernel modules:", so I'll skip to there.

This should be entirely reproducible, so ask if there is anything you'd like me to provide.

```
Loading kernel modules:
<6>[drm] amdgpu kernel modesetting enabled
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
sysctl_warn_reuse: can't reuse a leaf (hw.dri.debug)!
<6>[drm] initializing kernel modesetting (RENOIR 0x1002:0x1636 0x17AA:0x5081 0xD3)
<6>[drm] register mmio base: 0xFD300000
<6>[drm] register mmio size 524288
<6>[drm] add ip block number 0 <soc15_common>
<6>[drm] add ip block number 1 <gmc_v9_0>
<6>[drm] add ip block number 2 <vega10_ih>
<6>[drm] add ip block number 3 <psp>
<6>[drm] add ip block number 4 <smu>
<6>[drm] add ip block number 5 <gfx_v9_0>
<6>[drm] add ip block number 6 <sdma_v4_0>
<6>[drm] add ip block number 7 <vcn_v2_0>
drm0: failed to link firmware kernel module with mapped name: amdgpu_renoir_gpu_info_bin
amdgpu/renoir_gpu_info.bin: could not load firmware image, error 2
amdgpu/renoir_gpu_info.bin: could not load firmware image, error 2
drmn0: failed to load firmware with name: amdgpu/renoir_gpu_info.bin
drmn0: failed to load gpu_info firmware: "amdgpu/renoir_gpu_info.bin"
drmn0: fatal error during GPU init
unregister_acpi_notifier not implemented -- see your local kernel hacker
<6>[drm] amdpu: finishing device.
panic: mtx_lock() of spin mutex (null) @ /wrkdirs/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62-devel/drivers/gpu/drm/amd/amdgpu/amdgpu_irq.c:112
cpuid = 10
time= 1599860909
KDB: stack backtrace:
db_trace_self_wrapper()  (I'll skip the addresses; see attached image)
vpanic()
panic()
__mtx_lock_flags()
amdgpu_irq_disable_all()
amdgpu_device_fini()
amdgpu_driver_unload_kms()
amdgpu_driver_load_kms()
drm_dev_register()
amdgpu_pci_probe()
linux_pci_attach_device()
device_attach()
device_probe_and_attach()
bus_generic_driver_added()
devclass_driver_added()
devclass_add_driver()
_linux_pci_register_driver()
amdgpu_evh()
module_register_init()
linker_load_module()
kern_kldload()
sys_kldload()
amd64_syscall()
fast_syscall_common()
```
[image]

evadot, 2020-09-11:
The issue is with unloading the amdgpu module, it never really worked but now instead of hanging the machine it panics (don't know if it's better or not :P).
It seems that gpu-firmware needs an update because we don't ship the renoir firmware which in turn causes the driver to unattach.
I'll update the gpu-firmware modules next week if no one beats me to it.
Thanks for the report.

dnebdal, 2020-09-12:
Between the two, the panic seems ... better, I guess. :)

And yeah those firmware messages looked kind of relevant. I'll try again sometime next week.

evadot, 2020-09-14:
I've updated the gpu-firmware-kmod ports to include the renoir firmware files.
I'll generate some new image this week.