freebsd/drm-kmod #24: DRM_IOCTL_I915_GEM_VM_DESTROY fails with ENOENT

jbeich, 2020-09-20:
https://github.com/intel/compute-runtime/commit/519e75e3d683 (since https://github.com/freebsd/freebsd-ports/commit/09716dd57dc1) started to use VM ioctl. While "create" ioctl appears to succeed "destroy" fails, triggering assert on exit.

```
$ clinfo
[...]
Abort was called at 243 line in file:
shared/source/os_interface/linux/drm_neo.cpp
```

```c
$ cc a.c $(pkg-config libdrm --cflags --libs)
$ ./a.out
a.out: DRM_IOCTL_I915_GEM_VM_DESTROY failed: No such file or directory

$ cat a.c
#include <i915_drm.h>
#include <xf86drm.h>
#include <err.h>

// https://gitlab.freedesktop.org/mesa/drm/-/issues/53
#define DRM_I915_GEM_VM_CREATE		0x3a
#define DRM_I915_GEM_VM_DESTROY		0x3b
#define DRM_IOCTL_I915_GEM_VM_CREATE	DRM_IOWR(DRM_COMMAND_BASE + DRM_I915_GEM_VM_CREATE, struct drm_i915_gem_vm_control)
#define DRM_IOCTL_I915_GEM_VM_DESTROY	DRM_IOW (DRM_COMMAND_BASE + DRM_I915_GEM_VM_DESTROY, struct drm_i915_gem_vm_control)

int main()
{
  int fd;

  if ((fd = drmOpenWithType("i915", 0, DRM_NODE_RENDER)) < 0)
    err(1, "drmOpenWithType failed");

  struct drm_i915_gem_vm_control ctl = {};
  if (drmIoctl(fd, DRM_IOCTL_I915_GEM_VM_CREATE, &ctl) < 0)
    err(1, "DRM_IOCTL_I915_GEM_VM_CREATE failed");
  if (drmIoctl(fd, DRM_IOCTL_I915_GEM_VM_DESTROY, &ctl) < 0)
    err(1, "DRM_IOCTL_I915_GEM_VM_DESTROY failed");

  return 0;
}
```

jbeich, 2020-09-20:
I've submitted a workaround to NEO for feedback. However, the correct fix is probably to abort `DRM_IOCTL_I915_GEM_VM_CREATE` due to broken ppGTT on FreeBSD.
https://github.com/freebsd/drm-kmod/blob/9ff9e4a4c77b46ba039521f1939b81de1ceef8f5/drivers/gpu/drm/i915/gem/i915_gem_context.c#L805

evadot, 2022-06-06:
Your code seems to be working here on 5.10 now.

jbeich, 2022-06-12:
Indeed. Works fine on `drm-510-kmod` but not `drm-54-kmod`. Closing as `drm-510-kmod` is supported on FreeBSD 13.1+, `intel-compute-runtime` has the workaround upstreamed and `libva-intel-media-driver` only uses virtual engine (`FtrContextBasedScheduling`) on Gen11+.

closed 2022-06-12 by jbeich