freebsd/drm-kmod #32: amdgpu vaapi failure due to dma_resv_locking_ctx returning 0

myfreeweb, 2020-09-30:
Finally got around to debugging the video acceleration on Vega again. (HEVC *encoding* was working, H264 encoding and ALL *decoding* broke at some point)

Turns out this "double check" is failing: https://github.com/freebsd/drm-kmod/blob/2b7442a5bc2a7209908a6697ac60bb81ad10b9a0/drivers/gpu/drm/amd/amdgpu/amdgpu_cs.c#L1730-L1732

because the `dma_resv_locking_ctx(..)` result is zero. Everything is working fine when the `return` is replaced with a `printf`.

myfreeweb, 2020-10-04:
Looking a bit deeper..

```
dma_resv_locking_ctx == READ_ONCE(obj->lock.ctx)
struct dma_resv *obj -> struct ww_mutex lock
```

```c
struct ww_acquire_ctx {
};
struct ww_mutex {
	struct mutex base;
	struct cv condvar;
	struct ww_acquire_ctx *ctx;
};
```

there is no 'ctx =' in `ww_mutex.h`, seems like nothing ever sets that field. `ww_mutex_lock(_interruptible)` just ignores the `ctx` argument! I'll try setting it there.

myfreeweb, 2020-10-04:
https://reviews.freebsd.org/D26657

evadot, 2020-10-05:
So this can be closed now that it's commited ?

myfreeweb, 2020-10-05:
Yep

closed 2020-10-05 by myfreeweb