freebsd/drm-kmod #33: vmwgfx panics after loading

xtouqh, 2020-10-01:
Windows 10, VMware Workstation Pro 16.0 (same for previous versions of hypervisor).  Running 8 "core" FreeBSD VM (UEFI boot): `FreeBSD 13.0-CURRENT #0 r366332`; `drm-devel-kmod-5.4.62.g20200914`.  The workaround of using just 1 "core" doesn't not help and is not something I would find useful, it defeats this VM usage.  Tried setting different amount of VRAM in the VM settings, no changes.

getting the following panic right after loading vmwgfx.ko:
```
Unread portion of the kernel message buffer:
<6>[drm] Fifo max 0x00040000 min 0x00001000 cap 0x0000077f
kernel trap 12 with interrupts disabled

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x18
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff80bb6814
stack pointer           = 0x28:0xfffffe004d79b3f0
frame pointer           = 0x28:0xfffffe004d79b3f0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = resume, IOPL = 0
current process         = 0 (deadlkres)
trap number             = 12
panic: page fault
cpuid = 0
time = 1601584668
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe004d79b0a0
vpanic() at vpanic+0x182/frame 0xfffffe004d79b0f0
panic() at panic+0x43/frame 0xfffffe004d79b150
trap_fatal() at trap_fatal+0x387/frame 0xfffffe004d79b1b0
trap_pfault() at trap_pfault+0x97/frame 0xfffffe004d79b210
trap() at trap+0x2ab/frame 0xfffffe004d79b320
calltrap() at calltrap+0x8/frame 0xfffffe004d79b320
--- trap 0xc, rip = 0xffffffff80bb6814, rsp = 0xfffffe004d79b3f0, rbp = 0xfffffe004d79b3f0 ---
thread_lock_validate() at thread_lock_validate+0x4/frame 0xfffffe004d79b3f0
_thread_lock() at _thread_lock+0x45/frame 0xfffffe004d79b430
deadlkres() at deadlkres+0x14e/frame 0xfffffe004d79b470
fork_exit() at fork_exit+0x80/frame 0xfffffe004d79b4b0
fork_trampoline() at fork_trampoline+0xe/frame 0xfffffe004d79b4b0
--- trap 0, rip = 0, rsp = 0, rbp = 0 ---
Uptime: 33s
Dumping 565 out of 8151 MB:..3%..12%..23%..32%..43%..51%..63%..71%..83%..91%
```

relevant dmesg entries:
```
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] DMA map mode: Caching DMA mappings.
[drm] Capabilities:
[drm]   Rect copy.
[drm]   Cursor.
[drm]   Cursor bypass.
[drm]   Cursor bypass 2.
[drm]   8bit emulation.
[drm]   Alpha cursor.
[drm]   3D.
[drm]   Extended Fifo.
[drm]   Multimon.
[drm]   Pitchlock.
[drm]   Irq mask.
[drm]   Display Topology.
[drm]   GMR.
[drm]   Traces.
[drm]   GMR2.
[drm]   Screen Object 2.
[drm]   Command Buffers.
[drm]   Command Buffers 2.
[drm]   Guest Backed Resources.
[drm]   DX Features.
[drm]   HP Command Queue.
[drm] Capabilities2:
[drm]   Grow oTable.
[drm]   IntraSurface copy.
[drm] Max GMR ids is 64
[drm] Max number of GMR pages is 65536
[drm] Max dedicated hypervisor surface memory is 0 kiB
[drm] Maximum display memory size is 262144 kiB
[drm] VRAM at 0xf0000000 size is 131072 kiB
[drm] MMIO at 0xfb800000 size is 256 kiB
[TTM] Zone  kernel: Available graphics memory: 4173806 KiB
[TTM] Zone   dma32: Available graphics memory: 2097152 KiB
[TTM] Initializing pool allocator
[drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[drm] No driver support for vblank timestamp query.
[drm] Connector Virtual-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-2
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-3: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-3
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-4: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-4
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-5: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-5
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-6: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-6
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-7: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-7
[drm]   - kern.vt.fb.default_mode
[drm] Connector Virtual-8: get mode from tunables:
[drm]   - kern.vt.fb.modes.Virtual-8
[drm]   - kern.vt.fb.default_mode
[drm] Screen Target Display device initialized
[drm] width 1024
[drm] height 768
[drm] bpp 32
[drm] Fifo max 0x00040000 min 0x00001000 cap 0x0000077f
kernel trap 12 with interrupts disabled
```

xtouqh, 2020-10-08:
Sorry, that was me, can provide any additional needed information.

tuaris, 2020-11-27:
For me, if I have `ttm_load="YES"` in `/boot/loader.conf` (and loaded after a reboot) it doesn't crash right away.  First the following error is displayed right after typing in `kldload vmwgfx.ko`:

```
Nov 27 11:02:19 freebsd kernel: drmn0: <drmn> on vgapci0
Nov 27 11:02:19 freebsd kernel: vgapci0: child drmn0 requested pci_enable_io
Nov 27 11:02:19 freebsd syslogd: last message repeated 1 times
Nov 27 11:02:19 freebsd kernel: [drm] DMA map mode: Using physical TTM page addresses.
Nov 27 11:02:19 freebsd kernel: [drm] Capabilities:
Nov 27 11:02:19 freebsd kernel: [drm]   Rect copy.
Nov 27 11:02:19 freebsd kernel: [drm]   Cursor.
Nov 27 11:02:19 freebsd kernel: [drm]   Cursor bypass.
Nov 27 11:02:19 freebsd kernel: [drm]   Cursor bypass 2.
Nov 27 11:02:19 freebsd kernel: [drm]   8bit emulation.
Nov 27 11:02:19 freebsd kernel: [drm]   Alpha cursor.
Nov 27 11:02:19 freebsd kernel: [drm]   3D.
Nov 27 11:02:19 freebsd kernel: [drm]   Extended Fifo.
Nov 27 11:02:19 freebsd kernel: [drm]   Multimon.
Nov 27 11:02:19 freebsd kernel: [drm]   Pitchlock.
Nov 27 11:02:19 freebsd kernel: [drm]   Irq mask.
Nov 27 11:02:19 freebsd kernel: [drm]   Display Topology.
Nov 27 11:02:19 freebsd kernel: [drm]   GMR.
Nov 27 11:02:19 freebsd kernel: [drm]   Traces.
Nov 27 11:02:19 freebsd kernel: [drm]   GMR2.
Nov 27 11:02:19 freebsd kernel: [drm]   Screen Object 2.
Nov 27 11:02:19 freebsd kernel: [drm]   Command Buffers.
Nov 27 11:02:19 freebsd kernel: [drm]   Command Buffers 2.
Nov 27 11:02:19 freebsd kernel: [drm]   Guest Backed Resources.
Nov 27 11:02:19 freebsd kernel: [drm]   DX Features.
Nov 27 11:02:19 freebsd kernel: [drm] Max GMR ids is 64
Nov 27 11:02:19 freebsd kernel: [drm] Max number of GMR pages is 65536
Nov 27 11:02:19 freebsd kernel: [drm] Max dedicated hypervisor surface memory is 0 kiB
Nov 27 11:02:19 freebsd kernel: [drm] Maximum display memory size is 262144 kiB
Nov 27 11:02:19 freebsd kernel: [drm] VRAM at 0xe8000000 size is 131072 kiB
Nov 27 11:02:19 freebsd kernel: [drm] MMIO at 0xfe000000 size is 256 kiB
Nov 27 11:02:19 freebsd kernel: [drm] global init.
Nov 27 11:02:19 freebsd kernel: [drm:vmw_ttm_global_init] Failed setting up TTM memory accounting.
Nov 27 11:02:19 freebsd kernel: device_attach: drmn0 attach returned 12
```

A few seconds later, the system will panic:

```
Nov 27 11:04:22 freebsd syslogd: kernel boot file is /boot/kernel/kernel
Nov 27 11:04:22 freebsd kernel:
Nov 27 11:04:22 freebsd syslogd: last message repeated 1 times
Nov 27 11:04:22 freebsd kernel: Fatal trap 9: general protection fault while in kernel mode
Nov 27 11:04:22 freebsd kernel: cpuid = 0; apic id = 00
Nov 27 11:04:22 freebsd kernel: instruction pointer     = 0x20:0xffffffff80b89ef0
Nov 27 11:04:22 freebsd kernel: stack pointer           = 0x28:0xfffffe0025d9a7c0
Nov 27 11:04:22 freebsd kernel: frame pointer           = 0x28:0xfffffe0025d9a7c0
Nov 27 11:04:22 freebsd kernel: code segment            = base 0x0, limit 0xfffff, type 0x1b
Nov 27 11:04:22 freebsd kernel:                         = DPL 0, pres 1, long 1, def32 0, gran 1
Nov 27 11:04:22 freebsd kernel: processor eflags        = interrupt enabled, resume, IOPL = 0
Nov 27 11:04:22 freebsd kernel: current process         = 999 (ck-collect-session-)
Nov 27 11:04:22 freebsd kernel: trap number             = 9
Nov 27 11:04:22 freebsd kernel: panic: general protection fault
Nov 27 11:04:22 freebsd kernel: cpuid = 0
Nov 27 11:04:22 freebsd kernel: time = 1606474972
Nov 27 11:04:22 freebsd kernel: KDB: stack backtrace:
Nov 27 11:04:22 freebsd kernel: #0 0xffffffff80c0a8f5 at kdb_backtrace+0x65
Nov 27 11:04:22 freebsd kernel: #1 0xffffffff80bbeb1b at vpanic+0x17b
Nov 27 11:04:22 freebsd kernel: #2 0xffffffff80bbe993 at panic+0x43
Nov 27 11:04:22 freebsd kernel: #3 0xffffffff8108f911 at trap_fatal+0x391
Nov 27 11:04:22 freebsd kernel: #4 0xffffffff8108ed97 at trap+0x67
Nov 27 11:04:22 freebsd kernel: #5 0xffffffff81066f38 at calltrap+0x8
Nov 27 11:04:22 freebsd kernel: #6 0xffffffff80bae869 at cr_cansee+0x19
Nov 27 11:04:22 freebsd kernel: #7 0xffffffff80ba3f71 at pget+0x101
Nov 27 11:04:22 freebsd kernel: #8 0xffffffff80ba80dd at sysctl_kern_proc+0x8d
Nov 27 11:04:22 freebsd kernel: #9 0xffffffff80bcda3a at sysctl_root_handler_locked+0x8a
Nov 27 11:04:22 freebsd kernel: #10 0xffffffff80bcce69 at sysctl_root+0x249
Nov 27 11:04:22 freebsd kernel: #11 0xffffffff80bcd4d8 at userland_sysctl+0x178
Nov 27 11:04:22 freebsd kernel: #12 0xffffffff80bcd31f at sys___sysctl+0x5f
Nov 27 11:04:22 freebsd kernel: #13 0xffffffff810904c7 at amd64_syscall+0x387
Nov 27 11:04:22 freebsd kernel: #14 0xffffffff8106785e at fast_syscall_common+0xf8
Nov 27 11:04:22 freebsd kernel: Uptime: 1m9s
```

I uploaded crash dump to http://venus.morante.net/downloads/unibia/freebsd/misc/var_crash.tar.xz
It's a throwaway VM with no important data or private info.

HenryHu, 2021-02-04:
I've got the panic too, with VMWare Player 16.0.0 build-16894299 and 12.2-RELEASE.

tuaris, 2021-02-04:
I recently tried this on 12.2-RELEASE amd64.  The way to prevent the panic is to use UEFI mode.  However, there is a bug (https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=250580) with that too, but when you work around the bug, there is no panic.    I just haven't been able to get Xorg to use the 3D driver.

HenryHu, 2021-02-04:
Well, I worked around the UEFI boot issue by removing nvram, but loading the module still crashed the machine (I don't know if it's a panic, since the screen went blank and then rebooted - or not, sometimes). I think it panic, though. I wonder if it's supposed to work, or it's just a byproduct of compiling other kernel modules.

HenryHu, 2021-02-04:
With serial console, I can see different kinds of panics triggered when loading vmwgfx (with UEFI boot):
example:

panic: free: address 0xfffff800bbf45d00(0xfffff800bbf45000) has not been allocated.

cpuid = 0
time = 1612413099
KDB: stack backtrace:
kernel trap 12 with interrupts disabled

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x18
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff806ca653
stack pointer           = 0x28:0xfffffe004d5bffe0
frame pointer           = 0x28:0xfffffe004d5c0030
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = resume, IOPL = 0
current process         = 930 (kldload)
trap number             = 12
panic: page fault
cpuid = 0
time = 1612413099
KDB: stack backtrace:
kernel trap 12 with interrupts disabled

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x18
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff806ca653
stack pointer           = 0x28:0xfffffe004d5bfab0
frame pointer           = 0x28:0xfffffe004d5bfb00
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = resume, IOPL = 0
current process         = 930 (kldload)
trap number             = 12
panic: page fault
cpuid = 0
time = 1612413099
KDB: stack backtrace:
kernel trap 12 with interrupts disabled

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x18
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff806ca653
stack pointer           = 0x28:0xfffffe004d5bf580
frame pointer           = 0x28:0xfffffe004d5bf5d0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = resume, IOPL = 0
current process         = 930 (kldload)
trap number             = 12
panic: page fault
cpuid = 0
time = 1612413099
KDB: stack backtrace:
kernel trap 12 with interrupts disabled

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address   = 0x18
fault code              = supervisor read data, page not present
instruction pointer     = 0x20:0xffffffff806ca653
stack pointer           = 0x28:0xfffffe004d5bf050
frame pointer           = 0x28:0xfffffe004d5bf0a0
code segment            = base 0x0, limit 0xfffff, type 0x1b
                        = DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags        = resume, IOPL = 0
current process         = 930 (kldload)
trap number             = 12
......

Seems like that it can't even print the stack trace properly.

tuaris, 2021-02-18:
It seems that it's a known issue that the `vmwgfx` module doesn't work.  It was removed in ff9d303c7ea85cd8627d0a3dc0dbccceefd30687

Hopefully it's temporary.

agrajag9, 2022-01-06:
@evadot can we 

> We need people to maintain vmwgfx, I don't have the time or the will to port them again. Closing this issue.

Since you don't have the time, perhaps we should make this a feature request so that, if someone with the time and know-how finds it, they can pick it up?

ayeedshaikh, 2022-05-22:
https://github.com/ravynsoft/drm-kmod/tree/vmware-5.10.

The driver works fine in BIOS mode, We are facing an issue in UEFI mode, the driver loads without panic but doesn't draw content on the screen. Any suggestion would be appreciable.