freebsd/drm-kmod #36: Renoir: panic when starting X with the amdgpu(4x) Xorg driver

jiixyj, 2020-10-22:
I'm trying to start X on my Thinkpad T14 Gen1 AMD (Ryzen 7 Pro 4750U), but the system crashes to DDB.

I'm using latest -CURRENT and ports. The amdgpu kernel module from the 5.4.62.g20201003 drm-current-kmod port seems to load successfully when I put `hw.amdgpu.exp_hw_support=1` into `/boot/loader.conf`.

It only crashes when I use the `amdgpu` driver for X:
```
Section "Device"
        Identifier "Card0"
        Driver "amdgpu"
EndSection
```

With `scfb` X start successfully.

If I'm reading the stacktrace correctly, `page` is `NULL` at this line here: https://github.com/freebsd/drm-kmod/blob/f142faa7af1496b38e728e6acfcc3ab32f716b70/drivers/gpu/drm/ttm/ttm_bo_vm.c#L322

part of core.txt.0:
```
Thu Oct 22 22:53:13 CEST 2020

FreeBSD 13.0-CURRENT FreeBSD 13.0-CURRENT #0 r366953+44cf9eb23ade-c272379(master): Thu Oct 22 22:41:23 CEST 2020     root@toluca:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64

panic: page fault

GNU gdb 6.1.1 [FreeBSD]
Copyright 2004 Free Software Foundation, Inc.
GDB is free software, covered by the GNU General Public License, and you are
welcome to change it and/or distribute copies of it under certain conditions.
Type "show copying" to see the conditions.
There is absolutely no warranty for GDB.  Type "show warranty" for details.
This GDB was configured as "amd64-marcel-freebsd"...

Unread portion of the kernel message buffer:
<4>__pm_runtime_resume not implemented -- see your local kernel hacker
<4>pm_runtime_mark_last_busy not implemented -- see your local kernel hacker
<4>__pm_runtime_suspend not implemented -- see your local kernel hacker
<4>__pm_runtime_resume not implemented -- see your local kernel hacker
<4>pm_runtime_mark_last_busy not implemented -- see your local kernel hacker
<4>__pm_runtime_suspend not implemented -- see your local kernel hacker

Fatal trap 12: page fault while in kernel mode
cpuid = 0; apic id = 00
fault virtual address	= 0x20
fault code		= supervisor read data, page not present
instruction pointer	= 0x20:0xffffffff80eda064
stack pointer	        = 0x0:0xfffffe00e823b750
frame pointer	        = 0x0:0xfffffe00e823b780
code segment		= base 0x0, limit 0xfffff, type 0x1b
			= DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags	= interrupt enabled, resume, IOPL = 0
current process		= 1756 (Xorg)
trap number		= 12
panic: page fault
cpuid = 0
time = 1603399896
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe00e823b400
vpanic() at vpanic+0x182/frame 0xfffffe00e823b450
panic() at panic+0x43/frame 0xfffffe00e823b4b0
trap_fatal() at trap_fatal+0x387/frame 0xfffffe00e823b510
trap_pfault() at trap_pfault+0x4f/frame 0xfffffe00e823b570
trap() at trap+0x27d/frame 0xfffffe00e823b680
calltrap() at calltrap+0x8/frame 0xfffffe00e823b680
--- trap 0xc, rip = 0xffffffff80eda064, rsp = 0xfffffe00e823b750, rbp = 0xfffffe00e823b780 ---
vm_page_busy_acquire() at vm_page_busy_acquire+0x14/frame 0xfffffe00e823b780
ttm_bo_vm_fault() at ttm_bo_vm_fault+0x2de/frame 0xfffffe00e823b810
linux_cdev_pager_populate() at linux_cdev_pager_populate+0x11b/frame 0xfffffe00e823b880
vm_fault_allocate() at vm_fault_allocate+0x2a2/frame 0xfffffe00e823b8f0
vm_fault() at vm_fault+0x418/frame 0xfffffe00e823b9f0
vm_fault_trap() at vm_fault_trap+0x6d/frame 0xfffffe00e823ba40
[... 128 lines cut ...]
)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:557
#18 0xffffffff80ec3bd2 in vm_fault_allocate (fs=0xfffffe00e823b950)
    at src/sys/vm/vm_pager.h:171
#19 0xffffffff80ec2948 in vm_fault (map=<value optimized out>, 
    vaddr=34573647872, fault_type=2 '\002', fault_flags=<value optimized out>, 
    m_hold=<value optimized out>) at /usr/src/sys/vm/vm_fault.c:1404
#20 0xffffffff80ec240d in vm_fault_trap (map=0xfffffe00e789a000, 
    vaddr=<value optimized out>, fault_type=<value optimized out>, 
    fault_flags=0, signo=0xfffffe00e823bb84, ucode=0xfffffe00e823bb80)
    at /usr/src/sys/vm/vm_fault.c:631
#21 0xffffffff81024918 in trap_pfault (frame=0xfffffe00e823bbc0, 
    signo=<value optimized out>, ucode=<value optimized out>)
    at /usr/src/sys/amd64/amd64/trap.c:817
#22 0xffffffff81023f4d in trap (frame=0xfffffe00e823bbc0)
    at /usr/src/sys/amd64/amd64/trap.c:340
#23 0xffffffff80ff8628 in calltrap ()
    at /usr/src/sys/amd64/amd64/exception.S:290
#24 0x0000000800a985cb in ?? ()
Previous frame inner to this frame (corrupt stack?)
Current language:  auto; currently minimal
(kgdb) 
```

I could provide the full core.txt.0 and the vmcore if needed.

monwarez, 2020-11-20:
Does it crash with `modesetting` driver?

jiixyj, 2020-11-20:
It doesn't crash with `modesetting`, but it doesn't start up, either:
```
[   391.228] 
X.Org X Server 1.20.9
X Protocol Version 11, Revision 0
[   391.228] Build Operating System: FreeBSD 13.0-CURRENT amd64 
[   391.228] Current Operating System: FreeBSD host 13.0-CURRENT FreeBSD 13.0-CURRENT #0 e13becc87-c1(master): Fri Nov 20 22:08:01 CET 2020     root@host:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64
[   391.229] Build Date: 17 November 2020  05:31:06AM
[   391.229]  
[   391.229] Current version of pixman: 0.40.0
[   391.229] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   391.229] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   391.229] (==) Log file: "/var/log/Xorg.1.log", Time: Fri Nov 20 22:39:39 2020
[   391.229] (==) Using config directory: "/usr/local/etc/X11/xorg.conf.d"
[   391.229] (==) Using system config directory "/usr/local/share/X11/xorg.conf.d"
[   391.230] (==) No Layout section.  Using the first Screen section.
[   391.230] (==) No screen section available. Using defaults.
[   391.230] (**) |-->Screen "Default Screen Section" (0)
[   391.230] (**) |   |-->Monitor "<default monitor>"
[   391.230] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[   391.230] (**) |   |-->Device "Card0"
[   391.230] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   391.230] (==) Automatically adding devices
[   391.230] (==) Automatically enabling devices
[   391.230] (==) Not automatically adding GPU devices
[   391.230] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   391.230] (WW) The directory "/usr/local/share/fonts/misc/" does not exist.
[   391.230] 	Entry deleted from font path.
[   391.230] (WW) The directory "/usr/local/share/fonts/Type1/" does not exist.
[   391.230] 	Entry deleted from font path.
[   391.230] (WW) The directory "/usr/local/share/fonts/100dpi/" does not exist.
[   391.230] 	Entry deleted from font path.
[   391.230] (WW) The directory "/usr/local/share/fonts/75dpi/" does not exist.
[   391.230] 	Entry deleted from font path.
[   391.231] (==) FontPath set to:
	/usr/local/share/fonts/TTF/,
	/usr/local/share/fonts/OTF/,
	catalogue:/usr/local/etc/X11/fontpath.d
[   391.231] (==) ModulePath set to "/usr/local/lib/xorg/modules"
[   391.231] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   391.231] (II) Loader magic: 0x435470
[   391.231] (II) Module ABI versions:
[   391.231] 	X.Org ANSI C Emulation: 0.4
[   391.231] 	X.Org Video Driver: 24.1
[   391.231] 	X.Org XInput driver : 24.1
[   391.231] 	X.Org Server Extension : 10.0
[   391.231] (--) PCI:*(7@0:0:0) 1002:1636:17aa:5081 rev 209, Mem @ 0x860000000/268435456, 0x870000000/2097152, 0xfd300000/524288, I/O @ 0x00001000/256, BIOS @ 0x????????/65536
[   391.231] (II) LoadModule: "glx"
[   391.231] (II) Loading /usr/local/lib/xorg/modules/extensions/libglx.so
[   391.233] (II) Module glx: vendor="X.Org Foundation"
[   391.233] 	compiled for 1.20.9, module version = 1.0.0
[   391.233] 	ABI class: X.Org Server Extension, version 10.0
[   391.233] (II) LoadModule: "modesetting"
[   391.233] (II) Loading /usr/local/lib/xorg/modules/drivers/modesetting_drv.so
[   391.233] (II) Module modesetting: vendor="X.Org Foundation"
[   391.233] 	compiled for 1.20.9, module version = 1.20.9
[   391.233] 	Module class: X.Org Video Driver
[   391.233] 	ABI class: X.Org Video Driver, version 24.1
[   391.233] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   391.233] (--) Using syscons driver with X support (version 2.0)
[   391.233] (--) using VT number 9

[   391.235] (WW) Falling back to old probe method for modesetting
[   391.235] (WW) VGA arbiter: cannot open kernel arbiter, no multi-card support
[... 188 lines cut ...]
    subclass   = HDA
none7@pci0:7:0:2:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15df subvendor=0x17aa subdevice=0x5081
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) Platform Security Processor'
    class      = encrypt/decrypt
xhci1@pci0:7:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1639 subvendor=0x17aa subdevice=0x5081
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir USB 3.1'
    class      = serial bus
    subclass   = USB
xhci2@pci0:7:0:4:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1639 subvendor=0x17aa subdevice=0x5081
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir USB 3.1'
    class      = serial bus
    subclass   = USB
none8@pci0:7:0:5:	class=0x048000 rev=0x01 hdr=0x00 vendor=0x1022 device=0x15e2 subvendor=0x17aa subdevice=0x5081
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2/FireFlight/Renoir Audio Processor'
    class      = multimedia
hdac1@pci0:7:0:6:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x17aa subdevice=0x5081
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) HD Audio Controller'
    class      = multimedia
    subclass   = HDA
```

monwarez, 2020-11-20:
Just to be sure, without having any xorg.conf files does it fallback to sfcb or succeed with modessetting ?
Also you could try to build a more recent one: 5.4.62.g20201109 (or at least make sure that the kmod was build with the exact same revision as your kernel).

jiixyj, 2020-11-20:
Thank you for the hints! I forgot to mention in my previous post that I updated -CURRENT to the most recent commit (https://github.com/freebsd/freebsd/commit/e13becc870074c3fc5f40efe99b879a0390d9581) and rebuilt 5.4.62.g20201109 so there shouldn't be any weird KPI issues.

I'll try with the 5.5-wip ASAP. I've assumed that the Mesa 20.2.0 in ports (https://www.freshports.org/graphics/mesa-dri/) contains the required bits for Renoir, but I could be wrong of course.

I'll check again what happens if I leave out any Xorg config files, but I _think_ it did a fallback to the `modesetting` driver with the above error ("no screens found(EE)").

I've also tried to start Wayland with `sway` and it ran into the same crash in my first post.

jiixyj, 2020-11-20:
I also haven't tried to insert a `if (page == NULL) { goto fail; }` yet, maybe that makes it go on a bit longer.

jiixyj, 2020-11-21:
OK, when I do the `if (page == NULL) { goto fail; }` in `ttm_bo_vm.c` I get this segfault from Xorg. I rebuilt Mesa 20.2.0 from ports with `make WITH_DEBUG=1`.

```
#0  0x0000000800a78b0a in thr_kill () from /lib/libc.so.7
#1  0x00000008009ee784 in raise () from /lib/libc.so.7
#2  0x0000000800aa20d9 in abort () from /lib/libc.so.7
#3  0x000000000042004a in OsAbort ()
#4  0x000000000042710c in ?? ()
#5  0x000000000042581a in FatalError ()
#6  0x000000000041d8b5 in ?? ()
#7  0x0000000800927b9e in ?? () from /lib/libthr.so.3
#8  0x000000080092715f in ?? () from /lib/libthr.so.3
#9  <signal handler called>
#10 0x0000000800a985cb in memcpy () from /lib/libc.so.7
#11 0x00000008030b4c1e in si_buffer_subdata (ctx=0x8096405c0, buffer=0x8096c36c0, usage=258, offset=0, size=248, data=0x8096472e4)
    at ../src/gallium/drivers/radeonsi/si_buffer.c:622
#12 0x00000008030a3a9f in pipe_buffer_write (pipe=0x8096405c0, buf=0x8096c36c0, offset=0, size=248, data=0x8096472e4)
    at ../src/gallium/auxiliary/util/u_inlines.h:408
#13 0x00000008030a20a4 in si_create_context (screen=0x8013cc000, flags=0) at ../src/gallium/drivers/radeonsi/si_pipe.c:677
#14 0x00000008030a09ec in radeonsi_screen_create_impl (ws=0x8012f61c0, config=0x7fffffffcc30) at ../src/gallium/drivers/radeonsi/si_pipe.c:1268
#15 0x000000080319259c in amdgpu_winsys_create (fd=15, config=0x7fffffffcc30, screen_create=0x80309f650 <radeonsi_screen_create_impl>)
    at ../src/gallium/winsys/amdgpu/drm/amdgpu_winsys.c:538
#16 0x000000080309f60c in radeonsi_screen_create (fd=15, config=0x7fffffffcc30) at ../src/gallium/drivers/radeonsi/si_pipe.c:1315
#17 0x00000008022cf84b in pipe_radeonsi_create_screen (fd=15, config=0x7fffffffcc30) at ../src/gallium/auxiliary/target-helpers/drm_helper.h:169
#18 0x0000000802d3a371 in pipe_loader_drm_create_screen (dev=0x801210560, config=0x7fffffffcc30) at ../src/gallium/auxiliary/pipe-loader/pipe_loader_drm.c:324
#19 0x0000000802d39b46 in pipe_loader_create_screen (dev=0x801210560) at ../src/gallium/auxiliary/pipe-loader/pipe_loader.c:128
#20 0x00000008022d7020 in dri2_init_screen (sPriv=0x801315580) at ../src/gallium/frontends/dri/dri2.c:2075
#21 0x0000000802bc5dab in driCreateNewScreen2 (scrn=0, fd=13, extensions=0x801b75410, driver_extensions=0x803c03c60 <galliumdrm_driver_extensions>, 
    driver_configs=0x8012fd478, data=0x8012fd280) at ../src/mesa/drivers/dri/common/dri_util.c:155
#22 0x0000000801b6d638 in ?? () from /usr/local/lib/libgbm.so.1
#23 0x0000000801b6be15 in ?? () from /usr/local/lib/libgbm.so.1
#24 0x0000000801b6b5bc in gbm_create_device () from /usr/local/lib/libgbm.so.1
#25 0x0000000801b42200 in ?? () from /usr/local/lib/xorg/modules/drivers/amdgpu_drv.so
#26 0x00000000002e5fe4 in InitOutput ()
#27 0x00000000002a17db in ?? ()
#28 0x000000000028a300 in _start ()
quit
```

The `5.5-wip` branch errors out when doing `kldload amdgpu`. There is something wrong with `ttm_bo_mmap_obj`:
```
root@host:~ # kldload amdgpu
link_elf_obj: symbol ttm_bo_mmap_obj undefined
Warning: memory type debugfsint leaked memory on destroy (2 allocations, 80 bytes leaked).                                                                      
linker_load_file: /boot/modules/drm.ko - unsupported file type                                                                                                  
KLD amdgpu.ko: depends on drmn - not available or version mismatch                                                                                              
linker_load_file: /boot/modules/amdgpu.ko - unsupported file type                                                                                               
kldload: an error occurred while loading module amdgpu. Please check dmesg(8) for more details.                                                                 
root@host:~ #
```

monwarez, 2020-11-21:
for the `ttm_bo_vm.c`, if we check at the context before:
```
if (page == NULL) {
			if (bo->mem.bus.is_iomem) {
				pfn = ttm_bo_io_mem_pfn(bo, page_offset);
				page = PHYS_TO_VM_PAGE(IDX_TO_OFF(pfn));
			} else {
				page = ttm->pages[page_offset];
				if (page == NULL)
					goto fail;
				page->oflags &= ~VPO_UNMANAGED;
			}
```
The only path that would lead to `page == NULL` without fail is
```
if (bo->mem.bus.is_iomem) {
				pfn = ttm_bo_io_mem_pfn(bo, page_offset);
				page = PHYS_TO_VM_PAGE(IDX_TO_OFF(pfn));
			}
```

jiixyj, 2020-11-21:
Exactly, I inserted the null check like this:

```
			if (bo->mem.bus.is_iomem) {
				pfn = ttm_bo_io_mem_pfn(bo, page_offset);
				page = PHYS_TO_VM_PAGE(IDX_TO_OFF(pfn));
				if (page == NULL)
					goto fail;
			} else {
				page = ttm->pages[page_offset];
				if (page == NULL)
					goto fail;
				page->oflags &= ~VPO_UNMANAGED;
			}
```

Sadly this resulted in the Mesa segfault above, so there must be something trickier going on...

monwarez, 2020-11-21:
Did you check `pfn` ?

monwarez, 2020-11-21:
At line https://github.com/freebsd/drm-kmod/blob/f142faa7af1496b38e728e6acfcc3ab32f716b70/drivers/gpu/drm/ttm/ttm_bo_vm.c#L102
You could try to not use the bo_device, but not sure if it is a good idea. (you could also check which path is taken)
```
static unsigned long ttm_bo_io_mem_pfn(struct ttm_buffer_object *bo,
				       unsigned long page_offset)
{
	struct ttm_bo_device *bdev = bo->bdev;

	//if (bdev->driver->io_mem_pfn)
		//return bdev->driver->io_mem_pfn(bo, page_offset);

	return ((bo->mem.bus.base + bo->mem.bus.offset) >> PAGE_SHIFT)
		+ page_offset;
}
```

jiixyj, 2020-11-21:
I think I got a bit further: I noticed that `PHYS_TO_VM_PAGE` also looks at "fictitious" pages if it cannot find a `vm_page_t` for an address. In the past, there was this piece of code that registered some device memory (?) as fictitious pages: https://github.com/FreeBSDDesktop/kms-drm/pull/108/files

I put it back like this:

```patch
diff --git a/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c b/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c
index b4378fcff..c475ad236 100644
--- a/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c
+++ b/drivers/gpu/drm/amd/amdgpu/amdgpu_device.c
@@ -2864,6 +2864,26 @@ fence_driver_init:
 
 	amdgpu_fbdev_init(adev);
 
+#ifndef __linux__
+	r = vm_phys_fictitious_reg_range(adev->gmc.aper_base,
+	    adev->gmc.aper_base + adev->gmc.visible_vram_size,
+	    VM_MEMATTR_WRITE_COMBINING);
+
+	if (r != 0) {
+		DRM_ERROR(
+		    "failed to register fictitious range 0x%jx-0x%jx (%d).\n",
+		    (uintmax_t)adev->gmc.aper_base,
+		    (uintmax_t)adev->gmc.aper_base + adev->gmc.visible_vram_size,
+		    r);
+
+		return -r;
+	}
+
+	DRM_INFO("registered fictitious range 0x%jx-0x%jx\n",
+		(uintmax_t)adev->gmc.aper_base,
+		(uintmax_t)adev->gmc.aper_base + adev->gmc.visible_vram_size);
+#endif
+
 	if (amdgpu_sriov_vf(adev) && amdgim_is_hwperf(adev))
 		amdgpu_pm_virt_sysfs_init(adev);
 ```

...and now the X startup with `amdgpu` goes on a bit further, before stopping because no outputs are found:

```
[   714.238] 
X.Org X Server 1.20.9
X Protocol Version 11, Revision 0
[   714.238] Build Operating System: FreeBSD 13.0-CURRENT amd64 
[   714.238] Current Operating System: FreeBSD host 13.0-CURRENT FreeBSD 13.0-CURRENT #0 e13becc87-c1(master): Fri Nov 20 22:08:01 CET 2020     root@host:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG amd64
[   714.238] Build Date: 17 November 2020  05:31:06AM
[   714.238]  
[   714.238] Current version of pixman: 0.40.0
[   714.238] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   714.238] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   714.239] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Nov 21 19:03:09 2020
[   714.239] (==) Using config directory: "/usr/local/etc/X11/xorg.conf.d"
[   714.239] (==) Using system config directory "/usr/local/share/X11/xorg.conf.d"
[   714.239] (==) No Layout section.  Using the first Screen section.
[   714.239] (==) No screen section available. Using defaults.
[   714.239] (**) |-->Screen "Default Screen Section" (0)
[   714.239] (**) |   |-->Monitor "<default monitor>"
[   714.240] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[   714.240] (**) |   |-->Device "Card0"
[   714.240] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   714.240] (==) Automatically adding devices
[   714.240] (==) Automatically enabling devices
[   714.240] (==) Not automatically adding GPU devices
[   714.240] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   714.240] (WW) The directory "/usr/local/share/fonts/misc/" does not exist.
[... 70 lines cut ...]
[   714.387] (==) AMDGPU(0): DPI set to (96, 96)
[   714.387] (==) AMDGPU(0): Using gamma correction (1.0, 1.0, 1.0)
[   714.387] (II) Loading sub module "ramdac"
[   714.387] (II) LoadModule: "ramdac"
[   714.387] (II) Module "ramdac" already built-in
[   714.387] (EE) AMDGPU(0): No modes.
[   714.388] (II) UnloadModule: "amdgpu"
[   714.388] (II) UnloadSubModule: "glamoregl"
[   714.388] (II) Unloading glamoregl
[   714.388] (II) UnloadSubModule: "fb"
[   714.388] (II) Unloading fb
[   714.388] (EE) Screen(s) found, but none have a usable configuration.
[   714.388] (EE) 
Fatal server error:
[   714.388] (EE) no screens found(EE) 
[   714.388] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   714.388] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   714.388] (EE) 
[   714.390] (EE) Server terminated with error (1). Closing log file.
```

The `modesetting` driver still fails with "no screens found(EE)".

myfreeweb, 2020-11-22:
5.5 fix: https://github.com/freebsd/drm-kmod/pull/40

jiixyj, 2021-01-09:
Fantastic, using the `myfreeweb:5.5-wip-amd-pr` branch Xorg starts up using the amdgpu DDX driver. Backlight controls using `backlight(8)` work and 3D acceleration using Mesa seems to work as well (tested with glxgears).

evadot, 2021-04-19:
Closing as it works with 5.5-wip

closed 2021-04-19 by evadot