freebsd/drm-kmod #38: drm-devel-kmod fails to build on powerpc64(le)

pkubaj, 2020-11-11:
```
===> amd/amdgpu (all)
Warning: Object directory not changed from original /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/amd/amdgpu
cc  -O2 -pipe -mcpu=power9 -fno-strict-aliasing '-DKBUILD_MODNAME="amdgpu"' '-DLINUXKPI_PARAM_PREFIX=amdgpu_' -DDRM_SYSCTL_PARAM_PREFIX=_amdgpu -DLINUXKPI_VERSION=50000 -DCONFIG_DRM_AMDGPU_CIK -DCONFIG_DRM_AMDGPU_SI -DCONFIG_DRM_AMD_DC -DCONFIG_DRM_AMD_DC_FBC -DCONFIG_DRM_AMD_POWERPLAY -DCONFIG_DRM_I915_ALPHA_SUPPORT -DCONFIG_DRM_I915_FORCE_PROBE='"*"' -DCONFIG_DRM_I915_CAPTURE_ERROR -DCONFIG_DRM_I915_SPIN_REQUEST=5 -DCONFIG_DRM_I915_USERFAULT_AUTOSUSPEND=250 -DCONFIG_DRM_LOAD_EDID_FIRMWARE -DCONFIG_DRM_MIPI_DSI -DCONFIG_DRM_PANEL_ORIENTATION_QUIRKS -DCONFIG_DRM_VMWGFX_FBCON -DCONFIG_DRM_FBDEV_EMULATION -DCONFIG_DRM_FBDEV_OVERALLOC=100 -DCONFIG_DRM_LEGACY -DCONFIG_DRM_VM -DCONFIG_ARCH_HAVE_NMI_SAFE_CMPXCHG -DCONFIG_BACKLIGHT_CLASS_DEVICE -DCONFIG_DMI -DCONFIG_FB -DCONFIG_MTRR -DCONFIG_PCI -DCONFIG_PM -DCONFIG_SMP -mcpu=power9  -Werror -D_KERNEL -DKLD_MODULE -nostdinc  -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/linuxkpi/gplv2/include -I/usr/src/sys/compat/linuxkpi/common/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/linuxkpi/dummy/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/include/drm -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/include/uapi -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/acp/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/amdgpu -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/amdgpu_dm -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/dc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/dc/clk_mgr/ -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/dc/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/dc/inc/hw -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/display/modules/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/include -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/include/asic_reg -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/powerplay/hwmgr -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/powerplay/inc -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/powerplay/smumgr -I/tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/scheduler -include /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/amd/amdgpu/opt_global.h -I. -I/usr/src/sys -I/usr/src/sys/contrib/ck/include -fno-common  -fPIC -mlongcall -fno-omit-frame-pointer -fdebug-prefix-map=./machine=/usr/src/sys/powerpc/include     -MD  -MF.depend.amdgpu_bios.o -MTamdgpu_bios.o -mno-altivec -msoft-float -mabi=elfv2 -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error-tautological-compare -Wno-error-empty-body -Wno-error-parentheses-equality -Wno-error-unused-function -Wno-error-pointer-sign -Wno-error-shift-negative-value -Wno-address-of-packed-member -Wno-format-zero-length -Wno-pointer-arith -Wno-format -Wno-cast-qual -Wno-pointer-sign     -std=iso9899:1999 -c /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/amdgpu/amdgpu_bios.c -o amdgpu_bios.o
In file included from /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/drivers/gpu/drm/amd/amdgpu/amdgpu_bios.c:34:
In file included from /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/linuxkpi/gplv2/include/linux/acpi.h:56:
In file included from /usr/src/sys/compat/linuxkpi/common/include/linux/acpi.h:36:
In file included from /tmp/usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.4.62_4/linuxkpi/gplv2/include/acpi/acpi.h:24:
In file included from /usr/src/sys/compat/linuxkpi/common/include/acpi/acpi.h:48:
In file included from /usr/src/sys/contrib/dev/acpica/include/acpi.h:164:
In file included from /usr/src/sys/contrib/dev/acpica/include/platform/acenv.h:318:
/usr/src/sys/contrib/dev/acpica/include/platform/acfreebsd.h:181:10: fatal error: 'machine/acpica_machdep.h' file not found
#include <machine/acpica_machdep.h>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

The reason is that `acpica_machdep.h` is only for aarch64, amd64 and i386:
```
pkubaj@talos:$/usr/ports/graphics/drm-devel-kmod$ find /usr/src/ -name acpica_machdep.h
/usr/src/sys/arm64/include/acpica_machdep.h
/usr/src/sys/x86/include/acpica_machdep.h
/usr/src/sys/i386/include/acpica_machdep.h
/usr/src/sys/amd64/include/acpica_machdep.h
```

myfreeweb, 2020-11-11:
hmmm, the `#include <linux/acpi.h>` in `amdgpu_bios` is not guarded by `CONFIG_ACPI`. I guess upstream it relies on [the guard inside the file](https://github.com/torvalds/linux/blob/eccc876724927ff3b9ff91f36f7b6b159e948f0c/include/linux/acpi.h#L24).

@wulf7  our [linux/acpi.h](https://github.com/freebsd/freebsd/blob/master/sys/compat/linuxkpi/common/include/linux/acpi.h) should be `#if defined(__i386__) || defined(__amd64__) || defined(__aarch64__)` inside..

wulf7, 2020-11-11:
> The reason is that acpica_machdep.h is only for aarch64, amd64 and i386

@pkubaj Is drm-kmod can be compiled again if you truncate (or delete ACPI-related content) from `sys/compat/linuxkpi/common/include/linux/acpi.h` in base? Or some more headers require guard #ifdef-s?

pkubaj, 2020-11-12:
Yes, it's enough to disable `acpi/acpi.h` and `acpi/acpi_bus.h`.

pkubaj, 2020-11-14:
@evadot 
Could you then guard `acpi/acpi.h` and `acpi/acpi_bus.h` includes in `sys/compat/linuxkpi/common/include/linux/acpi.h` with `#if defined(__i386__) || defined(__amd64__) || defined(__aarch64__)`?

wulf7, 2020-11-14:
> guard `acpi/acpi.h` and `acpi/acpi_bus.h` includes

Done in r367673.

closed 2020-11-18 by pkubaj