freebsd/drm-kmod #41: Panic running "sysctl sys" with Radeon Pro W5700

bcran, 2020-11-27:
When I run "sysctl sys" on FreeBSD 13-CURRENT, I get a panic in an amdgpu sysctl handler.

```
__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
55              __asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
(kgdb) #0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  doadump (textdump=1) at /usr/src/sys/kern/kern_shutdown.c:399
#2  0xffffffff80bf677b in kern_reboot (howto=260)
    at /usr/src/sys/kern/kern_shutdown.c:486
#3  0xffffffff80bf6c00 in vpanic (fmt=<optimized out>, ap=<optimized out>)
    at /usr/src/sys/kern/kern_shutdown.c:919
#4  0xffffffff80bf6a03 in panic (fmt=<unavailable>)
    at /usr/src/sys/kern/kern_shutdown.c:843
#5  0xffffffff81029f17 in trap_fatal (frame=0xfffffe016e360760, eva=0)
    at /usr/src/sys/amd64/amd64/trap.c:915
#6  0xffffffff81029f6f in trap_pfault (frame=0xfffffe016e360760, 
    usermode=<optimized out>, signo=<optimized out>, ucode=<optimized out>)
    at /usr/src/sys/amd64/amd64/trap.c:732
#7  0xffffffff810295cd in trap (frame=0xfffffe016e360760)
    at /usr/src/sys/amd64/amd64/trap.c:398
#8  <signal handler called>
#9  0x0000000000000000 in ?? ()
#10 0xffffffff82a14c4e in amdgpu_device_get_pcie_replay_count ()
   from /boot/modules/amdgpu.ko
#11 0xffffffff82a14b80 in sysctl_handle_attr () from /boot/modules/amdgpu.ko
#12 0xffffffff80c06cc1 in sysctl_root_handler_locked (oid=0xfffffe02133ff000, 
    arg1=0xfffffe016e360980, arg2=-8724518803888, req=0xfffffe016e360980, 
    tracker=0xfffff81099af6280) at /usr/src/sys/kern/kern_sysctl.c:184
#13 0xffffffff80c0610c in sysctl_root (oidp=<optimized out>, 
    arg1=0xfffff810aa27e650, arg2=-2100190360, req=0xfffffe016e360980)
    at /usr/src/sys/kern/kern_sysctl.c:2211
#14 0xffffffff80c06783 in userland_sysctl (td=0xfffffe00f00b6100, 
    name=0xfffffe016e360a40, namelen=4, old=<optimized out>, 
    oldlenp=<optimized out>, inkernel=<optimized out>, new=0x0, newlen=0, 
    retval=0xfffffe016e360aa8, flags=0)
    at /usr/src/sys/kern/kern_sysctl.c:2368
#15 0xffffffff80c065cf in sys___sysctl (td=0xfffffe00f00b6100, 
    uap=0xfffffe00f00b64e8) at /usr/src/sys/kern/kern_sysctl.c:2241
#16 0xffffffff8102a81c in syscallenter (td=0xfffffe00f00b6100)
    at /usr/src/sys/amd64/amd64/../../kern/subr_syscall.c:189
#17 amd64_syscall (td=0xfffffe00f00b6100, traced=0)
    at /usr/src/sys/amd64/amd64/trap.c:1156
#18 <signal handler called>
#19 0x00000008003819ca in ?? ()
Backtrace stopped: Cannot access memory at address 0x7fffffffb618
(kgdb)
```
I'm running:
```
FreeBSD photon.int.bluestop.org 13.0-CURRENT FreeBSD 13.0-CURRENT #0 6db1a3e8098-c273171(master): Thu Nov 26 01:26:17 MST 2020     bcran@photon.int.bluestop.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64
```

adriaandegroot, 2020-11-27:
Can't immediately reproduce with:

- Freshly downloaded *FreeBSD-13.0-CURRENT-amd64-20201126-9e082d278b9-memstick.img*
- Installed on a new disk in an i7-6700 with an AMD R7 360
- Installed pkg, then installed bash (why not) and drm-kmod
- Loaded amdgpu
- Run `sysctl sys` (it produces lots of useful output)

Relevant bits of dmesg might be:

```
FreeBSD 13.0-CURRENT #0 9e082d278b9-c254726(main): Thu Nov 26 04:50:43 UTC 2020
    root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
CPU: Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz (3408.13-MHz K8-class CPU)
[drm] initializing kernel modesetting (BONAIRE 0x1002:0x665F 0x1787:0x2360 0x81).
```

**but** this is a lousy reproducer because it's an older OS, it looks like (odd, for a nightly installer image) and my packages come from the quarterly branch by default. So, time to build things ..

myfreeweb, 2020-11-28:
> amdgpu_device_get_pcie_replay_count

Duplicate of #15 which will be fixed by #37 (or rather importing a similar upstream commit)

@adriaandegroot you won't ever reproduce this on an older GPU. This bug only affects Navi.

myfreeweb, 2020-11-28:
Also the upstream fix is included in 5.5, so you can just try #40 :)

bcran, 2020-12-11:
Thanks, I'll try that!

myfreeweb, 2020-12-12:
(note, as we just realized in the chat: https://github.com/freebsd/drm-kmod/commit/76f5b3931c6906894a7e9418509613890bfb3016 is not in 5.5 so cherry-pick that one if you want `sysctl -a` to not panic)

evadot, 2021-04-19:
Duplicate of https://github.com/freebsd/drm-kmod/issues/15 so closing this one.

closed 2021-04-19 by evadot