freebsd/drm-kmod #43: panic: !drm_modeset_is_locked(&plane->mutex) failed at drm_atomic_helper.c:871

lwhsu, 2020-11-30:
FreeBSD x1c.lwhsu.org 13.0-CURRENT FreeBSD 13.0-CURRENT #1 r367582M: Wed Nov 11 16:16:00 CST 2020     root@x1c.lwhsu.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64

drm-devel-kmod-5.4.62.g20201109
gpu-firmware-kmod-g20200920

Lenovo x1 carbon 6th, Intel Kaby Lake.

lwhsu, 2020-11-30:
Sorry that I don't have dump file available for this machine, hope this help a little bit:
[image]
[image]

wulf7, 2020-12-02:
Both panics are zfs-related. _!drm_modeset_is_locked_ is just a SPAM message here.

pbpjackd, 2021-01-09:
@lwhsu I happened to help testing https://reviews.freebsd.org/D28061 and the issue seems to be gone for me. ( I saw this message when connecting a secondary monitor)
Most likely you don't need to follow the whole test procedure described there but just build a new kernel from head.

As @wulf7 said, it's  zfs related. So this discussion http://freebsd.1045724.x6.nabble.com/page-fault-due-to-close-2-possibly-drm-and-i915kms-related-td6432416.html would backup this.  The thread also points to commit https://svnweb.freebsd.org/changeset/base/368271 (Dec. 2nd 2020) as the fix for the issue.

Maybe have a try. Just stumbled over it today.

evadot, 2021-01-19:
Not a drm bug, closing.

closed 2021-01-19 by evadot