freebsd/drm-kmod #46: GPU hang w. UHD620 on a 3rd gen Thinkpad X1 Yoga. Reset works on -CURRENT (didn't on -STABLE)

mathiasp, 2021-01-13:
Hi all,

I got told in the forums to report GPU hangs here.

This was worse on -STABLE, the GPU reset wouldn't work and I had to reboot via ssh.

I switched to -CURRENT (jan4) and get the same hangs, but the reset works.

The X1 has the newest bios, I just updated it.

dmesg https://pastebin.com/RGA3VLiB
xorg.log https://pastebin.com/8C49UGKk
glxinfo https://pastebin.com/Qnw03VSc

This happens in blender, whille rotating the default cube or with simple strokes of the Grease Pencil. It's rare enough after switching to -CURRENT that I can follow a Grease Pencil tutorial.

Possibly connected question: I asked in the forums and on freebsd-x11@ about the very laggy, stutterng behaviour with blender on this machine (compared to ubuntu on this machine, but also ubunto on a *Surface Go* (!!!)), but got no answer. Any idea where I can find someone willing to look into this? Could there be a connection?

mathiasp, 2021-01-13:
Oh, I just noticed there is no hang in the dmesg ;)
It looks like this:
Jan  7 19:51:35 Danton kernel: drmn0: GPU HANG: ecode 9:1:0x87f99eb9, in blender [100796], hang on rcs0
Jan  7 19:51:35 Danton kernel: drmn0: Resetting rcs0 for hang on rcs0

;)

neelchauhan, 2021-01-16:
My work laptop is a X1 Yoga 3rd gen, but my main FreeBSD laptop is a HP Spectre.

On the Yoga, it has no problems with 12.2-RELEASE and GNOME. But the Yoga normally runs Windows since my job is Windows heavy (disclaimer: I work at Microsoft). In fact, FreeBSD is never booted on the Yoga outside of the occasional `pkg update`.

mathiasp, 2021-01-16:
Well, this only happens with blender so far. Otherwise it's rock solid.

But blender is my only real OpenGL user, so it might happen in other OpenGL applications.

mathiasp, 2021-01-17:
I solved my speed problem with blender by using DRI2 or vblank_mode=0.
Still not sure what'st the actual culprit, or if both are connected, I'm new to this.

But the hang just happened again:

```
Jan 17 13:15:33 Danton kernel: drmn0: GPU HANG: ecode 9:1:0x85df9e9f, in blender [102289], hang on rcs0
Jan 17 13:15:33 Danton kernel: drmn0: Resetting rcs0 for hang on rcs0
```

This is from just scribbling a few words with blender Grease Pencil in the 2D animation view.

The Screen hangs for a second or so when this happens.

Any hints on how to debug this?

evadot, 2022-05-20:
Please try again with latest drm-kmod/mesa/etc ....
If this is still an issue please open a new one with all the info asked when opening an issue.

closed 2022-05-20 by evadot