freebsd/drm-kmod #47: Blank Screen when loading i915kms on Intel TigerLake

neelchauhan, 2021-01-16:
When I load `i915kms` on a HP Spectre x360 13t-aw200, I get a blank screen. No console output, nothing.

This was on the 5.4 branch, both self-compiled and `drm-devel-kmod` package/port.

My Spectre is a Intel TigerLake/"Evo" model. I believe this may affect other laptops including newer ThinkPads (e.g. X1 Carbon 9th Gen and sister models), but can't confirm.

OpenBSD has Linux 5.7 GPU drivers and a TigerLake-based Lenovo "X1 Nano" has graphics support ([according to developer jcs@](https://twitter.com/jcs/status/1350221200157528064)).

evadot, 2021-01-16:
Tiger Lake support was just introduced to 5.4 so I'm not that surprised that it doesn't work.
I plan to update to the latest lts before 13.0 is released, it might help.

neelchauhan, 2021-03-03:
Are you still planning to update `drm-kmod` to the latest LTS before 13.0? I haven't noticed any GitHub branch with that work.

Right now I'm working with `scfb`, but wouldn't mind having accelerated graphics.

neelchauhan, 2021-12-08:
This ticket is no longer relevant, `i915kms` now supports TigerLake (another Spectre, 14" this time, but works).

closed 2021-12-08 by neelchauhan