freebsd/drm-kmod #48: Error building drm-current-kmod-5.4.62.g20210113 on recent CURRENT

pkubaj, 2021-01-16:
This happens on powerpc64le, but it doesn't seem to be architecture-dependent (although I can't test on other architectures, my amd64 box is currently on 12.2-RELEASE).

```
cc  -O2 -pipe -fno-strict-aliasing '-DKBUILD_MODNAME="linuxkpi_gplv2"' -DLINUXKPI_VERSION=50000 -DCONFIG_DRM_AMDGPU_CIK -DCONFIG_DRM_AMDGPU_SI -DCONFIG_DRM_AMD_DC -DCONFIG_DRM_AMD_DC_FBC -DCONFIG_DRM_AMD_POWERPLAY -DCONFIG_DRM_I915_ALPHA_SUPPORT -DCONFIG_DRM_I915_FORCE_PROBE='"*"' -DCONFIG_DRM_I915_CAPTURE_ERROR -DCONFIG_DRM_I915_SPIN_REQUEST=5 -DCONFIG_DRM_I915_USERFAULT_AUTOSUSPEND=250 -DCONFIG_DRM_LOAD_EDID_FIRMWARE -DCONFIG_DRM_MIPI_DSI -DCONFIG_DRM_PANEL_ORIENTATION_QUIRKS -DCONFIG_DRM_VMWGFX_FBCON -DCONFIG_DRM_FBDEV_EMULATION -DCONFIG_DRM_FBDEV_OVERALLOC=100 -DCONFIG_DRM_LEGACY -DCONFIG_DRM_VM -DCONFIG_ARCH_HAVE_NMI_SAFE_CMPXCHG -DCONFIG_BACKLIGHT_CLASS_DEVICE -DCONFIG_DMI -DCONFIG_FB -DCONFIG_MTRR -DCONFIG_PCI -DCONFIG_PM -DCONFIG_SMP  -Werror -D_KERNEL -DKLD_MODULE -nostdinc  -I/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/include -I/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/dummy/include -I/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/gplv2/include -I/usr/src/sys/compat/linuxkpi/common/include -include /tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/obj/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/opt_global.h -I. -I/usr/src/sys -I/usr/src/sys/contrib/ck/include -fno-common  -fPIC -mlongcall -fno-omit-frame-pointer -fdebug-prefix-map=./machine=/usr/src/sys/powerpc/include     -MD  -MF.depend.linux_notifier.o -MTlinux_notifier.o -mno-altivec -msoft-float -mabi=elfv2 -ffreestanding -fwrapv -fstack-protector -Wall -Wredundant-decls -Wnested-externs -Wstrict-prototypes -Wmissing-prototypes -Wpointer-arith -Wcast-qual -Wundef -Wno-pointer-sign -D__printf__=__freebsd_kprintf__ -Wmissing-include-dirs -fdiagnostics-show-option -Wno-unknown-pragmas -Wno-error-tautological-compare -Wno-error-empty-body -Wno-error-parentheses-equality -Wno-error-unused-function -Wno-error-pointer-sign -Wno-error-shift-negative-value -Wno-address-of-packed-member -Wno-format-zero-length -Wno-pointer-arith     -std=iso9899:1999 -c /tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/gplv2/src/linux_notifier.c -o linux_notifier.o
/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/gplv2/src/linux_notifier.c:28:31: error: too few arguments provided to function-like macro invocation
static LIST_HEAD(shrinker_list);
                              ^
/usr/src/sys/sys/queue.h:443:9: note: macro 'LIST_HEAD' defined here
#define LIST_HEAD(name, type)                                           \
        ^
/tmp/usr/ports/graphics/drm-current-kmod/work/drm-kmod-drm_v5.4.62_7/linuxkpi/gplv2/src/linux_notifier.c:28:8: error: type specifier missing, defaults to 'int' [-Werror,-Wimplicit-int]
static LIST_HEAD(shrinker_list);
~~~~~~ ^
2 errors generated.
*** Error code 1

Stop.
```

I'm currently on 79a5c790bdf08cb925693add4699f3e785c12bc6

evadot, 2021-01-16:
I can't reproduce here.
Running f64329bcdc4f

But the list and sx lock should be under #if __FreeBSD_version < 1300135 I seems to have missed that.

pkubaj, 2021-01-19:
5.4.62.g20210118 builds correctly, closing.

closed 2021-01-19 by pkubaj