freebsd/drm-kmod #49: kernel panic from intel_frontbuffer_put() running OBS

emaste, 2021-01-17:
**Describe the bug**
Running OBS built from git at https://github.com/obsproject/obs-studio/commit/95c4db3bc61cbd2b893d4eeb598f37b4684ed288 with a scene comprising a USB webcam, encountered the panic below. (Note was OCR'd from a photo so there are some errors)
This has been happening on occasion for a couple of months.

```
panic: general protection fault 
cpuid = 0 
time = 1610898758 
KDB: stack backtrace: 
db_trace_self_wrapper() at db_trace self wrapper+8x2b/frame 8xfffffe011567c5a0 
vpanic() at vpanic+em181/frame exffiffelli1567c5f8 
panic() at panic+48x43/frame Oxfffffe811567c658 
trap_fatal() at trap_fatal+8x367/frame 8xfffffe011567c6b8 
trap() at trap+Oxallframe 8xfffffe811567c7c8 
calltrap() at calltrap+Ox8/frame 8xfffffe011567c7c8 
--- trap 0x9. rip = 8xffffffff831698dd, rsp = 8xfffffe811567c898, rbp = 8xfffffe011567c8a0 --
intel_frontbuffer_put() at intel_frontbuffer_put+Oxd/frame Oxfffffe811567c8a8 
intel_user_framebuffer_destroy() at intel_user_framebuffer destrovexla/frame Oxfffffe811567(
_drm atomic_helper_plane_destroy_state() at __drm_atomic_helper_plane destroy_state+Ox1b/fri
drm_atomic_helper_plane_destroy_state() at drm_atomic helper_plane desirog_state+8x11/frame I
drm_atomic_state_default_clear() at drm_atomic state_aefault_clear;Oxldciframe Oxfffffe81156.
intel_atomic_state_clear() at intel_atomic_staie clear+Oxe/frame Oxfffffe011567c96B
__drm_atomic_state_free() at drm_atomic_state_iree+8x28/frame 8xfffffe811567c988intel_atomic_cleanup_work() at into l atomic_cleanup_work+8x43/frame exfffffe011567c9b0
linux_work_fn() at linux work_fn+Oxeilframe Oxfffffe011567ca88 
taskqueue_run locked() at taskqueue_run_locked+ftaa/frame Oxfffffe811567ca88 
taskqueue_thrad_loop() at taskqueue thread_loop+Ox94/frame 8xfffffe011567cab0
fork_exit() at fork exit+8x88/frame ixfffffe011567caf8 
fork trampoline() at fork_trampoline+Oxe/frame 8xfffffe811567caf8 
--- trap 0, rip = 0, rsp = 0, rbp = 0 ---
KDB: enter: panic 
[ thread pid 0 tid 188598 ]
Stopped at kdb_enter+8x37: mow, $8,8x18cf1be(rip)
db> 
```

**FreeBSD version**
`FreeBSD nazar 13.0-CURRENT FreeBSD 13.0-CURRENT #25 laptop-c255860-gc529d54f882e: Sat Jan 16 20:58:04 EST 2021     emaste@nazar:/usr/obj/usr/home/emaste/src/freebsd-git/head/amd64.amd64/sys/GENERIC  amd64`

This is upstream rev 685de460bc91737331eaf475b0f517517b2ddbd0 with some unrelated userland WIP.

**PCI Info**
```
vgapci0@pci0:0:2:0:     class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x3ea0 subvendor=0x17aa subdevice=0x2292
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620 (Whiskey Lake)'
    class      = display
    subclass   = VGA
```

**To Reproduce**
I am unsure of the specific trigger; it occurs at a seemingly random time while OBS is running (not in relation to some event such as scene switching, starting/stopping recording etc.)

emaste, 2021-01-23:
Linux issue that appears related:

- [freedesktop bug](https://gitlab.freedesktop.org/drm/intel/-/issues/827)
- [patchwork link](https://patchwork.freedesktop.org/patch/345256/?series=70930)
- [backport to 5.4 failed](https://lore.kernel.org/stable/15783126692333@kroah.com/)

emaste, 2021-01-23:
I have backported the patch from https://patchwork.freedesktop.org/series/70930/#rev3 and it seems good so far.

emaste, 2021-01-25:
Backport here: https://github.com/emaste/drm-kmod/commit/5dfe923be9e6ed38cab5144673b6c51c20a80452

evadot, 2021-04-19:
Closing as the commit is present in 5.4-lts now.

closed 2021-04-19 by evadot