freebsd/drm-kmod #50: FreeBSD 13.0-ALPHA1 linker_load_file: /boot/modules/radeonkms.ko - unsupported file type

grahamperrin, 2021-01-19:
After updating from FreeBSD-CURRENT `main-c572-g82397d791`: 

* `radeonkms` no longer loads. 

From today's <https://bsd-hardware.info/?probe=fcedf7a28d> for non-bugged `main-c572-g82397d791`: 

* Thames [Radeon HD 7550M/7570M/7650M]

Output from `uname -a` for one of the bugged builds: 

`FreeBSD mowa219-gjp4-8570p 13.0-ALPHA1 FreeBSD 13.0-ALPHA1 #76 main-c1739-gc3e75b6c1a: Tue Jan 19 00:51:44 GMT 2021     root@mowa219-gjp4-8570p:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64`

Towards bisection: at least one earlier ALPHA is similarly bugged but at the moment, I can't tell the build reference. 

```text
root@mowa219-gjp4-8570p:~ # grep -C 1 linker /var/log/messages
Jan 19 06:37:32 mowa219-gjp4-8570p kernel: link_elf_obj: symbol linux_pci_get_class undefined
Jan 19 06:37:32 mowa219-gjp4-8570p kernel: linker_load_file: /boot/modules/radeonkms.ko - unsupported file type
Jan 19 06:37:47 mowa219-gjp4-8570p login[2382]: ROOT LOGIN (root) ON ttyv1
--
Jan 19 06:53:52 mowa219-gjp4-8570p kernel: link_elf_obj: symbol linux_pci_get_class undefined
Jan 19 06:53:52 mowa219-gjp4-8570p kernel: linker_load_file: /boot/modules/radeonkms.ko - unsupported file type
Jan 19 06:54:08 mowa219-gjp4-8570p login[2300]: 2 LOGIN FAILURES ON ttyv1
--
Jan 19 06:57:38 mowa219-gjp4-8570p kernel: Warning: memory type linuxcurrent leaked memory on destroy (2 allocations, 320 bytes leaked).
Jan 19 06:57:38 mowa219-gjp4-8570p kernel: linker_load_file: /boot/modules/radeonkms.ko - unsupported file type
Jan 19 06:57:53 mowa219-gjp4-8570p devd[1637]: notify_clients: send() failed; dropping unresponsive client
root@mowa219-gjp4-8570p:~ # 
```

```text
root@mowa219-gjp4-8570p:~ # grep -v \# /etc/make.conf
WITH_DEBUG_PORTS=webcamd

ICA_CERTS=QuoVadisRootCA2.crt

WITHOUT_LLVM_TARGET_ALL=
root@mowa219-gjp4-8570p:~ # grep -v \# /etc/src.conf
PORTS_MODULES= graphics/gpu-firmware-kmod graphics/drm-devel-kmod

BOOT_FRAMEBUFFER_MODE=yes
root@mowa219-gjp4-8570p:~ # 
```

– and if I recall correctly, with the earlier ALPHA, `drm-current-kmod` was similarly bugged.

evadot, 2021-01-19:
https://svnweb.freebsd.org/ports?view=revision&revision=561978 is supposed to fix this.
Could you test again please ?

grahamperrin, 2021-01-19:
Thank you! 

Incidentally I should have found this through (experimental) https://cgit-dev.freebsd.org/ports/log/?h=main&qt=grep&q=radeon however https://cgit-dev.freebsd.org/ports/ defaults to a 2014 branch, not `main`, and I overlooked the branch specificity when performing an earlier search. I'll report the issue.

evadot, 2021-01-19:
So is it fixed for you ?
Also I did some template for issue reporting, why did you not use it ?

grahamperrin, 2021-01-19:
Building now (slowly), it might be a day or so before I can confirm a fix but if you like we can close this now. 

Sorry for not strictly following the template, I added the headings and link to the PCI information (within the hardware probe).

grahamperrin, 2021-01-20:
Fix confirmed, thanks. 

Recent praise for all the work on DRM: https://old.reddit.com/r/freebsd/comments/kyoxmc/freebsd_quarterly_status_report_fourth_quarter/gjiw3y3/ :-)

```text
% date ; uname -v
Wed 20 Jan 2021 09:49:05 GMT
FreeBSD 13.0-ALPHA1 #77 main-c1790-geed1cc6cdf: Wed Jan 20 06:47:36 GMT 2021     root@mowa219-gjp4-8570p:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG 
% kldstat | grep -i radeon
55    1 0xffffffff83856000   150c70 radeonkms.ko
57    1 0xffffffff839b7000     3258 radeon_TURKS_pfp_bin.ko
58    1 0xffffffff839bb000     3658 radeon_TURKS_me_bin.ko
59    1 0xffffffff839bf000     2cd8 radeon_BTC_rlc_bin.ko
60    1 0xffffffff839c2000     7ef8 radeon_TURKS_mc_bin.ko
61    1 0xffffffff839ca000     8138 radeon_TURKS_smc_bin.ko
62    1 0xffffffff839d3000    341f0 radeon_SUMO_uvd_bin.ko
% 
```

closed 2021-01-20 by grahamperrin