freebsd/drm-kmod #51: Display freezes for tens of seconds at times (exclusive sleep mutex lkpi-shrinker)

0mp, 2021-01-21:
**Describe the bug**
As I'm using FreeBSD on my laptop for everyday tasks, the system seems to freeze every now and then. It's not like the whole system freezes. however, it's only the display. When the display finally unfreezes, I can see that, e.g., the Firefox tab has been closed for example.

**FreeBSD version**
FreeBSD t480 13.0-ALPHA1 FreeBSD 13.0-ALPHA1 #3 main-c255947-g420d4be3e40a: Thu Jan 14 10:40:45 CET 2021     0mp@t480:/usr/obj/usr/home/0mp/.local/share/freebsd/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

<details>
<summary>pciconf -lv</summary>

```
hostb0@pci0:0:0:0:	class=0x060000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x5914 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x07 hdr=0x00 vendor=0x8086 device=0x5917 subvendor=0x17aa subdevice=0x225e
    vendor     = 'Intel Corporation'
    device     = 'UHD Graphics 620'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x08 hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d2f subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP USB 3.0 xHCI Controller'
    class      = serial bus
    subclass   = USB
pchtherm0@pci0:0:20:2:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d31 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Thermal subsystem'
    class      = dasp
ig4iic0@pci0:0:21:0:	class=0x118000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d60 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP Serial IO I2C Controller'
    class      = dasp
none2@pci0:0:22:0:	class=0x078000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d3a subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP CSME HECI'
    class      = simple comms
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d10 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:6:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d16 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:29:0:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d18 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib8@pci0:0:29:2:	class=0x060400 rev=0xf1 hdr=0x01 vendor=0x8086 device=0x9d1a subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
isab0@pci0:0:31:0:	class=0x060100 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d4e subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point LPC Controller/eSPI Controller'
    class      = bridge
    subclass   = PCI-ISA
none3@pci0:0:31:2:	class=0x058000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d21 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP PMC'
    class      = memory
hdac0@pci0:0:31:3:	class=0x040300 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d71 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP HD Audio'
    class      = multimedia
    subclass   = HDA
ichsmb0@pci0:0:31:4:	class=0x0c0500 rev=0x21 hdr=0x00 vendor=0x8086 device=0x9d23 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Sunrise Point-LP SMBus'
    class      = serial bus
    subclass   = SMBus
em0@pci0:0:31:6:	class=0x020000 rev=0x21 hdr=0x00 vendor=0x8086 device=0x15d8 subvendor=0x17aa subdevice=0x225d
    vendor     = 'Intel Corporation'
    device     = 'Ethernet Connection (4) I219-V'
    class      = network
    subclass   = ethernet
vgapci1@pci0:1:0:0:	class=0x030200 rev=0xa1 hdr=0x00 vendor=0x10de device=0x1d10 subvendor=0x17aa subdevice=0x225e
    vendor     = 'NVIDIA Corporation'
    device     = 'GP108M [GeForce MX150]'
    class      = display
    subclass   = 3D
iwm0@pci0:3:0:0:	class=0x028000 rev=0x78 hdr=0x00 vendor=0x8086 device=0x24fd subvendor=0x8086 subdevice=0x0010
    vendor     = 'Intel Corporation'
    device     = 'Wireless 8265 / 8275'
    class      = network
pcib4@pci0:4:0:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:5:0:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:5:1:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:5:2:0:	class=0x060400 rev=0x01 hdr=0x01 vendor=0x8086 device=0x15c0 subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 Bridge (Low Power) [Alpine Ridge LP 2016]'
    class      = bridge
    subclass   = PCI-PCI
none4@pci0:6:0:0:	class=0x088000 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15bf subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 NHI (Low Power) [Alpine Ridge LP 2016]'
    class      = base peripheral
xhci1@pci0:60:0:0:	class=0x0c0330 rev=0x01 hdr=0x00 vendor=0x8086 device=0x15c1 subvendor=0x2222 subdevice=0x1111
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016]'
    class      = serial bus
    subclass   = USB
nvme0@pci0:61:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x144d device=0xa808 subvendor=0x144d subdevice=0xa801
    vendor     = 'Samsung Electronics Co Ltd'
    device     = 'NVMe SSD Controller SM981/PM981/PM983'
    class      = mass storage
    subclass   = NVM
```
</details>

**DRM KMOD version**

drm-current-kmod 5.4.62.g20210118 drm-kmod g20190710

**To Reproduce**
The freezing happens much more frequently (and for longer periods I think) when using Firefox actively in than when I'm just in a full-screen terminal session (although it does happen as well). 

**Additional context**
I think it might be related to those messages I see a lot in `dmesg`:

```
might_sleep() with the following non-sleepable locks held:
exclusive sleep mutex lkpi-shrinker (lkpi-shrinker) r = 0 (0xffffffff839a4340) locked @ /usr/home/0mp/.local/share/freebsd/src/sys/compat/linuxkpi/common/src/linux_shrinker.c:96
stack backtrace:
#0 0xffffffff80c641c1 at witness_debugger+0x71
#1 0xffffffff80c652dd at witness_warn+0x40d
#2 0xffffffff837db79d at i915_vma_unbind+0x4d
#3 0xffffffff837bb8c8 at i915_gem_object_unbind+0x78
#4 0xffffffff838bf9cc at i915_gem_shrink+0x2dc
#5 0xffffffff838bfda8 at i915_gem_shrinker_scan+0x98
#6 0xffffffff8399beed at linuxkpi_vm_lowmem+0x7d
#7 0xffffffff80f03a80 at vm_pageout_worker+0x3d0
#8 0xffffffff80f0369e at vm_pageout+0x21e
#9 0xffffffff80bafb10 at fork_exit+0x80
#10 0xffffffff81024ade at fork_trampoline+0xe
```

evadot, 2021-01-22:
This should be fixed by 4af93235426012049161d8
Can you update and check ?
Thanks.

0mp, 2021-01-22:
> This should be fixed by 4af93235426012049161d8
> Can you update and check ?

Sure, I'll be happy to test the patch, but I cannot locate the commit you've referenced. Could you point me to the code I can test? Thanks!

0mp, 2021-01-23:
Sure, I'll test it out and report back. Thanks!

0mp, 2021-01-23:
It seems like the problem has been solved. I'll reopen this issue if I encounter this bug again. Thanks, @evadot

closed 2021-01-23 by 0mp