freebsd/drm-kmod #57: drm-devel-kmod FREEBSD14-current NAVI RADEON 5700XT

justfly1111, 2021-02-02:
So on FreeBSD 13-current drm-devel-kmod worked perfectly fine with my radeon 5700xt video cards i upgraded to 14-current and navi support wont work. i recompiled the most recent update of drm-devel-kmod on 14-current and still no dice i cant get navi support to work please help! the following is the dmesg of it being initialized:

AMD-Vi: IVRS Info VAsize = 64 PAsize = 48 GVAsize = 2 flags:0
driver bug: Unable to set devclass (class: ppc devname: (unknown))
ivhd0: <AMD-Vi/IOMMU ivhd with EFR> on acpi0
ivhd0: Flag:b0<IotlbSup,Coherent>
ivhd0: Features(type:0x11) MsiNumPPR = 0 PNBanks= 2 PNCounters= 0
ivhd0: Extended features[31:0]:22294ade<PPRSup,<b2>,NXSup,GTSup,IASup> HATS = 0x2 GATS = 0x0 GLXSup = 0x1 SmiFSup = 0x1 SmiFRC = 0x2 GAMSup = 0x1 DualPortLogSup = 0x2 DualEventLogSup = 0x2
ivhd0: Extended features[62:32]:58f77ef<USSup> Max PASID: 0x2f DevTblSegSup = 0x3 MarcSup = 0x1
ivhd0: supported paging level:7, will use only: 4
ivhd0: device range: 0x0 - 0xfffe
ivhd0: PCI cap 0x190b640f@0x40 feature:19<IOTLB,EFR,CapExt>
ivhd1: <AMD-Vi/IOMMU ivhd with EFR> on acpi0
ivhd1: Flag:b0<IotlbSup,Coherent>
ivhd1: Features(type:0x11) MsiNumPPR = 0 PNBanks= 2 PNCounters= 0
ivhd1: Extended features[31:0]:22294ade<PPRSup,<b2>,NXSup,GTSup,IASup> HATS = 0x2 GATS = 0x0 GLXSup = 0x1 SmiFSup = 0x1 SmiFRC = 0x2 GAMSup = 0x1 DualPortLogSup = 0x2 DualEventLogSup = 0x2
ivhd1: Extended features[62:32]:58f77ef<USSup> Max PASID: 0x2f DevTblSegSup = 0x3 MarcSup = 0x1
ivhd1: supported paging level:7, will use only: 4
ivhd1: device range: 0x0 - 0xfffe
ivhd1: PCI cap 0x190b640f@0x40 feature:19<IOTLB,EFR,CapExt>
ivhd2: <AMD-Vi/IOMMU ivhd with EFR> on acpi0
ivhd2: Flag:b0<IotlbSup,Coherent>
ivhd2: Features(type:0x11) MsiNumPPR = 0 PNBanks= 2 PNCounters= 0
ivhd2: Extended features[31:0]:22294ade<PPRSup,<b2>,NXSup,GTSup,IASup> HATS = 0x2 GATS = 0x0 GLXSup = 0x1 SmiFSup = 0x1 SmiFRC = 0x2 GAMSup = 0x1 DualPortLogSup = 0x2 DualEventLogSup = 0x2
ivhd2: Extended features[62:32]:58f77ef<USSup> Max PASID: 0x2f DevTblSegSup = 0x3 MarcSup = 0x1
ivhd2: supported paging level:7, will use only: 4
ivhd2: device range: 0x0 - 0x5ffe
ivhd2: PCI cap 0x190b640f@0x40 feature:19<IOTLB,EFR,CapExt>
ivhd3: <AMD-Vi/IOMMU ivhd with EFR> on acpi0
ivhd3: Flag:b0<IotlbSup,Coherent>
ivhd3: Features(type:0x11) MsiNumPPR = 0 PNBanks= 2 PNCounters= 0
ivhd3: Extended features[31:0]:22294ade<PPRSup,<b2>,NXSup,GTSup,IASup> HATS = 0x2 GATS = 0x0 GLXSup = 0x1 SmiFSup = 0x1 SmiFRC = 0x2 GAMSup = 0x1 DualPortLogSup = 0x2 DualEventLogSup = 0x2
ivhd3: Extended features[62:32]:58f77ef<USSup> Max PASID: 0x2f DevTblSegSup = 0x3 MarcSup = 0x1
ivhd3: supported paging level:7, will use only: 4
ivhd3: device range: 0x0 - 0x5ffe
ivhd3: PCI cap 0x190b640f@0x40 feature:19<IOTLB,EFR,CapExt>
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] initializing kernel modesetting (NAVI10 0x1002:0x731F 0x148C:0x2400 0xC0).
[drm] register mmio base: 0xE1C00000
[drm] register mmio size: 524288
[drm] set register base offset for ATHUB
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for CLKA
[drm] set register base offset for DF
[drm] set register base offset for DMU
[drm] set register base offset for GC
[drm] set register base offset for HDP
[drm] set register base offset for MMHUB
[drm] set register base offset for MP0
[drm] set register base offset for MP1
[drm] set register base offset for NBIF
[drm] set register base offset for NBIF
[drm] set register base offset for OSSSYS
[drm] set register base offset for SDMA0
[drm] set register base offset for SDMA1
[drm] set register base offset for SMUIO
[drm] set register base offset for THM
[drm] set register base offset for UVD
[drm] add ip block number 0 <nv_common>
[drm] add ip block number 1 <gmc_v10_0>
[drm] add ip block number 2 <navi10_ih>
[drm] add ip block number 3 <psp>
[drm] add ip block number 4 <smu>
[drm] add ip block number 5 <dm>
[drm] add ip block number 6 <gfx_v10_0>
[drm] add ip block number 7 <sdma_v5_0>
[drm] add ip block number 8 <vcn_v2_0>
KLD amdgpu_navi10_gpu_info_bin.ko: depends on kernel - not available or version mismatch
linker_load_file: /boot/modules/amdgpu_navi10_gpu_info_bin.ko - unsupported file type
drmn0: could not load firmware image 'amdgpu/navi10_gpu_info.bin'
drmn0: Failed to load gpu_info firmware "amdgpu/navi10_gpu_info.bin"
drmn0: Fatal error during GPU init
[drm] amdgpu: finishing device.
Warning: can't remove non-dynamic nodes (dri)!
device_attach: drmn0 attach returned 2

please  help thank you

unrelentingtech, 2021-02-08:
> amdgpu_navi10_gpu_info_bin.ko - unsupported file type

Recompile the gpu-firmware modules as well

evadot, 2021-04-19:
Closing for now, please reopen if recompiling the firmware didn't helped.

closed 2021-04-19 by evadot