freebsd/drm-kmod #59: amdgpu_dm_irq_schedule_work FAILED src 5

pkubaj, 2021-02-22:
**Describe the bug**
I have RX560 running on 13.0-BETA3 on amd64 workstation.

When my monitors are off, I'm getting the following errors on the console:
`<6>[drm] amdgpu_dm_irq_schedule_work FAILED src 5`

When it happens, X11 stops working, but I can still log in via SSH. Shortly after that, SSH and console logging stops working as well.

Restarting X11, in the time window where I can still log in, doesn't help. Xorg doesn't start at all.
Only reboot helps in that case.

**FreeBSD version**
_FreeBSD KGPE-D16 13.0-BETA3 FreeBSD 13.0-BETA3 #0 releng/13.0-n244525-150b4388d3b: Fri Feb 19 01:36:53 CET 2021     root@KGPE-D16:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64_

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
hostb0@pci0:0:0:0:	class=0x060000 rev=0x02 hdr=0x00 vendor=0x1002 device=0x5a10 subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890 Northbridge only dual slot (2x16) PCI-e GFX Hydra part'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1002 device=0x5a23 subvendor=0x1002 subdevice=0x5a23
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890S/RD990 I/O Memory Management Unit (IOMMU)'
    class      = base peripheral
    subclass   = IOMMU
pcib1@pci0:0:2:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a16 subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GFX port 0)'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:4:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a18 subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 0)'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:9:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a1c subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 4)'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:10:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a1d subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP Port 5)'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:11:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a1f subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD990 PCI to PCI bridge (PCI Express GFX2 port 0)'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:12:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a20 subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD990 PCI to PCI bridge (PCI Express GFX2 port 1)'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:13:0:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1002 device=0x5a1e subvendor=0x1002 subdevice=0x5a10
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'RD890/RD9x0/RX980 PCI to PCI bridge (PCI Express GPP2 Port 0)'
    class      = bridge
    subclass   = PCI-PCI
ahci0@pci0:0:17:0:	class=0x010601 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4394 subvendor=0x1043 subdevice=0x8163
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'SB7x0/SB8x0/SB9x0 SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
ohci0@pci0:0:18:0:	class=0x0c0310 rev=0x00 hdr=0x00 vendor=0x1002 device=0x4397 subvendor=0x1043 subdevice=0x8163
[... 216 lines cut ...]
    device     = 'FW322/323 [TrueFire] 1394a Controller'
    class      = serial bus
    subclass   = FireWire
emu10kx0@pci0:8:3:0:	class=0x040100 rev=0x0a hdr=0x00 vendor=0x1102 device=0x0002 subvendor=0x1102 subdevice=0x8065
    vendor     = 'Creative Labs'
    device     = 'EMU10k1 [Sound Blaster Live! Series]'
    class      = multimedia
    subclass   = audio
none2@pci0:8:3:1:	class=0x098000 rev=0x0a hdr=0x00 vendor=0x1102 device=0x7002 subvendor=0x1102 subdevice=0x0020
    vendor     = 'Creative Labs'
    device     = 'SB Live! Game Port'
    class      = input device

</details>

**DRM KMOD version**
drm-fbsd13-kmod 5.4.92.g20210202

**To Reproduce**
Steps to reproduce the behavior:
With DPMS on, allow your screen to be suspended and wait some time.

**Additional context**
I'm not sure whether it happens with one screen connected, I have two screens.
Everything worked fine on 12.2-RELEASE.

unrelentingtech, 2021-02-24:
Very similar trace: https://lkml.org/lkml/2020/6/28/107

pkubaj, 2021-03-02:
It looks like disabling DPMS does indeed workaround this issue, but `xset -dpms` alone is not enough.
I had to do:
```
xset s off
xset s noblank
xset -dpms
xset dpms 0 0 0
```

evadot, 2022-06-06:
Could you try again with 5.10 ?

pkubaj, 2022-06-06:
I moved all my desktop activities to powerpc64le workstation with a different GPU. I don't encounter the reported issue there.

closed 2022-06-06 by evadot