freebsd/drm-kmod #60: Finally Fixing The AMD EFIFB Conflict

unrelentingtech, 2021-03-02:
This years old issue https://github.com/FreeBSDDesktop/DEPRECATED-freebsd-base-graphics/issues/170 that required `hw.syscons.disable` on some systems. I know the answer now.

So, amdgpu actually kicks the system (EFI) framebuffer driver out very early on, to prevent it from messing with VRAM (BAR 0). There's this [`remove_conflicting_pci_framebuffers`](https://github.com/freebsd/drm-kmod/blob/4e1804b02c416377db1acb6a5a849570a8693def/drivers/gpu/drm/linux_fb.c#L372) function it uses to accomplish that. We have a very complete implementation of this stuff that supports all the registration and unregistration of *Linux* framebuffer drivers…

See where this is going, right?

.

.

**It only works with LinuxKPI fb drivers. It cannot kill our native efifb!**

So what's happening is, our efifb keeps happily writing to VRAM way after amdgpu expects it to stop, and if the framebuffer is large enough (e.g. 2560x1440 which is GOP mode 0 for me) it would directly overwrite stuff like the firmware the driver is loading into the card. \*facepalm\*

---

My current prototype solution is: adding a dummy vt driver where all methods are no-ops, adding a `force` flag to `vt_allocate` to avoid priority stuff, and doing the `vt_allocate(&vt_dummy_driver, NULL, true)` thing in `__remove_conflicting`.

If anyone has suggestions on less invasive / stupid ways to accomplish this, you're welcome. I have looked into replacing the driver with `NULL` (as if we booted with `syscons.disable` and none matched, right?) but the system freezes in that case.

Cacodemon345, 2021-05-18:
Booting with 640x480 GOP resolution seems to fix this in FreeBSD 13.0 and later.

unrelentingtech, 2021-05-18:
@Cacodemon345 of course. The "it works fine on lower resolution" observation is how I eventually arrived at the solution.

PR #61 (2021-07-06): Switch to dummy vt backend in __remove_conflicting when needed
AMD drivers use the remove_conflicting_pci_framebuffers function to kick the EFI framebuffer out early, to prevent it from messing with VRAM, corrupting things used by the driver loading process.

Our implementation could not actually remove native vt drivers, only LinuxKPI registered ones.

This commit solves the amdgpu/radeon efifb conflict, i.e. fixes #60

---

<details>
<summary>example dmesg log</summary>

```
[4] <6>[drm] amdgpu kernel modesetting enabled.
[4] drmn0: <drmn> on vgapci0
[4] VT: Replacing driver "efifb" with new [... cut]
closed 2021-07-06 by evadot