freebsd/drm-kmod #62: Add support for at least one qemu virtual gpu

jfouquart, 2021-03-03:
**Description**
Could you please add support for at least one of these qemu virtual devices:

- Virtio-gpu

- Qxl

- Bochs

**FreeBSD version**
13 or higher

**Additional context**
I have try by myself to support on of those, alas I always have the same result, when I load the kernel module, I obtain a black screen :(

Here are my git branches based on 5.4.x branch if it helps:

- https://github.com/jfouquart/drm-kmod/commits/bochs

- https://github.com/jfouquart/drm-kmod/commits/qxl

jfouquart, 2021-03-06:
I continued to investigate the problem but I still don't understand what's going on :(

When I load the kernel module always black screen, but the verbose seems to tell me that everything is fine (see [messages.log](https://github.com/freebsd/drm-kmod/files/6095683/messages.log)). If I boot an X, the kernel crashes on a ttm_bo_vm_fault ( backtrace in copy )

```
__curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
55		__asm("movq %%gs:%P1,%0" : "=r" (td) : "n" (offsetof(struct pcpu,
(kgdb) bt
#0  __curthread () at /usr/src/sys/amd64/include/pcpu_aux.h:55
#1  doadump (textdump=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:399
#2  0xffffffff80c09856 in kern_reboot (howto=260) at /usr/src/sys/kern/kern_shutdown.c:486
#3  0xffffffff80c09cd0 in vpanic (fmt=<optimized out>, ap=<optimized out>) at /usr/src/sys/kern/kern_shutdown.c:919
#4  0xffffffff80c09ad3 in panic (fmt=<unavailable>) at /usr/src/sys/kern/kern_shutdown.c:843
#5  0xffffffff8108a187 in trap_fatal (frame=0xfffffe00984ce480, eva=32) at /usr/src/sys/amd64/amd64/trap.c:915
#6  0xffffffff8108a1df in trap_pfault (frame=frame@entry=0xfffffe00984ce480, usermode=false, signo=<optimized out>, 
    signo@entry=0x0, ucode=<optimized out>, ucode@entry=0x0) at /usr/src/sys/amd64/amd64/trap.c:732
#7  0xffffffff8108983d in trap (frame=0xfffffe00984ce480) at /usr/src/sys/amd64/amd64/trap.c:398
#8  <signal handler called>
#9  vm_page_busy_acquire (m=0x0, allocflags=16) at /usr/src/sys/vm/vm_page.c:885
#10 0xffffffff8360902a in ttm_bo_vm_fault () from /boot/modules/ttm.ko
#11 0xffffffff8351832c in qxl_ttm_fault () from /boot/modules/qxl.ko
#12 0xffffffff80e3f1fb in linux_cdev_pager_populate (vm_obj=0xfffff8002f9ace70, pidx=<optimized out>, 
    fault_type=<optimized out>, max_prot=<optimized out>, first=0xfffffe00984ce7f8, last=0xfffffe00984ce7f0)
    at /usr/src/sys/compat/linuxkpi/common/src/linux_compat.c:549
#13 0xffffffff80f27e62 in vm_pager_populate (object=0x0, pidx=16, fault_type=1024, max_prot=<optimized out>, 
    first=0xfffffe00984ce7f8, last=0xfffffe00984ce7f0) at /usr/src/sys/vm/vm_pager.h:177
#14 vm_fault_populate (fs=0xfffffe00984ce890) at /usr/src/sys/vm/vm_fault.c:443
#15 vm_fault_allocate (fs=fs@entry=0xfffffe00984ce890) at /usr/src/sys/vm/vm_fault.c:1072
#16 0xffffffff80f26b96 in vm_fault (map=<optimized out>, map@entry=0xfffffe00a1f543e0, vaddr=vaddr@entry=34552676352, 
    fault_type=fault_type@entry=2 '\002', fault_flags=<optimized out>, fault_flags@entry=0, m_hold=<optimized out>, 
    m_hold@entry=0x0) at /usr/src/sys/vm/vm_fault.c:1404
#17 0xffffffff80f2665d in vm_fault_trap (map=0xfffffe00a1f543e0, vaddr=<optimized out>, vaddr@entry=34552676352, 
    fault_type=<optimized out>, fault_flags=fault_flags@entry=0, signo=0xfffffe00984ceac4, ucode=0xfffffe00984ceac0)
    at /usr/src/sys/vm/vm_fault.c:631
--Type <RET> for more, q to quit, c to continue without paging--
#18 0xffffffff8108a388 in trap_pfault (frame=frame@entry=0xfffffe00984ceb00, usermode=true, signo=0xfffff8006a6c7a80, 
    signo@entry=0xfffffe00984ceac4, ucode=0x0, ucode@entry=0xfffffe00984ceac0) at /usr/src/sys/amd64/amd64/trap.c:817
#19 0xffffffff810899bd in trap (frame=0xfffffe00984ceb00) at /usr/src/sys/amd64/amd64/trap.c:340
#20 <signal handler called>
#21 0x0000000800858382 in ?? ()
```

Can someone help me please.

jfouquart, 2022-04-27:
Sadly, since one year, I don't have time to work on this project but some hopes are allowed since @[arichardson](https://github.com/arichardson) is working on this PR.  https://github.com/freebsd/drm-kmod/pull/119 #crossfinger

grahamperrin, 2026-11-04:
Plasma (Wayland) on 16.0-CURRENT in Virtual Machine Manager on Kubuntu: 

<details>
<summary>Screenshot</summary>

[image]

</details>

This might be more difficult to achieve with `releng/15.1`. I'm discussing elsewhere (what's pictured above is not a FreeBSD Project development).