freebsd/drm-kmod #64: panic on 14-CURRENT with AMD NAVI10

thisplacestinksoffascism, 2021-03-09:
**Describe the bug**
System panics when issuing `sysctl -a`

**FreeBSD version**
FreeBSD direwolf.local. 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n245338-221622ec0c8-dirty: Tue Mar  9 15:30:10 AWST 2021     agh@direwolf.local.:/tmp/direwolf/usr/src/amd64.amd64/sys/DIREWOLF_DEBUG  amd64

-dirty because of:
index 30d6d94723f..fbcd176457d 100644
--- a/sbin/sysctl/sysctl.c
+++ b/sbin/sysctl/sysctl.c
@@ -336,6 +336,12 @@ parse_numeric(const char *newvalstr, const char *fmt, u_int kind,
        return (true);
 }

+/*
+#define                sysctl(...) ({ \
+       usleep(100000); \
+       sysctl(__VA_ARGS__); })
+*/
+

Which is used for debugging this panic.

**PCI Info**

hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1022 subdevice=0x1480
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1481 subvendor=0x1022 subdevice=0x1481
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib14@pci0:0:3:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
[... 203 lines cut ...]
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Cryptographic Coprocessor PSPCPP'
    class      = encrypt/decrypt
xhci2@pci0:19:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x149c subvendor=0x1849 subdevice=0xffff
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse USB 3.0 Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:19:0:4:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1487 subvendor=0x1849 subdevice=0x2221
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse HD Audio Controller'
    class      = multimedia
    subclass   = HDA

**DRM KMOD version**

drm-devel-kmod 5.4.92.g20210128_1

**To Reproduce**
Steps to reproduce the behavior:
sysctl -a

**Additional context**

https://docs.freebsd.org/cgi/getmsg.cgi?fetch=202014+0+archive/2021/freebsd-current/20210207.freebsd-current

hselasky, 2021-03-09:
> Always panics at:
> $ /tmp/sysctl-with-delay -a
> [...]
> p1003_1b.mapped_files: 200112
> p1003_1b.memlock: 0
> p1003_1b.memlock_range: 0
> p1003_1b.memory_protection: 0
> p1003_1b.message_passing: 0
> p1003_1b.prioritized_io: 0
> p1003_1b.priority_scheduling: 200112
> p1003_1b.realtime_signals: 200112
> p1003_1b.semaphores: 0
> p1003_1b.fsync: 200112
> p1003_1b.shared_memory_objects: 200112
> p1003_1b.synchronized_io: 0
> p1003_1b.timers: 200112
> p1003_1b.aio_listio_max: 256
> p1003_1b.aio_max: 1024
> p1003_1b.aio_prio_delta_max: 0
> p1003_1b.delaytimer_max: 2147483647
> p1003_1b.mq_open_max: 0
> p1003_1b.pagesize: 4096
> p1003_1b.rtsig_max: 62
> p1003_1b.sem_nsems_max: 0
> p1003_1b.sem_value_max: 0
> p1003_1b.sigqueue_max: 128
> p1003_1b.timer_max: 32
>

Hi,

On my system:

p1003_1b.timer_max: 32
sys.class.drm.card0-HDMI-A-1.modes: 1680x1050
1600x1200
1280x1024
1440x900
1280x960
1024x768
800x600
640x480
720x400

So that means some modes sysctl is causing the panic.

hselasky, 2021-03-09:
@evadot

unrelentingtech, 2021-03-09:
Is this #15 / #41 again? The last comment in #15 says @evadot didn't do the promised cherry-pick of the upstream fix for that :)

hselasky, 2021-03-09:
@unrelentingtech : Looks very similar!

evadot, 2021-03-11:
The fixing commit is now in 5.4-lts, if you could try again.
Thanks

thisplacestinksoffascism, 2021-03-12:
Works great on 14.0-CURRENT #0 main-n245400-e75eac2cb81
```
$ sysctl sys.class.drm
sys.class.drm.card0-HDMI-A-1.modes: 
sys.class.drm.card0-HDMI-A-1.dpms: Off
sys.class.drm.card0-HDMI-A-1.enabled: disabled
sys.class.drm.card0-HDMI-A-1.status: disconnected
sys.class.drm.card0-DP-3.modes: 
sys.class.drm.card0-DP-3.dpms: Off
sys.class.drm.card0-DP-3.enabled: disabled
sys.class.drm.card0-DP-3.status: disconnected
sys.class.drm.card0-DP-2.modes: 2560x1440
2560x1440
2560x1440
2560x1440
1920x1200
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1600x1200
1680x1050
1600x900
1280x1024
1280x1024
1440x900
1280x800
1152x864
1280x720
1280x720
1280x720
1280x720
1024x768
1024x768
800x600
800x600
720x480
720x480
640x480
640x480
640x480
640x480
sys.class.drm.card0-DP-2.dpms: On
sys.class.drm.card0-DP-2.enabled: enabled
sys.class.drm.card0-DP-2.status: connected
sys.class.drm.card0-DP-1.modes: 2560x1440
2560x1440
2560x1440
2560x1440
1920x1200
1920x1080
1920x1080
1920x1080
1920x1080
1920x1080
1600x1200
1680x1050
1600x900
1280x1024
1280x1024
1440x900
1280x800
1152x864
1280x720
1280x720
1280x720
1280x720
1024x768
1024x768
800x600
800x600
720x480
720x480
640x480
640x480
640x480
640x480
sys.class.drm.card0-DP-1.dpms: On
sys.class.drm.card0-DP-1.enabled: enabled
sys.class.drm.card0-DP-1.status: connected
sys.class.drm.ttm.buffer_objects.bo_count: 394
sys.class.drm.ttm.memory_accounting.pool.pool_allocation_size: 2048
sys.class.drm.ttm.memory_accounting.pool.pool_small_allocation: 64
sys.class.drm.ttm.memory_accounting.pool.pool_max_size: 16752464
sys.class.drm.ttm.memory_accounting.dma32.used_memory: 8544
sys.class.drm.ttm.memory_accounting.dma32.swap_limit: 1572864
sys.class.drm.ttm.memory_accounting.dma32.available_memory: 2097152
sys.class.drm.ttm.memory_accounting.dma32.emergency_memory: 3145728
sys.class.drm.ttm.memory_accounting.dma32.zone_memory: 4194304
sys.class.drm.ttm.memory_accounting.kernel.used_memory: 64016
sys.class.drm.ttm.memory_accounting.kernel.swap_limit: 25128700
sys.class.drm.ttm.memory_accounting.kernel.available_memory: 33504934
sys.class.drm.ttm.memory_accounting.kernel.emergency_memory: 50257401
sys.class.drm.ttm.memory_accounting.kernel.zone_memory: 67009868
sys.class.drm.ttm.memory_accounting.lower_mem_limit: 0
sys.class.drm.version: drm 1.1.0 20060810
```

evadot, 2021-03-12:
Thanks for testing, I'll update the ports this weekend probably.

closed 2021-03-12 by evadot