freebsd/drm-kmod #66: screen disabled when loading radeonkms.ko with SUMO chip

Zirias, 2021-03-12:
**Describe the bug**
Loading `radeonkms.ko` disables the screen with error message `[drm] Cannot find any crtc or sizes`

**FreeBSD version**
FreeBSD 13.0-RC2 #11 releng/13.0-n244684-13c22f74953: Fri Mar 12 09:49:28 CET 2021

**PCI Info**

<details>
  <summary>pciconf -lv</summary>

hostb0@pci0:0:0:0:	class=0x060000 card=0x17051849 chip=0x17051022 rev=0x00 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 12h Processor Root Complex'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:1:0:	class=0x030000 card=0x96401849 chip=0x964a1002 rev=0x00 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Sumo [Radeon HD 6530D]'
    class      = display
    subclass   = VGA
hdac0@pci0:0:1:1:	class=0x040300 card=0x17141849 chip=0x17141002 rev=0x00 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'BeaverCreek HDMI Audio [Radeon HD 6500D and 6400G-6600G series]'
    class      = multimedia
    subclass   = HDA
pcib1@pci0:0:4:0:	class=0x060400 card=0x12341022 chip=0x17091022 rev=0x00 hdr=0x01
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 12h Processor Root Port'
    class      = bridge
    subclass   = PCI-PCI
xhci0@pci0:0:16:0:	class=0x0c0330 card=0x78121849 chip=0x78121022 rev=0x03 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB XHCI Controller'
    class      = serial bus
    subclass   = USB
xhci1@pci0:0:16:1:	class=0x0c0330 card=0x78121849 chip=0x78121022 rev=0x03 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB XHCI Controller'
    class      = serial bus
    subclass   = USB
ahci0@pci0:0:17:0:	class=0x010601 card=0x78001849 chip=0x78011022 rev=0x40 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
ohci0@pci0:0:18:0:	class=0x0c0310 card=0x78071849 chip=0x78071022 rev=0x11 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB OHCI Controller'
    class      = serial bus
    subclass   = USB
ehci0@pci0:0:18:2:	class=0x0c0320 card=0x78081849 chip=0x78081022 rev=0x11 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB EHCI Controller'
    class      = serial bus
    subclass   = USB
ohci1@pci0:0:19:0:	class=0x0c0310 card=0x78071849 chip=0x78071022 rev=0x11 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB OHCI Controller'
    class      = serial bus
    subclass   = USB
ehci1@pci0:0:19:2:	class=0x0c0320 card=0x78081849 chip=0x78081022 rev=0x11 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH USB EHCI Controller'
    class      = serial bus
    subclass   = USB
intsmb0@pci0:0:20:0:	class=0x0c0500 card=0x780b1849 chip=0x780b1022 rev=0x13 hdr=0x00
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
[... 145 lines cut ...]
Mar 12 13:06:37 nexus kernel: [drm] ib test on ring 3 succeeded in 0 usecs
Mar 12 13:06:37 nexus kernel: [drm] ib test on ring 5 succeeded
Mar 12 13:06:37 nexus kernel: [drm] Connector DVI-D-1: get mode from tunables:
Mar 12 13:06:37 nexus kernel: [drm]   - kern.vt.fb.modes.DVI-D-1
Mar 12 13:06:37 nexus kernel: [drm]   - kern.vt.fb.default_mode
Mar 12 13:06:37 nexus kernel: [drm] Connector DVI-D-2: get mode from tunables:
Mar 12 13:06:37 nexus kernel: [drm]   - kern.vt.fb.modes.DVI-D-2
Mar 12 13:06:37 nexus kernel: [drm]   - kern.vt.fb.default_mode
Mar 12 13:06:37 nexus kernel: [drm] Radeon Display Connectors
Mar 12 13:06:37 nexus kernel: [drm] Connector 0:
Mar 12 13:06:37 nexus kernel: [drm]   DVI-D-1
Mar 12 13:06:37 nexus kernel: [drm]   HPD2
Mar 12 13:06:37 nexus kernel: [drm]   DDC: 0x6440 0x6440 0x6444 0x6444 0x6448 0x6448 0x644c 0x644c
Mar 12 13:06:37 nexus kernel: [drm]   Encoders:
Mar 12 13:06:37 nexus kernel: [drm]     DFP2: INTERNAL_UNIPHY2
Mar 12 13:06:37 nexus kernel: [drm] Connector 1:
Mar 12 13:06:37 nexus kernel: [drm]   DVI-D-2
Mar 12 13:06:37 nexus kernel: [drm]   HPD1
Mar 12 13:06:37 nexus kernel: [drm]   DDC: 0x6430 0x6430 0x6434 0x6434 0x6438 0x6438 0x643c 0x643c
Mar 12 13:06:37 nexus kernel: [drm]   Encoders:
Mar 12 13:06:37 nexus kernel: [drm]     DFP1: INTERNAL_UNIPHY2
Mar 12 13:06:37 nexus kernel: [drm] Cannot find any crtc or sizes
Mar 12 13:06:37 nexus kernel: [drm] Initialized radeon 2.50.0 20080528 for drmn0 on minor 0
Mar 12 13:06:39 nexus kernel: [drm] Cannot find any crtc or sizes
```

Zirias, 2021-03-12:
For comparison, here's the log when loading `radeonkms.ko` from `drm-fbsd11.2-kmod` on FreeBSD 12.2-RELEASE-p3:
```
Mar 12 14:13:32 nexus kernel: [drm] radeon kernel modesetting enabled.
Mar 12 14:13:32 nexus kernel: drmn0: <drmn> on vgapci0
Mar 12 14:13:32 nexus kernel: vgapci0: child drmn0 requested pci_enable_io
Mar 12 14:13:32 nexus syslogd: last message repeated 1 times
Mar 12 14:13:32 nexus kernel: [drm] initializing kernel modesetting (SUMO 0x1002:0x964A 0x1849:0x9640 0x00).
Mar 12 14:13:32 nexus kernel: [drm] register mmio base: 0xFEB00000
Mar 12 14:13:32 nexus kernel: [drm] register mmio size: 262144
Mar 12 14:13:32 nexus kernel: [drm:radeon_device_init] Unable to find PCI I/O BAR
Mar 12 14:13:32 nexus kernel: [drm:radeon_atombios_init] Unable to find PCI I/O BAR; using MMIO for ATOM IIO
Mar 12 14:13:32 nexus kernel: ATOM BIOS: General
Mar 12 14:13:32 nexus kernel: drmn0: VRAM: 256M 0x0000000000000000 - 0x000000000FFFFFFF (256M used)
Mar 12 14:13:32 nexus kernel: drmn0: GTT: 1024M 0x0000000010000000 - 0x000000004FFFFFFF
Mar 12 14:13:32 nexus kernel: Failed to add WC MTRR for [0xc0000000-0xcfffffff]: -22; performance may suffer
Mar 12 14:13:32 nexus kernel: [drm] Detected VRAM RAM=256M, BAR=256M
Mar 12 14:13:32 nexus kernel: [drm] RAM width 32bits DDR
Mar 12 14:13:32 nexus kernel: [TTM] Zone  kernel: Available graphics memory: 4030824 kiB
Mar 12 14:13:32 nexus kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 kiB
Mar 12 14:13:32 nexus kernel: [TTM] Initializing pool allocator
Mar 12 14:13:32 nexus kernel: [drm] radeon: 256M of VRAM memory ready
Mar 12 14:13:32 nexus kernel: [drm] radeon: 1024M of GTT memory ready.
Mar 12 14:13:32 nexus kernel: [drm] Loading SUMO Microcode
Mar 12 14:13:32 nexus kernel: radeon/SUMO_pfp.bin: could not load firmware image, error 2
Mar 12 14:13:32 nexus syslogd: last message repeated 2 times
Mar 12 14:13:32 nexus kernel: radeon/SUMO_me.bin: could not load firmware image, error 2
Mar 12 14:13:32 nexus syslogd: last message repeated 2 times
Mar 12 14:13:32 nexus kernel: radeon/SUMO_rlc.bin: could not load firmware image, error 2
Mar 12 14:13:32 nexus syslogd: last message repeated 2 times
Mar 12 14:13:32 nexus kernel: [drm] Internal thermal controller without fan control
Mar 12 14:13:32 nexus kernel: [drm] Found smc ucode version: 0x00011200
Mar 12 14:13:32 nexus kernel: [drm] radeon: dpm initialized
Mar 12 14:13:32 nexus kernel: radeon/SUMO_uvd.bin: could not load firmware image, error 2
Mar 12 14:13:32 nexus syslogd: last message repeated 2 times
Mar 12 14:13:32 nexus kernel: [drm] GART: num cpu pages 262144, num gpu pages 262144
Mar 12 14:13:32 nexus kernel: [drm] PCIE GART of 1024M enabled (table at 0x0000000000162000).
Mar 12 14:13:32 nexus kernel: drmn0: WB enabled
Mar 12 14:13:32 nexus kernel: drmn0: fence driver on ring 0 use gpu addr 0x0000000010000c00 and cpu addr 0x0xfffff800152e1c00
Mar 12 14:13:32 nexus kernel: drmn0: fence driver on ring 3 use gpu addr 0x0000000010000c0c and cpu addr 0x0xfffff800152e1c0c
Mar 12 14:13:32 nexus kernel: drmn0: fence driver on ring 5 use gpu addr 0x0000000000072118 and cpu addr 0x0xfffff800c0072118
Mar 12 14:13:32 nexus kernel: [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
Mar 12 14:13:32 nexus kernel: [drm] Driver supports precise vblank timestamp query.
Mar 12 14:13:32 nexus kernel: drmn0: radeon: MSI limited to 32-bit
Mar 12 14:13:32 nexus kernel: [drm] radeon: irq initialized.
Mar 12 14:13:32 nexus kernel: [drm] ring test on 0 succeeded in 1 usecs
Mar 12 14:13:32 nexus kernel: [drm] ring test on 3 succeeded in 3 usecs
Mar 12 14:13:32 nexus kernel: [drm] ring test on 5 succeeded in 1 usecs
Mar 12 14:13:32 nexus kernel: [drm] UVD initialized successfully.
Mar 12 14:13:32 nexus kernel: [drm] ib test on ring 0 succeeded in 0 usecs
Mar 12 14:13:32 nexus kernel: [drm] ib test on ring 3 succeeded in 0 usecs
Mar 12 14:13:32 nexus kernel: [drm] ib test on ring 5 succeeded
Mar 12 14:13:32 nexus kernel: [drm] hw_i2c forced on, you may experience display detection problems!
Mar 12 14:13:32 nexus kernel: [drm] Connector DVI-D-1: get mode from tunables:
Mar 12 14:13:32 nexus kernel: [drm]   - kern.vt.fb.modes.DVI-D-1
Mar 12 14:13:32 nexus kernel: [drm]   - kern.vt.fb.default_mode
Mar 12 14:13:32 nexus kernel: [drm] Connector DVI-D-2: get mode from tunables:
Mar 12 14:13:32 nexus kernel: [drm]   - kern.vt.fb.modes.DVI-D-2
Mar 12 14:13:32 nexus kernel: [drm]   - kern.vt.fb.default_mode
Mar 12 14:13:32 nexus kernel: [drm] Radeon Display Connectors
Mar 12 14:13:32 nexus kernel: [drm] Connector 0:
Mar 12 14:13:32 nexus kernel: [drm]   DVI-D-1
Mar 12 14:13:32 nexus kernel: [drm]   HPD2
Mar 12 14:13:32 nexus kernel: [drm]   DDC: 0x6440 0x6440 0x6444 0x6444 0x6448 0x6448 0x644c 0x644c
Mar 12 14:13:32 nexus kernel: [drm]   Encoders:
Mar 12 14:13:32 nexus kernel: [drm]     DFP2: INTERNAL_UNIPHY2
Mar 12 14:13:32 nexus kernel: [drm] Connector 1:
Mar 12 14:13:32 nexus kernel: [drm]   DVI-D-2
Mar 12 14:13:32 nexus kernel: [drm]   HPD1
Mar 12 14:13:32 nexus kernel: [drm]   DDC: 0x6430 0x6430 0x6434 0x6434 0x6438 0x6438 0x643c 0x643c
Mar 12 14:13:32 nexus kernel: [drm]   Encoders:
Mar 12 14:13:32 nexus kernel: [drm]     DFP1: INTERNAL_UNIPHY2
Mar 12 14:13:32 nexus kernel: [drm] fb mappable at 0xC0366000
Mar 12 14:13:32 nexus kernel: [drm] vram apper at 0xC0000000
Mar 12 14:13:32 nexus kernel: [drm] size 7299072
Mar 12 14:13:32 nexus kernel: [drm] fb depth is 24
Mar 12 14:13:32 nexus kernel: [drm]    pitch is 6912
Mar 12 14:13:32 nexus kernel: VT: Replacing driver "vga" with new "fb".
```
It's interesting that it always complains about not being able to load firmware…

evadot, 2021-03-12:
Could you before loading the module do : 
kenv hw.dri.drm_debug=0xffffffff
And post the dmesg, it might contain more info.
Same thing with 12.2 but replace the hw.dri.drm_debug by dev.drm.drm_debug

Thanks.

Zirias, 2021-03-12:
Well, this didn't give any more output as far as I could see, but it made me *yet another time* look very closely at the logs, and this time I noticed this peculiarity from the log of the old, working version:
`[drm] hw_i2c forced on, you may experience display detection problems!`

So, I tried setting `hw.radeonkms.hw_i2c=1` and ... **it works!** – the option warning about possible display detection problems makes display detection work here…

Now, is this still considered a bug? I'm fine with just closing it (and pretty happy).

evadot, 2021-04-19:
We default to hw_i2c=0 in https://github.com/freebsd/drm-kmod/commit/2e424b6df2cc89fa3db3709e397a35478ce9671b
So there seems to be case where 1 is better and some others when 0 is better ...
Nothing much I can do about that, especially since radeon isn't really maintained either here or upstream.

closed 2021-04-19 by evadot