freebsd/drm-kmod #67: [i915kms] broken suspend/resume on MacbookPro15/13.0-RC3

huanghwh, 2021-03-22:
After upgrade to 13.0-RC3 from 12.2-RELEASE, suspend/resume is not working any more:

% pkg info -x drm
drm-fbsd13-kmod-5.4.92.g20210202
drm-kmod-g20190710_1
libdrm-2.4.104,1

before suspend:
#dmesg|grep drm
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
drmn0: Found 128MB of eDRAM
[drm] Got stolen memory base 0x79a00000, size 0x6000000
[drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[drm] Driver supports precise vblank timestamp query.
[drm] Connector eDP-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.eDP-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector DP-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.DP-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector DP-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.DP-2
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-2
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-3: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-3
[drm]   - kern.vt.fb.default_mode
[drm] Initialized i915 1.6.0 20190822 for drmn0 on minor 0
name=drmn0 flags=0x0 stride=11776 bpp=32
drmn0: fb0: i915drmfb frame buffer device

#  glxinfo | grep OpenGL
OpenGL vendor string: Intel Open Source Technology Center
OpenGL renderer string: Mesa DRI Intel(R) Iris(R) Pro Graphics P5200 (HSW GT3)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 20.2.3
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.0 Mesa 20.2.3
OpenGL shading language version string: 1.30
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.1 Mesa 20.2.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
OpenGL ES profile extensions:

after resume:
#dmesg|grep drm
% dmesg | grep drm
drmn0: <drmn> on vgapci0
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
[drm] Unable to create a private tmpfs mount, hugepage support will be disabled(-19).
drmn0: Found 128MB of eDRAM
[drm] Got stolen memory base 0x79a00000, size 0x6000000
[drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[drm] Driver supports precise vblank timestamp query.
[drm] Connector eDP-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.eDP-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector DP-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.DP-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-1: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-1
[drm]   - kern.vt.fb.default_mode
[drm] Connector DP-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.DP-2
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-2: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-2
[drm]   - kern.vt.fb.default_mode
[drm] Connector HDMI-A-3: get mode from tunables:
[drm]   - kern.vt.fb.modes.HDMI-A-3
[drm]   - kern.vt.fb.default_mode
[drm] Initialized i915 1.6.0 20190822 for drmn0 on minor 0
name=drmn0 flags=0x0 stride=11776 bpp=32
drmn0: fb0: i915drmfb frame buffer device
vgapci0: child drmn0 requested pci_set_powerstate
vgapci0: child drmn0 requested pci_set_powerstate
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
drmn0: Failed to idle engines, declaring wedged!

#glxinfo | grep OpenGL
OpenGL vendor string: Mesa/X.org
OpenGL renderer string: llvmpipe (LLVM 10.0.1, 256 bits)
OpenGL core profile version string: 4.5 (Core Profile) Mesa 20.2.3
OpenGL core profile shading language version string: 4.50
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile
OpenGL core profile extensions:
OpenGL version string: 3.1 Mesa 20.2.3
OpenGL shading language version string: 1.40
OpenGL context flags: (none)
OpenGL extensions:
OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.2.3
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20
OpenGL ES profile extensions:

grahamperrin, 2021-07-03:
Reproducible with 13.0-RELEASE-p3?

eaosfu, 2021-09-28:
@grahamperrin 

I think I'm running into a similar issue. I'm running on a Thinkpad T450. I see the same 'wedged' message after suspend/resume.

It's difficult to reproduce since I can get through several such cycles before I encounter the error.

I was encountering the bug on FreeBSD 13.0-RELEASE-p2, recently upgraded to 13.0-RELEASE-p4 and still encounter the problem:

Sep 27 13:55:45  acpi[3602]: resumed at 20210927 13:55:45
Sep 27 13:55:45  kernel: ugen0.2: <vendor 0x8087 product 0x0a2a> at usbus0
Sep 27 13:55:45  kernel: ubt0 on uhub0
Sep 27 13:55:45  kernel: ubt0: <vendor 0x8087 product 0x0a2a, class 224/1, rev 2.01/0.01, addr 1> on usbus0
Sep 27 13:55:46  kernel: ugen0.3: <J8AFCNEGC Integrated Camera> at usbus0
Sep 27 13:55:47  kernel: uhub1: 3 ports with 3 removable, self powered
Sep 27 13:55:47  kernel: ugen1.2: <vendor 0x8087 product 0x8001> at usbus1
Sep 27 13:55:47  kernel: uhub2 on uhub1
Sep 27 13:55:47  kernel: uhub2: <vendor 0x8087 product 0x8001, class 9/0, rev 2.00/0.03, addr 2> on usbus1
Sep 27 13:55:50  kernel: uhub2: 8 ports with 8 removable, self powered
Sep 27 13:55:50  kernel: ubt0: ubt_ctrl_write_callback:906: control transfer failed: USB_ERR_TIMEOUT
Sep 27 13:55:50  kernel: ng_hci_process_command_timeout: ubt0hci - unable to complete HCI command OGF=0x3, OCF=0x3. Timeout
Sep 27 13:55:51  root[3640]: /etc/rc.d/bluetooth: ERROR: Unable to setup Bluetooth stack for device ubt0
Sep 27 13:56:00  acpi[3681]: suspend at 20210927 13:56:00
Sep 27 22:23:05  kernel: uhub1: at usbus1, port 1, addr 1 (disconnected)
Sep 27 22:23:05  kernel: ugen1.2: <vendor 0x8087 product 0x8001> at usbus1 (disconnected)
Sep 27 22:23:05  kernel: uhub2: at uhub1, port 1, addr 2 (disconnected)
Sep 27 22:23:05  kernel: uhub2: detached
Sep 27 22:23:05  kernel: uhub1: detached
Sep 27 22:23:05  kernel: rtsx0: Suspend
Sep 27 22:23:05  kernel: uhub0: at usbus0, port 1, addr 1 (disconnected)
Sep 27 22:23:05  kernel: ugen0.2: <vendor 0x8087 product 0x0a2a> at usbus0 (disconnected)
Sep 27 22:23:05  kernel: ubt0: at uhub0, port 7, addr 1 (disconnected)
Sep 27 22:23:05  kernel: ubt0: detached
Sep 27 22:23:05  kernel: ugen0.3: <J8AFCNEGC Integrated Camera> at usbus0 (disconnected)
Sep 27 22:23:05  kernel: uhub0: detached
Sep 27 22:23:05  kernel: vgapci0: child drmn0 requested pci_set_powerstate
Sep 27 22:23:05  kernel: i915 raw-wakerefs=3 wakelocks=3 on cleanup
Sep 27 22:23:05  kernel: pci0: failed to set ACPI power state D3 on \_SB_.PCI0.VID_: AE_BAD_PARAMETER
Sep 27 22:23:05  kernel: pcib0: failed to set ACPI power state D2 on \_SB_.PCI0: AE_BAD_PARAMETER
Sep 27 22:23:05  kernel: acpi0: cleared fixed power button status
Sep 27 22:23:05  kernel: vgapci0: child drmn0 requested pci_set_powerstate
Sep 27 22:23:05  kernel: vgapci0: child drmn0 requested pci_enable_io
Sep 27 22:23:05  syslogd: last message repeated 1 times
Sep 27 22:23:05  kernel: drmn0: Failed to idle engines, declaring wedged!

eaosfu, 2021-10-01:
This happens more frequently under 13.0-RELEASE-p4 than it did under p2

evadot, 2022-06-06:
13.0 only support 5.4 max and this is known to have problems with suspend/resume.
Please upgrade to 13.1 and 5.10

closed 2022-06-06 by evadot