freebsd/drm-kmod #68: amdgpu navi10, panic on 13-STABLE or 14-CURRENT

korasuku, 2021-04-12:
**Describe the bug**

I'm using a laptop, connected to a Thunderbolt 3 enclosure, with an AMD navi10 GPU. Laptop contains an Intel IGP (Kaby Lake), but as I use it in clamshell while connected to TB enclosure, I only kldload amdgpu, framebuffer swap to the external monitor (connected via DP directly to the navi10 card) leaving integrated panel orphan. So far so good ! :)

Issue raise a few seconds to minutes after loading the driver:

- If I just use my terminal as usual, let's say launching a "make install clean" in the ports tree, screen will goes blank at some point (randomly within a two minutes) preventing me to see of the panic output, it will reboot accordingly to sysctl parameters in case of panic.
- If I launch a "high throughput command" like "sysctl -a", panic is instant (see attached pictures), followed by a few seconds from panic to screen going also blank (for what it's worth signal is still sent to monitor as DPMS is not triggered)

I'm posting with information from -CURRENT but had the same with 13-RC* or -STABLE.

**FreeBSD version**

FreeBSD gorthaur.int.7c5.fr 14.0-CURRENT FreeBSD 14.0-CURRENT #0: Mon Apr 12 03:37:29 CEST 2021     zarock@gorthaur.int.7c5.fr:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
root@gorthaur:/home/zarock # pciconf -lv
hostb0@pci0:0:0:0:	class=0x060000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x3e34 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Coffee Lake HOST and DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x3ea0 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'WhiskeyLake-U GT2 [UHD Graphics 620]'
    class      = display
    subclass   = VGA
none0@pci0:0:4:0:	class=0x118000 rev=0x0c hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
    class      = dasp
none1@pci0:0:8:0:	class=0x088000 rev=0x00 hdr=0x00 vendor=0x8086 device=0x1911 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model'
    class      = base peripheral
pchtherm0@pci0:0:18:0:	class=0x118000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df9 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Thermal Controller'
    class      = dasp
xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9ded subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP USB 3.1 xHCI Controller'
    class      = serial bus
    subclass   = USB
none2@pci0:0:20:2:	class=0x050000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9def subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP Shared SRAM'
    class      = memory
    subclass   = RAM
iwm0@pci0:0:20:3:	class=0x028000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df0 subvendor=0x8086 subdevice=0x0034
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP CNVi [Wireless-AC]'
    class      = network
none3@pci0:0:22:0:	class=0x078000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de0 subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP MEI Controller'
    class      = simple comms
pcib1@pci0:0:28:0:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9db8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:28:4:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9dbc subvendor=0x17aa subdevice=0x2286
    vendor     = 'Intel Corporation'
    device     = 'Cannon Point-LP PCI Express Root Port'
    class      = bridge
[... 140 lines cut ...]
    vendor     = 'Intel Corporation'
    device     = 'JHL6240 Thunderbolt 3 USB 3.1 Controller (Low Power) [Alpine Ridge LP 2016]'
    class      = serial bus
    subclass   = USB
nvme0@pci0:61:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x15b7 device=0x5006 subvendor=0x15b7 subdevice=0x5006
    vendor     = 'Sandisk Corp'
    device     = 'WD Black SN750 / PC SN730 NVMe SSD'
    class      = mass storage
    subclass   = NVM

</details>

**DRM KMOD version**
drm-current-kmod 5.4.92.g20210128_2
drm-kmod g20190710_1

**To Reproduce**
kldload amdgpu and wait a few seconds

**Screenshots**
[image]

**Additional context**

- I'm not sure if "lag" implied by Thunderbolt 3 link can create this kind of problem, wondering because of different behaviours between high throughput and quiet command, still, in the end, panic will happens. I tried to play with pci lane cap sysctl for examples, but no change at all :)

korasuku, 2021-04-12:
Extra data gathered from ssh after kenv hw.dri.drm_debug=0xffffffff then kldload amdgpu

```
Apr 12 14:28:40 gorthaur login[91815]: ROOT LOGIN (root) ON ttyv0
Apr 12 14:28:47 gorthaur kernel: [drm] amdgpu kernel modesetting enabled.
Apr 12 14:28:47 gorthaur kernel: drmn1: <drmn> on vgapci1
Apr 12 14:28:47 gorthaur kernel: vgapci1: child drmn1 requested pci_enable_io
Apr 12 14:28:47 gorthaur syslogd: last message repeated 1 times
Apr 12 14:28:47 gorthaur kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
Apr 12 14:28:47 gorthaur kernel: [drm] initializing kernel modesetting (NAVI10 0x1002:0x731F 0x1043:0x04E2 0xC1).
Apr 12 14:28:47 gorthaur kernel: [drm] register mmio base: 0xD2B00000
Apr 12 14:28:47 gorthaur kernel: [drm] register mmio size: 524288
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for ATHUB
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for CLKA
Apr 12 14:28:48 gorthaur syslogd: last message repeated 4 times
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for DF
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for DMU
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for GC
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for HDP
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for MMHUB
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for MP0
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for MP1
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for NBIF
Apr 12 14:28:48 gorthaur syslogd: last message repeated 1 times
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for OSSSYS
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for SDMA0
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for SDMA1
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for SMUIO
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for THM
Apr 12 14:28:48 gorthaur kernel: [drm] set register base offset for UVD
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 0 <nv_common>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 1 <gmc_v10_0>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 2 <navi10_ih>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 3 <psp>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 4 <smu>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 5 <dm>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 6 <gfx_v10_0>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 7 <sdma_v5_0>
Apr 12 14:28:48 gorthaur kernel: [drm] add ip block number 8 <vcn_v2_0>
Apr 12 14:28:48 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_gpu_info.bin'
Apr 12 14:28:48 gorthaur kernel: ATOM BIOS: 115-D182PI0-100
Apr 12 14:28:48 gorthaur kernel: [drm] VCN decode is enabled in VM mode
Apr 12 14:28:48 gorthaur kernel: [drm] VCN encode is enabled in VM mode
Apr 12 14:28:48 gorthaur kernel: [drm] VCN jpeg decode is enabled in VM mode
Apr 12 14:28:48 gorthaur kernel: [drm] GPU posting now...
Apr 12 14:28:48 gorthaur kernel: [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
Apr 12 14:28:48 gorthaur kernel: drmn1: VRAM: 8176M 0x0000008000000000 - 0x00000081FEFFFFFF (8176M used)
Apr 12 14:28:48 gorthaur kernel: drmn1: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
Apr 12 14:28:48 gorthaur kernel: Failed to add WC MTRR for [0xa0000000-0xafffffff]: -22; performance may suffer
Apr 12 14:28:48 gorthaur kernel: [drm] Detected VRAM RAM=8176M, BAR=256M
Apr 12 14:28:48 gorthaur kernel: [drm] RAM width 256bits GDDR6
Apr 12 14:28:48 gorthaur kernel: [TTM] Zone  kernel: Available graphics memory: 8268990 KiB
Apr 12 14:28:48 gorthaur kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 KiB
Apr 12 14:28:48 gorthaur kernel: [TTM] Initializing pool allocator
Apr 12 14:28:48 gorthaur kernel: [drm] amdgpu: 8176M of VRAM memory ready
Apr 12 14:28:48 gorthaur kernel: [drm] amdgpu: 8176M of GTT memory ready.
Apr 12 14:28:48 gorthaur kernel: [drm] GART: num cpu pages 131072, num gpu pages 131072
Apr 12 14:28:48 gorthaur kernel: [drm] PCIE GART of 512M enabled (table at 0x0000008000000000).
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_sos.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_asd.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_smc.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_pfp.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_me.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_ce.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_rlc.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_mec.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_mec2.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_sdma.bin'
Apr 12 14:28:50 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_sdma1.bin'
Apr 12 14:28:50 gorthaur kernel: [drm] use_doorbell being set to: [true]
[... 71 lines cut ...]
[... cut ...]
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 5 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 6 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 7 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 8 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 9 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 10 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 11 succeeded
Apr 12 14:28:52 gorthaur kernel: [drm] ib test on ring 12 succeeded
```

korasuku, 2021-04-12:
OK, so after some more tests:

- My first guess about console throughput is wrong: in fact, just doing a sysctl -a make the system panic (can't figure out on which oid yet), tested via SSH.
- At full console idle crash will occurs later,
- While using the console, when the monitor goes blank after a few minutes, I can't to have a dump, but referring to debugging tips in https://wiki.freebsd.org/Graphics; I have these lines thanks to dev.drm.drm_debug=-1, from module loading to the first syslog line after reboot.

<details>
  <summary>message.log from load to reboot with a few cut</summary>
Apr 12 21:34:59 gorthaur kernel: [drm] amdgpu kernel modesetting enabled.
Apr 12 21:34:59 gorthaur kernel: drmn1: <drmn> on vgapci1
Apr 12 21:34:59 gorthaur kernel: vgapci1: child drmn1 requested pci_enable_io
Apr 12 21:34:59 gorthaur syslogd: last message repeated 1 times
Apr 12 21:34:59 gorthaur kernel: sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
Apr 12 21:34:59 gorthaur kernel: [drm] initializing kernel modesetting (NAVI10 0x1002:0x731F 0x1043:0x04E2 0xC1).
Apr 12 21:34:59 gorthaur kernel: [drm] register mmio base: 0xD2B00000
Apr 12 21:34:59 gorthaur kernel: [drm] register mmio size: 524288
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for ATHUB
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for CLKA
Apr 12 21:34:59 gorthaur syslogd: last message repeated 4 times
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for DF
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for DMU
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for GC
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for HDP
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for MMHUB
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for MP0
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for MP1
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for NBIF
Apr 12 21:34:59 gorthaur syslogd: last message repeated 1 times
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for OSSSYS
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for SDMA0
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for SDMA1
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for SMUIO
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for THM
Apr 12 21:34:59 gorthaur kernel: [drm] set register base offset for UVD
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 0 <nv_common>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 1 <gmc_v10_0>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 2 <navi10_ih>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 3 <psp>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 4 <smu>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 5 <dm>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 6 <gfx_v10_0>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 7 <sdma_v5_0>
Apr 12 21:34:59 gorthaur kernel: [drm] add ip block number 8 <vcn_v2_0>
Apr 12 21:34:59 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_gpu_info.bin'
Apr 12 21:35:00 gorthaur kernel: ATOM BIOS: 115-D182PI0-100
Apr 12 21:35:00 gorthaur kernel: [drm] VCN decode is enabled in VM mode
Apr 12 21:35:00 gorthaur kernel: [drm] VCN encode is enabled in VM mode
Apr 12 21:35:00 gorthaur kernel: [drm] VCN jpeg decode is enabled in VM mode
Apr 12 21:35:00 gorthaur kernel: [drm] GPU posting now...
Apr 12 21:35:00 gorthaur kernel: [drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
Apr 12 21:35:00 gorthaur kernel: drmn1: VRAM: 8176M 0x0000008000000000 - 0x00000081FEFFFFFF (8176M used)
Apr 12 21:35:00 gorthaur kernel: drmn1: GART: 512M 0x0000000000000000 - 0x000000001FFFFFFF
Apr 12 21:35:00 gorthaur kernel: Failed to add WC MTRR for [0xa0000000-0xafffffff]: -22; performance may suffer
Apr 12 21:35:00 gorthaur kernel: [drm] Detected VRAM RAM=8176M, BAR=256M
Apr 12 21:35:00 gorthaur kernel: [drm] RAM width 256bits GDDR6
Apr 12 21:35:00 gorthaur kernel: [TTM] Zone  kernel: Available graphics memory: 8268990 KiB
Apr 12 21:35:00 gorthaur kernel: [TTM] Zone   dma32: Available graphics memory: 2097152 KiB
Apr 12 21:35:00 gorthaur kernel: [TTM] Initializing pool allocator
Apr 12 21:35:00 gorthaur kernel: [drm] amdgpu: 8176M of VRAM memory ready
Apr 12 21:35:00 gorthaur kernel: [drm] amdgpu: 8176M of GTT memory ready.
Apr 12 21:35:00 gorthaur kernel: [drm] GART: num cpu pages 131072, num gpu pages 131072
Apr 12 21:35:00 gorthaur kernel: [drm] PCIE GART of 512M enabled (table at 0x0000008000000000).
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_sos.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_asd.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_smc.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_pfp.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_me.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_ce.bin'
Apr 12 21:35:01 gorthaur kernel: drmn1: successfully loaded firmware image 'amdgpu/navi10_rlc.bin'
[... cut ...]
Apr 12 21:35:50 gorthaur kernel: [drm:drm_calc_timestamping_constants] crtc 62: hwmode: htotal 4040, vtotal 1235, vdisplay 1200
Apr 12 21:35:50 gorthaur kernel: [drm:drm_calc_timestamping_constants] crtc 62: clock 718470 kHz framedur 6944479 linedur 5623
Apr 12 21:35:50 gorthaur kernel: [drm:amdgpu_dm_atomic_commit_tail] amdgpu_crtc id:0 crtc_state_flags: enable:1, active:1, planes_changed:1, mode_changed:0,active_changed:0,connectors_changed:0
Apr 12 21:35:50 gorthaur kernel: [drm:amdgpu_dm_atomic_

korasuku, 2021-04-14:
I took another chance tonight, with repo mentioned in #45 
- If using console, it will goes blank and hang.
- If launching a compositor (for example Sway), I can work without issue ! (whereas it was going to blank with the release in ports)
- If I launching an infamous sysctl -a, it will crash but this time I'll have more messages, don't know if they can be of any help of if they are related to the fact that panic in ongoing :)

```
Fatal trap 12: page fault while in kernel mode
cpuid = 4; apic id = 04
fault virtual address	= 0x0
fault code		= supervisor read instruction, page not present
instruction pointer	= 0x20:0x0
stack pointer	        = 0x28:0xfffffe01242c7728
frame pointer	        = 0x28:0xfffffe01242c7740
code segment		= base 0x0, limit 0xfffff, type 0x1b
			= DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags	= interrupt enabled, resume, IOPL = 0
current process		= 79772 (sysctl)
trap number		= 12
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /usr/home/zarock/Code/repos/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:621
#0 0xffffffff80e6dc93 at linux_dump_stack+0x23
#1 0xffffffff83d273b3 at drm_atomic_helper_check_modeset+0xb3
#2 0xffffffff83b14b5f at amdgpu_dm_atomic_check+0x2f
#3 0xffffffff83d2635f at drm_atomic_check_only+0x3ff
#4 0xffffffff83d26783 at drm_atomic_commit+0x13
#5 0xffffffff83d334f8 at drm_client_modeset_commit_atomic+0x148
#6 0xffffffff83d33259 at drm_client_modeset_commit_force+0x69
#7 0xffffffff83d7460a at drm_fb_helper_restore_fbdev_mode_unlocked+0x7a
#8 0xffffffff83d6e096 at vt_kms_postswitch+0x166
#9 0xffffffff80a76d8b at vt_window_switch+0x12b
#10 0xffffffff80a73e2f at vtterm_cngrab+0x4f
#11 0xffffffff80bbd716 at cngrab+0x16
#12 0xffffffff80c251dc at vpanic+0xec
#13 0xffffffff80c250e3 at panic+0x43
#14 0xffffffff810c4317 at trap_fatal+0x387
#15 0xffffffff810c436f at trap_pfault+0x4f
#16 0xffffffff810c39cd at trap+0x27d
#17 0xffffffff81099d18 at calltrap+0x8
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /usr/home/zarock/Code/repos/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:621
#0 0xffffffff80e6dc93 at linux_dump_stack+0x23
#1 0xffffffff83d273b3 at drm_atomic_helper_check_modeset+0xb3
#2 0xffffffff83b14b5f at amdgpu_dm_atomic_check+0x2f
#3 0xffffffff83d2635f at drm_atomic_check_only+0x3ff
#4 0xffffffff83d26783 at drm_atomic_commit+0x13
#5 0xffffffff83d334f8 at drm_client_modeset_commit_atomic+0x148
#6 0xffffffff83d33259 at drm_client_modeset_commit_force+0x69
#7 0xffffffff83d7460a at drm_fb_helper_restore_fbdev_mode_unlocked+0x7a
#8 0xffffffff83d6e096 at vt_kms_postswitch+0x166
#9 0xffffffff80a76d8b at vt_window_switch+0x12b
#10 0xffffffff80a73e2f at vtterm_cngrab+0x4f
#11 0xffffffff80bbd716 at cngrab+0x16
#12 0xffffffff80c251dc at vpanic+0xec
#13 0xffffffff80c250e3 at panic+0x43
#14 0xffffffff810c4317 at trap_fatal+0x387
#15 0xffffffff810c436f at trap_pfault+0x4f
#16 0xffffffff810c39cd at trap+0x27d
#17 0xffffffff81099d18 at calltrap+0x8
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /usr/home/zarock/Code/repos/drm-kmod/drivers/gpu/drm/drm_atomic_helper.c:621
#0 0xffffffff80e6dc93 at linux_dump_stack+0x23
#1 0xffffffff83d273b3 at drm_atomic_helper_check_modeset+0xb3
#2 0xffffffff83b14b5f at amdgpu_dm_atomic_check+0x2f
#3 0xffffffff83d2635f at drm_atomic_check_only+0x3ff
#4 0xffffffff83d26783 at drm_atomic_commit+0x13
#5 0xffffffff83d334f8 at drm_client_modeset_commit_atomic+0x148
#6 0xffffffff83d33259 at drm_client_modeset_commit_force+0x69
#7 0xffffffff83d7460a at drm_fb_helper_restore_fbdev_mode_unlocked+0x7a
#8 0xffffffff83d6e096 at vt_kms_postswitch+0x166
#9 0xffffffff80a76d8b at vt_window_switch+0x12b
#10 0xffffffff80a73e2f at vtterm_cngrab+0x4f
#11 0xffffffff80bbd716 at cngrab+0x16
#12 0xffffffff80c251dc at vpanic+0xec
[... 366 lines cut ...]
#17 0xffffffff83a14e50 at sysctl_handle_attr+0x70
kernel trap 22 with interrupts disabled
 panic: page fault
cpuid = 4
time = 1618432617
KDB: stack backtrace:
db_trace_self_wrapper() at db_trace_self_wrapper+0x2b/frame 0xfffffe01242c73d0
vpanic() at vpanic+0x181/frame 0xfffffe01242c7420
panic() at panic+0x43/frame 0xfffffe01242c7480
trap_fatal() at trap_fatal+0x387/frame 0xfffffe01242c74e0
trap_pfault() at trap_pfault+0x4f/frame 0xfffffe01242c7540
trap() at trap+0x27d/frame 0xfffffe01242c7650
calltrap() at calltrap+0x8/frame 0xfffffe01242c7650
--- trap 0xc, rip = 0, rsp = 0xfffffe01242c7728, rbp = 0xfffffe01242c7740 ---
??() at 0/frame 0xfffffe01242c7740
sysctl_handle_attr() at sysctl_handle_attr+0x70/frame 0xfffffe01242c7790
sysctl_root_handler_locked() at sysctl_root_handler_locked+0x91/frame 0xfffffe01242c77e0
sysctl_root() at sysctl_root+0x2d0/frame 0xfffffe01242c7860
userland_sysctl() at userland_sysctl+0x173/frame 0xfffffe01242c7910
sys___sysctl() at sys___sysctl+0x5f/frame 0xfffffe01242c79c0
amd64_syscall() at amd64_syscall+0x10c/frame 0xfffffe01242c7af0
fast_syscall_common() at fast_syscall_common+0xf8/frame 0xfffffe01242c7af0
--- syscall (202, FreeBSD ELF64, sys___sysctl), rip = 0x1020e0fa15ba, rsp = 0x7fffff13d098, rbp = 0x7fffff13d0d0 ---
KDB: enter: panic
```

mateuszkwiatkowski, 2021-04-19:
I can also trigger panic with `sysctl -a` on 13.0-RELEASE and drm-kmod installed with `pkg`.
Packages:
```
$ pkg info | grep amdgpu
xf86-video-amdgpu-19.1.0_1     X.Org amdgpu display driver
$ pkg info | grep drm
drm-fbsd13-kmod-5.4.92.g20210202 DRM modules for the linuxkpi-based KMS components
drm-kmod-g20190710_1           Metaport of DRM modules for the linuxkpi-based KMS components
libdrm-2.4.104,1               Userspace interface to kernel Direct Rendering Module services
linux-c7-libdrm-2.4.97         Interface to kernel Direct Rendering Module (Linux CentOS 7.9.2009)
```

GPU:
```
vgapci0@pci0:43:0:0:    class=0x030000 rev=0xca hdr=0x00 vendor=0x1002 device=0x731f subvendor=0x1462 subdevice=0x381e
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Navi 10 [Radeon RX 5600 OEM/5600 XT / 5700/5700 XT]'
    class      = display
    subclass   = VGA
```

evadot, 2021-04-19:
Should be fixed by :
https://cgit.freebsd.org/ports/commit/?h=2021Q2&id=cdbfe8c1c9831078e142376de877ab7bab1a3a00
and
https://cgit.freebsd.org/ports/commit/?id=7c1c47427fb765b9856e46cf4eed5dd84c32ec2d

Sorry that I forgot to update the ports.
Please let me know if you still have issue after updating the packages/ports.

evadot, 2021-04-19:
Duplicate of https://github.com/freebsd/drm-kmod/issues/15 so closing this one.

closed 2021-04-19 by evadot