freebsd/drm-kmod #69: kernel panic from linux_dump_stack() caused by drm_atomic_helper.c:621

sbz, 2021-04-16:
**Describe the bug**
kernel panic from _linux_dump_stack()_ caused by [drm_atomic_helper.c:621](https://github.com/freebsd/drm-kmod/blob/master/drivers/gpu/drm/drm_atomic_helper.c#L621) while I was running Firefox and compiling kernel sources.

relevant ports source path _graphics/drm-fbsd13-kmod/work/drm-kmod-drm_v5.4.92_2/drivers/gpu/drm/drm_atomic_helper.c:621_

```
Fatal trap 9: general protection fault while in kernel mode
cpuid = 2; apic id = 04
instruction pointer	= 0x20:0xffffffff81075b79
stack pointer	        = 0x28:0xfffffe00d2d63a10
frame pointer	        = 0x28:0xfffffe00d2d63b60
code segment		= base 0x0, limit 0xfffff, type 0x1b
			= DPL 0, pres 1, long 1, def32 0, gran 1
processor eflags	= interrupt enabled, resume, IOPL = 0
current process		= 15413 (c++)
trap number		= 9
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /wrkdirs/usr/ports/graphics/drm-fbsd13-kmod/work/drm-kmod-drm_v5.4.92_2/drivers/gpu/drm/drm_atomic_helper.c:621
#0 0xffffffff80e3acc3 at linux_dump_stack+0x23
#1 0xffffffff82a863c3 at drm_atomic_helper_check_modeset+0xb3
#2 0xffffffff8297eb8d at intel_atomic_check+0x8d
#3 0xffffffff82a85360 at drm_atomic_check_only+0x400
#4 0xffffffff82a85793 at drm_atomic_commit+0x13
#5 0xffffffff82a923b8 at drm_client_modeset_commit_atomic+0x148
#6 0xffffffff82a92119 at drm_client_modeset_commit_force+0x69
#7 0xffffffff82ad253a at drm_fb_helper_restore_fbdev_mode_unlocked+0x7a
#8 0xffffffff82acc527 at vt_kms_postswitch+0x167
#9 0xffffffff80a622cb at vt_window_switch+0x12b
#10 0xffffffff80a5f36f at vtterm_cngrab+0x4f
#11 0xffffffff80ba06d6 at cngrab+0x16
#12 0xffffffff80c05d6e at vpanic+0xee
#13 0xffffffff80c05c73 at panic+0x43
#14 0xffffffff810850c7 at trap_fatal+0x387
#15 0xffffffff81084577 at trap+0x67
#16 0xffffffff8105b8d8 at calltrap+0x8
#17 0xffffffff80f26a9f at vmspace_exit+0x7f
WARNING !drm_modeset_is_locked(&crtc->mutex) failed at /wrkdirs/usr/ports/graphics/drm-fbsd13-kmod/work/drm-kmod-drm_v5.4.92_2/drivers/gpu/drm/drm_atomic_helper.c:621
```

**FreeBSD version**

I'm running 13.0-STABLE on this laptop.

```
FreeBSD laptop 13.0-STABLE FreeBSD 13.0-STABLE #4 stable/13-659d71186: Wed Apr 14 00:01:48 IST 2021     sbz@laptop:/usr/obj/usr/src/amd64.amd64/sys/T410  amd64
```

**PCI Info**

Not a very recent Graphic controller

<details>
  <summary>pciconf -lv</summary>
  
```
vgapci0@pci0:0:2:0:	class=0x030000 rev=0x02 hdr=0x00 vendor=0x8086 device=0x0046 subvendor=0x17aa subdevice=0x215a
    vendor     = 'Intel Corporation'
    device     = 'Core Processor Integrated Graphics Controller'
    class      = display
    subclass   = VGA

```

</details>

**DRM KMOD version**

```
drm-fbsd13-kmod 5.4.92.g20210202
drm-kmod g20190710_1
```

**To Reproduce**
Unsure if it will be possible to reproduce, I was running Firefox and compiling kernel source at the same time and suddenly it raised the panic.

**Additional context**
Thinkpad T410 running with xorg server xorg-server-1.20.9_3,1 with intel kms I have `kld_list="i915kms"` in _rc.conf_

```
kldstat |grep i915
 4    1 0xffffffff82920000   158458 i915kms.ko
```

dmesg report Intel [Ironlake](https://en.wikichip.org/wiki/intel/hd_graphics_(ironlake)) 

```
agp0: <Intel Ironlake (M) SVGA controller> on vgapci0
drmn0: <drmn> on vgapci0
vgapci0: <VGA-compatible display> port 0x1800-0x1807 mem 0xf2000000-0xf23fffff,0xd0000000-0xdfffffff irq 16 at device 2.0 on pci1
vgapci0: child drmn0 requested pci_enable_io
drmn0: fb0: i915drmfb frame buffer device
```

evadot, 2021-04-19:
This is probably not the problem, this stacktrace is always printed when entering the debugger.
Please check the full trace for other clue.

sbz, 2021-04-19:
Here the full stacktrace when I enter in kgdb https://gist.github.com/sbz/3724ac5f4fe63d02c062e6f368af73d7

sbz, 2021-04-29:
FYI, I have updated to a recent **14.0-CURRENT** from a week ago with the new `drm-current-kmod` version, and the panic was not reproduced so far. 

```
14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-45aec4624: Wed Apr 21 23:36:48 IST 2021
```

```
drm-current-kmod-5.4.92.g20210419
```

I think that issue can be closed.

closed 2021-04-29 by sbz

ctipper, 2021-05-02:
I have exact same stack trace but amdgpu and trap number 12. Why was this issue closed?

ctipper, 2021-05-02:
+Root mount waiting for: usbus0
+ugen0.3: <Microsoft Microsoft Basic Optical Mouse v2.0> at usbus0
+[drm] amdgpu kernel modesetting enabled.
+drmn0: <drmn> on vgapci0
+vgapci0: child drmn0 requested pci_enable_io
+vgapci0: child drmn0 requested pci_enable_io
+sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
+[drm] initializing kernel modesetting (RAVEN 0x1002:0x15D8 0x1002:0x15D8 0xC9).
+[drm] register mmio base: 0xFCC00000
+[drm] register mmio size: 524288
+[drm] add ip block number 0 <soc15_common>
+[drm] add ip block number 1 <gmc_v9_0>
+[drm] add ip block number 2 <vega10_ih>
+[drm] add ip block number 3 <psp>
+[drm] add ip block number 4 <gfx_v9_0>
+[drm] add ip block number 5 <sdma_v4_0>
+[drm] add ip block number 6 <powerplay>
+[drm] add ip block number 7 <dm>
+[drm] add ip block number 8 <vcn_v1_0>
+drmn0: successfully loaded firmware image 'amdgpu/picasso_gpu_info.bin'
+[drm] BIOS signature incorrect 0 0
+ATOM BIOS: 113-PICASSO-118
+drmn0: successfully loaded firmware image 'amdgpu/picasso_sdma.bin'
+[drm] VCN decode is enabled in VM mode
+[drm] VCN encode is enabled in VM mode
+[drm] VCN jpeg decode is enabled in VM mode
+[drm] vm size is 262144 GB, 4 levels, block size is 9-bit, fragment size is 9-bit
+drmn0: VRAM: 2048M 0x000000F400000000 - 0x000000F47FFFFFFF (2048M used)
+drmn0: GART: 1024M 0x0000000000000000 - 0x000000003FFFFFFF
+drmn0: AGP: 267419648M 0x000000F800000000 - 0x0000FFFFFFFFFFFF
+Successfully added WC MTRR for [0xd0000000-0xdfffffff]: 0; 
+[drm] Detected VRAM RAM=2048M, BAR=256M
+[drm] RAM width 128bits DDR4
+[TTM] Zone  kernel: Available graphics memory: 7281118 KiB
+[TTM] Zone   dma32: Available graphics memory: 2097152 KiB
+[TTM] Initializing pool allocator
+[drm] amdgpu: 2048M of VRAM memory ready
+[drm] amdgpu: 3072M of GTT memory ready.
+[drm] GART: num cpu pages 262144, num gpu pages 262144
+[drm] PCIE GART of 1024M enabled (table at 0x000000F400900000).
+drmn0: successfully loaded firmware image 'amdgpu/picasso_asd.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_pfp.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_me.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_ce.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_rlc_am4.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_mec.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_mec2.bin'
+[drm] use_doorbell being set to: [true]
+amdgpu: [powerplay] hwmgr_sw_init smu backed is smu10_smu
+drmn0: successfully loaded firmware image 'amdgpu/raven_dmcu.bin'
+drmn0: successfully loaded firmware image 'amdgpu/picasso_vcn.bin'
+[drm] Found VCN firmware Version ENC: 1.12 DEC: 2 VEP: 0 Revision: 5
+[drm] PSP loading VCN firmware
+[drm] reserve 0x400000 from 0xf47f800000 for PSP TMR
+[drm] DM_PPLIB: values for F clock
+[drm] DM_PPLIB:	 1200000 in kHz, 4399 in mV
+[drm] DM_PPLIB: values for DCF clock
+[drm] DM_PPLIB:	 300000 in kHz, 3099 in mV
+[drm] DM_PPLIB:	 600000 in kHz, 3574 in mV
+[drm] DM_PPLIB:	 626000 in kHz, 4250 in mV
+[drm] DM_PPLIB:	 654000 in kHz, 4399 in mV
+[drm] Display Core initialized with v3.2.48!
+[drm] Connector HDMI-A-1: get mode from tunables:
+[drm]   - kern.vt.fb.modes.HDMI-A-1
+[drm]   - kern.vt.fb.default_mode
+[drm] SADs count is: -2, don't need to read it
+[drm] Connector HDMI-A-2: get mode from tunables:
+[drm]   - kern.vt.fb.modes.HDMI-A-2
+[drm]   - kern.vt.fb.default_mode
+[drm] Connector DP-1: get mode from tunables:
[... 332 lines cut ...]
+ panic: page fault
+cpuid = 1
+time = 1619884911
+KDB: stack backtrace:
+#0 0xffffffff80c57345 at kdb_backtrace+0x65
+#1 0xffffffff80c09d21 at vpanic+0x181
+#2 0xffffffff80c09b93 at panic+0x43
+#3 0xffffffff8108b187 at trap_fatal+0x387
+#4 0xffffffff8108b1df at trap_pfault+0x4f
+#5 0xffffffff8108a83d at trap+0x27d
+#6 0xffffffff810617a8 at calltrap+0x8
+#7 0xffffffff82153123 at zfs_acl_next_ace+0xb3
+#8 0xffffffff8215621b at zfs_zaccess_aces_check+0x13b
+#9 0xffffffff82155962 at zfs_zaccess+0xb2
+#10 0xffffffff822e4b48 at zfs_access+0x68
+#11 0xffffffff82166a68 at zfs_freebsd_access+0x48
+#12 0xffffffff80cfbd6a at vn_open_vnode+0x10a
+#13 0xffffffff80cfba02 at vn_open_cred+0x452
+#14 0xffffffff80cf18cd at kern_openat+0x25d
+#15 0xffffffff8108ba8c at amd64_syscall+0x10c
+#16 0xffffffff810620ce at fast_syscall_common+0xf8
+Uptime: 8m31s
+---<<BOOT>>---
+Copyright (c) 1992-2021 The FreeBSD Project.
+Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994

ctipper, 2021-05-02:
```
# uname -a
FreeBSD proteus 13.0-RELEASE FreeBSD 13.0-RELEASE #0 releng/13.0-n244733-ea31abc261f: Fri Apr  9 04:24:09 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

drm-fbsd13-kmod-5.4.92.g20210419
drm-kmod-g20190710_1
gpu-firmware-kmod-g20210330 
```

evadot, 2021-05-02:
This is a zfs panic, see the last lines :

+#7 0xffffffff82153123 at zfs_acl_next_ace+0xb3
+#8 0xffffffff8215621b at zfs_zaccess_aces_check+0x13b
+#9 0xffffffff82155962 at zfs_zaccess+0xb2
+#10 0xffffffff822e4b48 at zfs_access+0x68
+#11 0xffffffff82166a68 at zfs_freebsd_access+0x48
+#12 0xffffffff80cfbd6a at vn_open_vnode+0x10a
+#13 0xffffffff80cfba02 at vn_open_cred+0x452
+#14 0xffffffff80cf18cd at kern_openat+0x25d
+#15 0xffffffff8108ba8c at amd64_syscall+0x10c
+#16 0xffffffff810620ce at fast_syscall_common+0xf8

ctipper, 2021-05-02:
So I don't know how to report that it still seems to bail out while using Firefox as TR reported, same in my case. I've had two crashes so far, the one mailed (above) occurred on startup, the second during a desktop session. I'm way out of my depth I don't think I can use crash dumps with a generic kernel.