freebsd/drm-kmod #70: How to merge code from Linux

neelchauhan, 2021-04-22:
What is the process for merging code from Linux? I mean, which commands are used to merge and cherry pick commits from Linux to `drm-kmod`?

I'm interested in this mainly because I'm thinking about getting 5.5-wip working with i915kms (to support TigerLake), and am willing to do (at least some of) the dirty work myself. Maybe this will land into a PR.

evadot, 2021-04-23:
You have two scripts in the scripts directory.
drmgeneratepatch will help you generate needed patches to go from one linux version to one other.
drmpatch will help you applying those patches.

neelchauhan, 2021-04-26:
Thanks for letting me know. Sorry for the delayed response.

Closing this ticket.

closed 2021-04-26 by neelchauhan