freebsd/drm-kmod #72: amdgpu no hw acceleration on gnome3 ??

rut78HR, 2021-04-27:
Did anyone have a problem similar to mine with Gnome3. Namely, when starting, I have a big lag on loading that never goes away. If i switch betwen consoles only then is it seen that Gnome3 "works" . Other windows managers works OK (i try Mate/Xfce/Kde and i dont see any problem). HW acceleration works OK. Have tryed both 13-RELEASE and 14-current and same problem on both.

Is this mutter/gnome3 related problem or maye is something with drm-kmod-devel on amdgpu ??

rut@pila:~ % pciconf -lv
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d0 subvendor=0x1022 subdevice=0x15d0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15d1 subvendor=0x1022 subdevice=0x15d1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:7:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15d3 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 PCIe GPP Bridge [6:0]'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1452 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 00h-1fh) PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x15db subvendor=0x1022 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Internal PCIe GPP Bridge 0 to Bus A'
    class      = bridge
    subclass   = PCI-PCI
intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x61 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SMBus Controller'
    class      = serial bus
    subclass   = SMBus
isab0@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH LPC Bridge'
    class      = bridge
    subclass   = PCI-ISA
hostb3@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e8 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 0'
    class      = bridge
    subclass   = HOST-PCI
hostb4@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e9 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 1'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:24:2:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ea subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 2'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:24:3:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15eb subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 3'
    class      = bridge
    subclass   = HOST-PCI
hostb7@pci0:0:24:4:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ec subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 4'
    class      = bridge
    subclass   = HOST-PCI
hostb8@pci0:0:24:5:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ed subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 5'
    class      = bridge
    subclass   = HOST-PCI
hostb9@pci0:0:24:6:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ee subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 6'
    class      = bridge
    subclass   = HOST-PCI
hostb10@pci0:0:24:7:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15ef subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2 Device 24: Function 7'
    class      = bridge
    subclass   = HOST-PCI
iwm0@pci0:1:0:0:	class=0x028000 rev=0x93 hdr=0x00 vendor=0x8086 device=0x08b4 subvendor=0x8086 subdevice=0x8270
    vendor     = 'Intel Corporation'
    device     = 'Wireless 3160'
    class      = network
nvme0@pci0:2:0:0:	class=0x010802 rev=0x00 hdr=0x00 vendor=0x144d device=0xa808 subvendor=0x144d subdevice=0xa801
    vendor     = 'Samsung Electronics Co Ltd'
    device     = 'NVMe SSD Controller SM981/PM981/PM983'
    class      = mass storage
    subclass   = NVM
vgapci0@pci0:3:0:0:	class=0x030000 rev=0xc2 hdr=0x00 vendor=0x1002 device=0x15d8 subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Picasso'
    class      = display
    subclass   = VGA
hdac0@pci0:3:0:1:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0x15de subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Raven/Raven2/Fenghuang HDMI/DP Audio Controller'
    class      = multimedia
    subclass   = HDA
none1@pci0:3:0:2:	class=0x108000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15df subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) Platform Security Processor'
    class      = encrypt/decrypt
xhci0@pci0:3:0:3:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e0 subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven USB 3.1'
    class      = serial bus
    subclass   = USB
xhci1@pci0:3:0:4:	class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e1 subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven USB 3.1'
    class      = serial bus
    subclass   = USB
none2@pci0:3:0:5:	class=0x048000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e2 subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Raven/Raven2/FireFlight/Renoir Audio Processor'
    class      = multimedia
hdac1@pci0:3:0:6:	class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x19e5 subdevice=0x3e14
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) HD Audio Controller'
    class      = multimedia
    subclass   = HDA

rut@pila:~ % pkg info | grep drm-
drm-devel-kmod-5.4.92.g20210419 DRM modules for the linuxkpi-based KMS components (development version)
libdrm-2.4.104,1               Userspace interface to kernel Direct Rendering Module services

**To Reproduce**
start gdm or gnome-session (latest from ports or binary pkg)

https://i.postimg.cc/9F6zDk3r/20210420-112811.jpg
https://i.postimg.cc/rpcmJtwT/20210420-112823.jpg
https://i.postimg.cc/SNCYf4fW/20210420-112842.jpg
https://i.postimg.cc/B6jBqy9M/20210420-113439.jpg

https://i.postimg.cc/MT1Jr9Pr/20210420-113120.jpg

rut78HR, 2021-04-27:
i manage to found workaround for this problem .. in xorg.conf i have to disable DRI3 and use DRI2 to this freeezee/laggg go away .. 

##Option     "DRI3"    "1"
Option     "DRI" "2"      

so is somthing broken with xorg and amdgpu to use DRI3 ??

rut@pila:/var/log % cat Xorg.0.log
[    17.650] (--) Log file renamed from "/var/log/Xorg.pid-1260.log" to "/var/log/Xorg.0.log"
[    17.653] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    17.653] Build Operating System: FreeBSD 14.0-CURRENT amd64 
[    17.653] Current Operating System: FreeBSD pila 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n246078-14d0cd7225e: Thu Apr 15 08:36:43 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
[    17.653] Build Date: 27 April 2021  10:37:33AM
[    17.653]  
[    17.653] Current version of pixman: 0.40.0
[    17.653] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    17.653] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    17.653] (==) Log file: "/var/log/Xorg.0.log", Time: Tue Apr 27 13:32:29 2021
[    17.655] (==) Using config file: "/etc/X11/xorg.conf"
[    17.655] (==) Using system config directory "/usr/local/share/X11/xorg.conf.d"
[    17.657] (==) ServerLayout "X.org Configured"
[    17.657] (**) |-->Screen "Screen0" (0)
[    17.657] (**) |   |-->Monitor "Monitor0"
[    17.657] (**) |   |-->Device "Card0"
[    17.657] (**) |-->Input Device "Mouse0"
[    17.657] (**) |-->Input Device "Keyboard0"
[    17.657] (==) Automatically adding devices
[    17.657] (==) Automatically enabling devices
[    17.657] (==) Not automatically adding GPU devices
[    17.657] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    17.658] (WW) The directory "/usr/local/share/fonts/misc/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/Type1/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/100dpi/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/75dpi/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/misc/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/Type1/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/100dpi/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (WW) The directory "/usr/local/share/fonts/75dpi/" does not exist.
[    17.658] 	Entry deleted from font path.
[    17.658] (**) FontPath set to:
	/usr/local/share/fonts/TTF/,
	/usr/local/share/fonts/OTF/,
	catalogue:/usr/local/etc/X11/fontpath.d,
	/usr/local/share/fonts/TTF/,
	/usr/local/share/fonts/OTF/,
	catalogue:/usr/local/etc/X11/fontpath.d
[    17.658] (**) ModulePath set to "/usr/local/lib/xorg/modules"
[    17.658] (WW) Hotplugging is on, devices using drivers 'kbd', 'mouse' or 'vmmouse' will be disabled.
[    17.658] (WW) Disabling Mouse0
[    17.658] (WW) Disabling Keyboard0
[    17.658] (II) Loader magic: 0x4359f0
[    17.658] (II) Module ABI versions:
[    17.658] 	X.Org ANSI C Emulation: 0.4
[    17.658] 	X.Org Video Driver: 24.1
[    17.658] 	X.Org XInput driver : 24.1
[    17.658] 	X.Org Server Extension : 10.0
[    17.659] (--) PCI:*(3@0:0:0) 1002:15d8:19e5:3e14 rev 194, Mem @ 0xb0000000/268435456, 0xc0000000/2097152, 0xc0500000/524288, I/O @ 0x00001000/256, BIOS @ 0x????????/65536
[    17.659] (II) "glx" will be loaded. This was enabled by default and also specified in the config file.
[... 256 lines cut ...]
[    18.447] (**) ELAN2602:05 04F3:3109 TouchPad: Applying InputClass "evdev pointer catchall"
[    18.447] (**) ELAN2602:05 04F3:3109 TouchPad: Applying InputClass "evdev touchpad catchall"
[    18.447] (**) ELAN2602:05 04F3:3109 TouchPad: Applying InputClass "libinput pointer catchall"
[    18.447] (**) ELAN2602:05 04F3:3109 TouchPad: Applying InputClass "libinput touchpad catchall"
[    18.447] (II) Using input driver 'libinput' for 'ELAN2602:05 04F3:3109 TouchPad'
[    18.447] (**) ELAN2602:05 04F3:3109 TouchPad: always reports core events
[    18.447] (**) Option "Device" "/dev/input/event7"
[    18.447] (**) Option "_source" "server/udev"
[    18.447] (II) event7  - ELAN2602:05 04F3:3109 TouchPad: is tagged by udev as: Mouse Touchpad
[    18.450] (II) event7  - ELAN2602:05 04F3:3109 TouchPad: device is a touchpad
[    18.450] (II) event7  - ELAN2602:05 04F3:3109 TouchPad: device removed
[    18.451] (**) Option "config_info" "udev:/dev/input/event7"
[    18.451] (II) XINPUT: Adding extended input device "ELAN2602:05 04F3:3109 TouchPad" (type: TOUCHPAD, id 12)
[    18.452] (**) Option "AccelerationScheme" "none"
[    18.453] (**) ELAN2602:05 04F3:3109 TouchPad: (accel) selected scheme none/0
[    18.453] (**) ELAN2602:05 04F3:3109 TouchPad: (accel) acceleration factor: 2.000
[    18.453] (**) ELAN2602:05 04F3:3109 TouchPad: (accel) acceleration threshold: 4
[    18.454] (II) event7  - ELAN2602:05 04F3:3109 TouchPad: is tagged by udev as: Mouse Touchpad
[    18.455] (II) event7  - ELAN2602:05 04F3:3109 TouchPad: device is a touchpad
[    22.030] (II) AMDGPU(0): EDID vendor "CMN", prod id 33281
[    22.030] (II) AMDGPU(0): Printing DDC gathered Modelines:
[    22.030] (II) AMDGPU(0): Modeline "2160x1440"x0.0  206.02  2160 2208 2240 2320  1440 1443 1453 1480 -hsync -vsync (88.8 kHz eP)
[    53.583] (II) AMDGPU(0): EDID vendor "CMN", prod id 33281
[    53.583] (II) AMDGPU(0): Printing DDC gathered Modelines:
[    53.583] (II) AMDGPU(0): Modeline "2160x1440"x0.0  206.02  2160 2208 2240 2320  1440 1443 1453 1480 -hsync -vsync (88.8 kHz eP)

grahamperrin, 2022-08-06:
Also, FreeBSD bugs: 

* [253746 – x11-wm/mutter: Freezes on AMD GPU's with DRI3](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=253746)
* [257810 – x11/gnome3(-lite) & x11/gdm lags significantly on amdgpu](https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=257810)

grahamperrin, 2023-05-13:
> … in xorg.conf i have to disable DRI3 and use DRI2 to this freeezee/laggg go away ..
> 
> ##Option "DRI3" "1" Option "DRI" "2" …

What, exactly, should be the content of a `.conf` file? 

I can't interpret the manual page for xorg.conf(5). It makes my brain bleed.

grahamperrin, 2023-05-13:
I worked around as outlined at <https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=253746#c12>.

grahamperrin, 2024-03-28:
<https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=253746#c20> (2024-02-17): 

> … i do run GNOME now for a long time and have no issues so far.
> 
> Regarding that i would like to close this issue if its fine.

@rut78HR is this issue #72 still reproducible?

rut78HR, 2024-03-28:
Hi ..U can close it now. Its ok. Sent from my iPhone

evadot, 2024-04-17:
Closing as OP said it's fixed.

closed 2024-04-17 by evadot