freebsd/drm-kmod #74: adev->powerplay.pp_funcs  not set

sinetek, 2021-05-03:
the bug is that "adev->powerplay.pp_funcs"  is not set. where is this normally set?
my chip is a Renoir one in the AMD Ryzen 5 4600H, vendor=0x1002 device=0x1636.

When I unplug or plug the power connector I get an ACPI event, which propagates to amdgpu_pm_acpi_event_handler,
which then wants to call the powerplay table

```
        if (adev->powerplay.pp_funcs->enable_bapm) {    <------ nasty crash
            amdgpu_dpm_enable_bapm(adev, adev->pm.ac_power);
        }

```

sinetek, 2021-05-03:
Also want to thank a million for this amazing project. I'm looking at the code and can't believe what has been ported to FreeBSD, huge achievement this is.

unrelentingtech, 2021-05-08:
We have the fix in 5.4-lts: https://github.com/freebsd/drm-kmod/commit/3ed2827cbc3601ea3836a4f431d02f7fc38dde0e so this one should be cherry-picked onto 5.5-wip

sinetek, 2021-05-08:
> We have the fix in 5.4-lts: 3ed2827
> so this one should be cherry-picked onto 5.5-wip

wow that would have saved a lot of time, but i learned a lot hunting down
this bug.

id like to help
In general is it okay to sync portions of the code or should we only cherry
pick commits?

evadot, 2022-01-27:
3ed2827 is in master and 5.7 (which is now drm-devel-kmod), closing the issue.

closed 2022-01-27 by evadot