freebsd/drm-kmod #76: Patch:  Fix the backlight on Renoir

sinetek, 2021-05-07:
Hi, this patch will fix a bug with the backlight on amdgpu.
on bootup the brightness control does not work fully, but on resume it does work fully.

this patch forces the method that does work all the time.

sinetek, 2021-05-07:
```
--- a/drivers/gpu/drm/amd/display/dc/dce/dce_abm.c
+++ b/drivers/gpu/drm/amd/display/dc/dce/dce_abm.c
@@ -424,17 +424,7 @@ static bool dce_abm_set_backlight_level_pwm(
 {
        struct dce_abm *abm_dce = TO_DCE_ABM(abm);

-       DC_LOG_BACKLIGHT("New Backlight level: %d (0x%X)\n",
-                       backlight_pwm_u16_16, backlight_pwm_u16_16);
-
-       /* If DMCU is in reset state, DMCU is uninitialized */
-       if (use_smooth_brightness)
-               dmcu_set_backlight_level(abm_dce,
-                               backlight_pwm_u16_16,
-                               frame_ramp,
-                               controller_id);
-       else
-               driver_set_backlight_level(abm_dce, backlight_pwm_u16_16);
+       driver_set_backlight_level(abm_dce, backlight_pwm_u16_16);

        return true;
 }
```

unrelentingtech, 2021-05-08:
From a quick look at the git history of that file, I suppose we need to backport this fix (from 5.8 and up):

https://github.com/torvalds/linux/commit/67d09292f857d47e3577e69236e426cbc31439f0

sinetek, 2021-05-24:
Can this be pulled in? The hack can be removed when we backport.

evadot, 2022-05-20:
This is present now, closing.

closed 2022-05-20 by evadot