freebsd/drm-kmod #78: drm ERROR :amdgpu_job_timedout] ring gfx timeout, but soft recovered

rozhuk-im, 2021-05-19:
**Describe the bug**
Some times I get:
notebook
```
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e00000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e04000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
Limiting closed port RST response from 309 to 100 packets/sec
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e02000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e02000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e03000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e04000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
[drm ERROR :amdgpu_dm_commit_planes] Waiting for fences timed out!drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e04000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
drmn0:   in page starting at address 0x0000800104e02000 from client 27
drmn0: VM_L2_PROTECTION_FAULT_STATUS:0x00541051
drmn0: 	 MORE_FAULTS: 0x1
drmn0: 	 WALKER_ERROR: 0x0
drmn0: 	 PERMISSION_FAULTS: 0x5
drmn0: 	 MAPPING_ERROR: 0x0
drmn0: 	 RW: 0x1
drmn0: [gfxhub0] retry page fault (src_id:0 ring:0 vmid:5 pasid:32772, for process  pid 101038 thread  pid 101038)
[... 262 lines cut ...]
ahci2@pci0:16:0:2:	class=0x010601 rev=0x51 hdr=0x00 vendor=0x1022 device=0x7901 subvendor=0x1849 subdevice=0x7901
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA

</details>

**DRM KMOD version**
drm-fbsd13-kmod 5.4.92.g20210419

**To Reproduce**
Steps to reproduce the behavior:
On notebook this start after I move WiFi driver to byhive.
On desktopp - then zoom map in google pas.

**Additional context**
I found workaround for this:
add to /boot/loader conf
```
compat.linuxkpi.amdgpu_gpu_recovery="1"		# Enable GPU recovery mechanism, (1 = enable, 0 = disable, -1 = auto)
compat.linuxkpi.amdgpu_job_hang_limit="32"	# how much times allow a job hang and not drop it (default 0) / number of times to allow a job to hang before dropping it
```

Probably this should be noted in pkg-message or driver defaults changed.

evilham, 2021-06-08:
Thank you for posting this work-around, this has been bugging me for days, hopefully this fixes my issue.
On my system this sometimes happens every few mins and sometimes not at all for many hours, the main problem is that the GUI (wayland) does not recover and the system is not usable until reboot.

evilham, 2021-06-09:
Ouch, sadly that didn't work for me >,< I was hoping it would.

<details>
<summary>So, this is my `/var/log/messages` for a session: boot + couple minutes usage + drm crash + hardware shutdown button.
Towards the end there is a `VM_L2_PROTECTION_FAULT_STATUS:0x00341051`</summary>

```
Jun  9 19:24:22 verdandi syslogd: kernel boot file is /boot/kernel/kernel
Jun  9 19:24:22 verdandi kernel: ---<<BOOT>>---
Jun  9 19:24:22 verdandi kernel: Copyright (c) 1992-2021 The FreeBSD Project.
Jun  9 19:24:22 verdandi kernel: Copyright (c) 1979, 1980, 1983, 1986, 1988, 1989, 1991, 1992, 1993, 1994
Jun  9 19:24:22 verdandi kernel: 	The Regents of the University of California. All rights reserved.
Jun  9 19:24:22 verdandi kernel: FreeBSD is a registered trademark of The FreeBSD Foundation.
Jun  9 19:24:22 verdandi kernel: FreeBSD 13.0-RELEASE-p1 #0: Wed May 26 22:15:09 UTC 2021
Jun  9 19:24:22 verdandi kernel:     root@amd64-builder.daemonology.net:/usr/obj/usr/src/amd64.amd64/sys/GENERIC amd64
Jun  9 19:24:22 verdandi kernel: FreeBSD clang version 11.0.1 (git@github.com:llvm/llvm-project.git llvmorg-11.0.1-0-g43ff75f2c3fe)
Jun  9 19:24:22 verdandi kernel: VT(efifb): resolution 1024x768
Jun  9 19:24:22 verdandi kernel: CPU: AMD Ryzen 7 PRO 2700U w/ Radeon Vega Mobile Gfx (2195.93-MHz K8-class CPU)
Jun  9 19:24:22 verdandi kernel:   Origin="AuthenticAMD"  Id=0x810f10  Family=0x17  Model=0x11  Stepping=0
Jun  9 19:24:22 verdandi kernel:   Features=0x178bfbff<FPU,VME,DE,PSE,TSC,MSR,PAE,MCE,CX8,APIC,SEP,MTRR,PGE,MCA,CMOV,PAT,PSE36,CLFLUSH,MMX,FXSR,SSE,SSE2,HTT>
Jun  9 19:24:22 verdandi kernel:   Features2=0x7ed8320b<SSE3,PCLMULQDQ,MON,SSSE3,FMA,CX16,SSE4.1,SSE4.2,MOVBE,POPCNT,AESNI,XSAVE,OSXSAVE,AVX,F16C,RDRAND>
Jun  9 19:24:22 verdandi kernel:   AMD Features=0x2e500800<SYSCALL,NX,MMX+,FFXSR,Page1GB,RDTSCP,LM>
Jun  9 19:24:22 verdandi kernel:   AMD Features2=0x35c233ff<LAHF,CMP,SVM,ExtAPIC,CR8,ABM,SSE4A,MAS,Prefetch,OSVW,SKINIT,WDT,TCE,Topology,PCXC,PNXC,DBE,PL2I,MWAITX>
Jun  9 19:24:22 verdandi kernel:   Structured Extended Features=0x209c01a9<FSGSBASE,BMI1,AVX2,SMEP,BMI2,RDSEED,ADX,SMAP,CLFLUSHOPT,SHA>
Jun  9 19:24:22 verdandi kernel:   XSAVE Features=0xf<XSAVEOPT,XSAVEC,XINUSE,XSAVES>
Jun  9 19:24:22 verdandi kernel:   AMD Extended Feature Extensions ID EBX=0x1007<CLZERO,IRPerf,XSaveErPtr,IBPB>
Jun  9 19:24:22 verdandi kernel:   SVM: NP,NRIP,VClean,AFlush,DAssist,NAsids=32768
Jun  9 19:24:22 verdandi kernel:   TSC: P-state invariant, performance statistics
Jun  9 19:24:22 verdandi kernel: real memory  = 34359738368 (32768 MB)
Jun  9 19:24:22 verdandi kernel: avail memory = 32144011264 (30654 MB)
Jun  9 19:24:22 verdandi kernel: Event timer "LAPIC" quality 600
Jun  9 19:24:22 verdandi kernel: ACPI APIC Table: <LENOVO TP-R0W  >
Jun  9 19:24:22 verdandi kernel: FreeBSD/SMP: Multiprocessor System Detected: 8 CPUs
Jun  9 19:24:22 verdandi kernel: FreeBSD/SMP: 1 package(s) x 4 core(s) x 2 hardware threads
Jun  9 19:24:22 verdandi kernel: random: registering fast source Intel Secure Key RNG
Jun  9 19:24:22 verdandi kernel: random: fast provider: "Intel Secure Key RNG"
Jun  9 19:24:22 verdandi kernel: random: unblocking device.
Jun  9 19:24:22 verdandi kernel: ioapic0: MADT APIC ID 32 != hw id 0
Jun  9 19:24:22 verdandi kernel: ioapic1: MADT APIC ID 33 != hw id 0
Jun  9 19:24:22 verdandi kernel: ioapic0 <Version 2.1> irqs 0-23
Jun  9 19:24:22 verdandi kernel: ioapic1 <Version 2.1> irqs 24-55
Jun  9 19:24:22 verdandi kernel: Launching APs: 4 6 2 5 1 3 7
Jun  9 19:24:22 verdandi kernel: Timecounter "TSC-low" frequency 1097965473 Hz quality 1000
Jun  9 19:24:22 verdandi kernel: KTLS: Initialized 8 threads
Jun  9 19:24:22 verdandi kernel: Cuse v0.1.36 @ /dev/cuse
Jun  9 19:24:22 verdandi kernel: random: entropy device external interface
Jun  9 19:24:22 verdandi kernel: [ath_hal] loaded
Jun  9 19:24:22 verdandi kernel: WARNING: Device "kbd" is Giant locked and may be deleted before FreeBSD 14.0.
Jun  9 19:24:22 verdandi kernel: kbd1 at kbdmux0
Jun  9 19:24:22 verdandi kernel: 000.000060 [4354] netmap_init               netmap: loaded module
Jun  9 19:24:22 verdandi kernel: mlx5en: Mellanox Ethernet driver 3.6.0 (December 2020)
Jun  9 19:24:22 verdandi kernel: nexus0
Jun  9 19:24:22 verdandi kernel: efirtc0: <EFI Realtime Clock>
Jun  9 19:24:22 verdandi kernel: efirtc0: registered as a time-of-day clock, resolution 1.000000s
Jun  9 19:24:22 verdandi kernel: cryptosoft0: <software crypto>
Jun  9 19:24:22 verdandi kernel: aesni0: <AES-CBC,AES-CCM,AES-GCM,AES-ICM,AES-XTS,SHA1,SHA256>
Jun  9 19:24:22 verdandi kernel: acpi0: <LENOVO TP-R0W>
Jun  9 19:24:22 verdandi kernel: acpi0: Power Button (fixed)
Jun  9 19:24:22 verdandi kernel: cpu0: <ACPI CPU> on acpi0
Jun  9 19:24:22 verdandi kernel: atrtc0: <AT realtime clock> port 0x70-0x71 on acpi0
Jun  9 19:24:22 verdandi kernel: atrtc0: registered as a time-of-day clock, resolution 1.000000s
Jun  9 19:24:22 verdandi kernel: Event timer "RTC" frequency 32768 Hz quality 0
Jun  9 19:24:22 verdandi kernel: attimer0: <AT timer> port 0x40-0x43 irq 0 on acpi0
Jun  9 19:24:22 verdandi kernel: Timecounter "i8254" frequency 1193182 Hz quality 0
Jun  9 19:24:22 verdandi kernel: Event timer "i8254" frequency 1193182 Hz quality 100
Jun  9 19:24:22 verdandi kernel: hpet0: <High Precision Event Timer> iomem 0xfed00000-0xfed003ff on acpi0
Jun  9 19:24:22 verdandi kernel: Timecounter "HPET" frequency 14318180 Hz quality 950
[... cut ...]
drm-fbsd13-kmod 5.4.92.g20210419
drm-kmod g20190710_1
```

**To Reproduce**
Steps to reproduce the behavior:

Very often after using videoconferencing software (e.g. jitsi on firefox) in a wayland composer, graphics stop updating after a while and everything becomes unresponsive. Only pressing the hardware poweroff button enables a proper shutdown process.

**Additional context**

This has been happening for a couple months now and setting the aforementioned `sysctl`s related to `amdgpu` did not help.

evadot, 2021-10-06:
@evilham please create a new issue for your problem.

evilham, 2021-10-06:
Hey, I had it in my queue to get back to this issue. Just for completeness sake and in case this helps OP:

it looked to me like quite the same issue, but in reality it was related to the GPU firmware, I've been testing this patch for `graphics/gpu-firmware-kmod` for a couple weeks now with the latest `graphics/drm-kmod` too and my issue is totally gone: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=257786

evadot, 2022-06-09:
Firmware were updated so closing this issue.

closed 2022-06-09 by evadot