freebsd/drm-kmod #81: drm-kmod crash on i915kms when Xorg is loaded

neelchauhan, 2021-06-02:
**Describe the bug**

When I load `i915kms` and then `gdm`, I get a coredump. This happens on the first boot, but not subsequent boots.

Coredump: [core.0.txt](https://github.com/freebsd/drm-kmod/files/6581781/core.0.txt)

I am using Xorg. The system is a HP Spectre x360 13-ap0053dx (WhiskeyLake).

**FreeBSD version**

    FreeBSD spectre.local 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n246975-8d5c7813061: Thu May 27 09:39:01 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

    hostb0@pci0:0:0:0:	class=0x060000 rev=0x0b hdr=0x00 vendor=0x8086 device=0x3e34 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Coffee Lake HOST and DRAM Controller'
        class      = bridge
        subclass   = HOST-PCI
    vgapci0@pci0:0:2:0:	class=0x030000 rev=0x00 hdr=0x00 vendor=0x8086     device=0x3ea0 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'WhiskeyLake-U GT2 [UHD Graphics 620]'
        class      = display
        subclass   = VGA
    none0@pci0:0:4:0:	class=0x118000 rev=0x0b hdr=0x00 vendor=0x8086 device=0x1903 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem'
        class      = dasp
    pchtherm0@pci0:0:18:0:	class=0x118000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df9 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP Thermal Controller'
        class      = dasp
    none1@pci0:0:19:0:	class=0x070000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9dfc subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP Integrated Sensor Hub'
        class      = simple comms
        subclass   = UART
    xhci0@pci0:0:20:0:	class=0x0c0330 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9ded subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP USB 3.1 xHCI Controller'
        class      = serial bus
        subclass   = USB
    none2@pci0:0:20:2:	class=0x050000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9def subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP Shared SRAM'
        class      = memory
        subclass   = RAM
    iwm0@pci0:0:20:3:	class=0x028000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9df0 subvendor=0x8086 subdevice=0x0034
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP CNVi [Wireless-AC]'
        class      = network
    ig4iic0@pci0:0:21:0:	class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de8 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP Serial IO I2C Controller'
        class      = serial bus
    ig4iic1@pci0:0:21:1:	class=0x0c8000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de9 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP Serial IO I2C Controller'
        class      = serial bus
    none3@pci0:0:22:0:	class=0x078000 rev=0x30 hdr=0x00 vendor=0x8086 device=0x9de0 subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = 'Cannon Point-LP MEI Controller'
        class      = simple comms
    ahci0@pci0:0:23:0:	class=0x010400 rev=0x30 hdr=0x00 vendor=0x8086 device=0x282a subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
        device     = '82801 Mobile SATA Controller [RAID mode]'
        class      = mass storage
        subclass   = RAID
    pcib1@pci0:0:28:0:	class=0x060400 rev=0xf0 hdr=0x01 vendor=0x8086 device=0x9dba subvendor=0x103c subdevice=0x8514
        vendor     = 'Intel Corporation'
[... 78 lines cut ...]
        vendor     = 'Intel Corporation'
        class      = mass storage
        subclass   = NVM

</details>

**DRM KMOD version**

`5.5-wip` branch, but with cherry-picked `e5194b8949b316adfcda4895fc117cf1045d9336` and `b8a85669f61ffea54860a76f72b8c5c0bb647127` to unbreak the build.

**To Reproduce**

Checkout 5.5-wip branch.

Cherry pick commits `e5194b8949b316adfcda4895fc117cf1045d9336` and `b8a85669f61ffea54860a76f72b8c5c0bb647127` from `master`.

**Screenshots**

N/A.

**Additional context**

This happens (on my Spectre) on the first boot. Subsequent reboots after the first panic are not affected for me, and I can use gdm, gnome, firefox (WebGL fish), wine (Toontown Rewritten) normally.

@evadot has mentioned there is a crash when running Xorg or Weston, and I believe this is the same issue manifested differently.

neelchauhan, 2021-06-02:
New coredump: [core.1.txt](https://github.com/freebsd/drm-kmod/files/6581911/core.1.txt)

Previous one lacked debugging symbols, sorry.

evadot, 2021-09-23:
This have been fixed since.
Closing.

closed 2021-09-23 by evadot