freebsd/drm-kmod #85: Importing my WIP 5.7 branch

neelchauhan, 2021-06-07:
I have a WIP branch to merge 5.7 here: https://github.com/neelchauhan/drm-kmod/tree/5.7-wip-demo

This is based on the branch for my PR #83, but can be rebased.

Why 5.7 and not 5.6? Two things: I want to get off `scfb` on my TigerLake laptop and I had a lot of time to burn this weekend so I figured out why not jump directly. This is my first time doing this so I could absolutely have done it wrong or something.

However, we should merge #82 and #83 before proceeding.

unrelentingtech, 2021-06-07:
On a first glance it seemed like you handled the #45 / "amdgpu: Wrap FPU dependent functions in dc20" thing correctly, but there's one leftover unmatched `kernel_fpu_end` in dcn20 that definitely shouldn't be there:

https://github.com/neelchauhan/drm-kmod/blob/1f7b91b5a22ed8779d99bc2f7bc287f3c58e9ee0/drivers/gpu/drm/amd/display/dc/dcn20/dcn20_resource.c#L3956-L3958

Plus a matching pair in `dcn21_validate_bandwidth` that shouldn't be necessary anymore either. But that should be tested on Renoir hardware.

unrelentingtech, 2021-06-07:
I have 5.7 amdgpu working for the most part https://github.com/neelchauhan/drm-kmod/pull/1

evadot, 2021-06-07:
Yeah I was afraid that other stuff broke.
I've ask Neel in private to make sure that 
1/ all commits build (easier to bisect)
2/ split by linux version (easier to test regression and then bisect).

neelchauhan, 2021-06-08:
Sorry if I haven't done those yet. Amdgpu now builds, and I'm attempting to get i915kms to compile in my branch: https://github.com/neelchauhan/drm-kmod/tree/5.7-wip

neelchauhan, 2021-06-08:
On my branch, I made i915kms compile (yay!) i haven't tested it yet however.

neelchauhan, 2021-06-08:
I have some good news: I got i915kms working, being able to load the console Xorg and Fluxbox (since GNOME is broken in Ports, at least the last time I checked) both Firefox and Chromium being able to run WebGL demos.

I tested this on my WhiskeyLake system (HP Spectre x360 13-ap0053dx). My TigerLake system (HP Spectre x360 13t-aw200) will follow.

These aren't the most through tests, but it's something.

neelchauhan, 2021-07-07:
Closing this issue. @evadot said in #91 that he wants to go to 5.6 first.

While my branch can load Xorg on amdgpu and i915, there are some bugs (noticable on TigerLake since it's new, less so on WhiskeyLake) and doesn't incorporate some of @wulf7's 5.5 fixes.

Some of the changes in my 5.7-wip branch can be rebased on the 5.6 branch, and the rest on the official 5.7-wip branch.

closed 2021-07-07 by neelchauhan