freebsd/drm-kmod #87: rx580/AMDGPU System Crash (Display Manager freezes and unable to switch to tty) when using VAAPI

AmeliaFloof, 2021-06-09:
When running this command: (which tests vaapi video), my FREEBSD 13-RELEASE system freezes up
 mpv --no-config --msg-level=vo/gpu=v,vd=v --hwdec=vaapi --vo=gpu https://jell.yfish.us/media/jellyfish-3-mbps-hd-h264.mkv

**FreeBSD version**
 13.0-RELEASE FreeBSD 13.0-RELEASE #0 releng/13.0-n244733-ea31abc261f: Fri Apr  9 04:24:09 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**
hostb0@pci0:0:0:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1022 subdevice=0x1480
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:       class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1481 subvendor=0x1022 subdevice=0x1481
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib9@pci0:0:3:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:7:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib10@pci0:0:7:1:      class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1484 subvendor=0x1022 subdevice=0x1484
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Internal PCIe GPP Bridge 0 to bus[E:B]'
    class      = bridge
    subclass   = PCI-PCI
hostb7@pci0:0:8:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
[... 144 lines cut ...]
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Cryptographic Coprocessor PSPCPP'
    class      = encrypt/decrypt
xhci2@pci0:11:0:3:      class=0x0c0330 rev=0x00 hdr=0x00 vendor=0x1022 device=0x149c subvendor=0x1458 subdevice=0x5007
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Matisse USB 3.0 Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:11:0:4:      class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1487 subvendor=0x1458 subdevice=0xa0cf
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse HD Audio Controller'
    class      = multimedia
    subclass   = HDA

**DRM KMOD version**
drm-fbsd13-kmod 5.4.92.g20210419
drm-kmod g20190710_1

**To Reproduce**
Dispatch this command while using an rx580 (I was also using Sway at the time)
 mpv --no-config --msg-level=vo/gpu=v,vd=v --hwdec=vaapi --vo=gpu https://jell.yfish.us/media/jellyfish-3-mbps-hd-h264.mkv
Either 5 seconds in the video or at the end of the 30, the system will crash.

I am using:
Rx 580 + Ryzen 9 3900x and have had major compatibillity issues whilst running earlier freeBSD releases.

AmeliaFloof, 2021-06-09:
Admendment to the isuse:
At the time I was using the most recent of mesa-devel in tandem with mesa-dri

Unfortunately at the time of the crash i did not have the nessecary logging things set up to retreive a full stacktrace

grahamperrin, 2021-07-03:
Correction to the command: 

`mpv --no-config --msg-level=vo/gpu=v,vd=v --hwdec=vaapi --vo=gpu https://jell.yfish.us/media/jellyfish-3-mbps-hd-h264.mkv`

evadot, 2022-05-20:
No news so closing the issue.

closed 2022-05-20 by evadot