freebsd/drm-kmod #89: Intel fixes for 5.5-wip

wulf7, 2021-07-05:
Hopefully, most Intel's outstanding issues of 5.5-wip are fixed in this branch now:
https://github.com/wulf7/drm-kmod/commits/5.5-wip
I tested it on KabyLake for 2 weeks and got only one unexpected crash which happened at i915kms.ko module load
Also this branch got import of ~20 missing commits, port of 5.5 dma-buf changes and some mismerge cleanups

5.5.19 commits have been imported too. They can be found here:
https://github.com/wulf7/drm-kmod/commits/5.5.y-wip

I can not do a pull-request as I had to insert some of missing commits in between of already existing that broke hashes and made my branches unmergeable.

Requires very recent current having __FreeBSD_version >= 1400025

neelchauhan, 2021-07-05:
Thanks for your work, @wulf7!

I will rebase my 5.7-wip branch on your 5.5-wip branch (when/if this branch gets merged). Right now, I'm battling an issue with a NVMe bug on my TigerLake laptop so I wasn't able to do as much work on drm-kmod.

evadot, 2021-07-05:
Awesome.
I guess that the easiest way for everyone is that you open a MR with the master branch as the target.
I can then apply that, tag and update drm-devel-kmod.

wulf7, 2021-07-05:
Here you are: https://github.com/freebsd/drm-kmod/pull/90

If this changes are ok, please create 5.5 stable branch to make MR for 5.5.y-wip

hericium, 2021-07-17:
5.5-wip that existed here previously was the only one that worked for me with Ryzen/Renoir graphics on 4750G and 4650U CPUs (both 1002:1363). Currently 5.5-wip at wulf7/drm-kmod does not compile for me on 13.0-STABLE (
```
In file included from /build/drm-kmod/drivers/dma-buf//dma-buf.c:54:
/build/drm-kmod/linuxkpi/gplv2/include/linux/dma-resv.h:45:10: fatal error: 'linux/seqlock.h' file not found
```

After blindly copying this file from 5.5-wip I had from here: 
```
/build/drm-kmod/drivers/gpu/drm/drm_client.c:390:2: error: implicit declaration of function 'strscpy' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        strscpy(buffer->fb->comm, client->name, TASK_COMM_LEN);
        ^
```

wulf7, 2021-07-17:
Some required LKPI changes are not MFC-ed yet. You may try to checkout commit 787ffa8c and compile it

hericium, 2021-07-19:
I switched from a recent 13.0-STABLE to 13.0-RELEASE-p3 releng/13.0-n244752-b368bb75bad8 but there's the below error on both with this commit:

```
/build/drm-kmod/drivers/gpu/drm/amd/amdgpu/amdgpu_dma_buf.c:256:10: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'struct dma_attrs *' [-Werror,-Wint-conversion]
                              DMA_ATTR_SKIP_CPU_SYNC))
                              ^~~~~~~~~~~~~~~~~~~~~~
/usr/src/sys/compat/linuxkpi/common/include/linux/dma-attrs.h:39:33: note: expanded from macro 'DMA_ATTR_SKIP_CPU_SYNC'
#define DMA_ATTR_SKIP_CPU_SYNC          (1 << 5)
                                        ^~~~~~~~
/usr/src/sys/compat/linuxkpi/common/include/linux/dma-mapping.h:193:52: note: passing argument to parameter 'attrs' here
    enum dma_data_direction dir, struct dma_attrs *attrs)
```

If this isn't right place and you'd rather like me to open a new issue, please forward me accordingly but for the time being I'll leave some details here. I have two seemingly similar (1002:1636) Ryzen/Radeon/Renoir platforms:

### Laptop with Ryzen 4650U 
To be perfectly honest I am not 100% sure I ever had working acceleration in Xorg (used scfb, too) but I had the driver working in the console and it was causing Xorg panics if "amdgpu" driver was configured for the "Card" Section. 
```
[drm] Initialized amdgpu 3.35.0 20150101 for drmn0 on minor 0
```

The commit from May that got me accelerated graphics on 13.0-STABLE desktop (below) gives me this on the (both 13.0-STABLE and -RELEASE) laptop:
```
[drm] amdgpu kernel modesetting enabled.
drmn0: <drmn> on vgapci0
[drm] This hardware requires experimental hardware support.
See modparam exp_hw_support
device_attach: drmn0 attach returned 19
```
pciconf -lv:
```
vgapci0@pci0:6:0:0:     class=0x030000 rev=0xd3 hdr=0x00 vendor=0x1002 device=0x1636 subvendor=0x17aa subdevice=0x5082
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Renoir'
    class      = display
    subclass   = VGA
```

### Desktop with Ryzen 4750G:
13.0-STABLE stable/13-n245875-70e13c4cffd with commit e5194b8949b316adfcda4895fc117cf1045d9336 works almost flawlessly:

```
[drm] amdgpu kernel modesetting enabled.
[drm] amdgpu: 512M of VRAM memory ready
[drm] amdgpu: 3072M of GTT memory ready.
[drm] Initialized amdgpu 3.36.0 20150101 for drmn0 on minor 0
```

```
[  5410.145] (II) AMDGPU(0): glamor X acceleration enabled on AMD RENOIR (DRM 3.36.0, 13.0-STABLE, LLVM 10.0.1)
[  5410.155] (II) AMDGPU(0): [DRI2] Setup complete
[  5410.155] (II) AMDGPU(0): [DRI2]   DRI driver: radeonsi
[  5410.155] (II) AMDGPU(0): [DRI2]   VDPAU driver: radeonsi
[  5410.191] (II) AMDGPU(0): Front buffer pitch: 7680 bytes
[  5410.192] (II) AMDGPU(0): SYNC extension fences enabled
[  5410.193] (II) AMDGPU(0): Present extension enabled
[  5410.193] (==) AMDGPU(0): DRI3 enabled
[  5410.193] (==) AMDGPU(0): Backing store enabled
[  5410.193] (II) AMDGPU(0): Direct rendering enabled
[  5410.209] (II) AMDGPU(0): Use GLAMOR acceleration.
[  5410.209] (II) AMDGPU(0): Acceleration enabled
[  5410.209] (==) AMDGPU(0): DPMS enabled
[  5410.209] (==) AMDGPU(0): Silken mouse enabled
[  5410.209] (II) AMDGPU(0): Set up textured video (glamor)
```
pciconf -lv:
```
vgapci0@pci0:11:0:0:    class=0x030000 rev=0xd8 hdr=0x00 vendor=0x1002 device=0x1636 subvendor=0x17aa subdevice=0x318e
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Renoir'
    class      = display
    subclass   = VGA
```

The "almost flawlessly" part is that it sometimes crashes the system during boot but after it loads (sometimes after 2-3 crashes) there seems to be no more issues.

wulf7, 2021-07-20:
```
/build/drm-kmod/drivers/gpu/drm/amd/amdgpu/amdgpu_dma_buf.c:256:10: error: incompatible integer to pointer conversion passing 'int' to parameter of type 'struct dma_attrs *' [-Werror,-Wint-conversion]
                              DMA_ATTR_SKIP_CPU_SYNC))
```
You can replace DMA_ATTR_SKIP_CPU_SYNC with NULL to compile this line.

> If this isn't right place and you'd rather like me to open a new issue, please forward me accordingly

I am going to MFC required changes to 13-STABLE soon, so you will be able to build 5.5-stable on it.

wulf7, 2021-07-21:
> I am going to MFC required changes to 13-STABLE soon, so you will be able to build 5.5-stable on it.

All required changes are MFC-ed. You can try to build 5.5-stable again

hericium, 2021-07-26:
@wulf7 5.5-stable works like a charm on Ryzen 4650U with `stable/13-n246457-1ea187f8a1b`. Thank you for your work! 

```
[drm] amdgpu kernel modesetting enabled.
[drm] amdgpu: 512M of VRAM memory ready
[drm] amdgpu: 3072M of GTT memory ready.
[drm] Initialized amdgpu 3.36.0 20150101 for drmn0 on minor 0
```

```
[   366.624] (II) AMDGPU(0): glamor X acceleration enabled on AMD RENOIR (DRM 3.36.0, 13.0-STABLE, LLVM 10.0.1)
[   366.629] (II) AMDGPU(0): [DRI2] Setup complete
[   366.629] (II) AMDGPU(0): [DRI2]   DRI driver: radeonsi
[   366.629] (II) AMDGPU(0): [DRI2]   VDPAU driver: radeonsi
[   366.675] (II) AMDGPU(0): Front buffer pitch: 7680 bytes
[   366.676] (II) AMDGPU(0): SYNC extension fences enabled
[   366.677] (II) AMDGPU(0): Present extension enabled
[   366.677] (==) AMDGPU(0): DRI3 enabled
[   366.677] (==) AMDGPU(0): Backing store enabled
[   366.677] (II) AMDGPU(0): Direct rendering enabled
[   366.696] (II) AMDGPU(0): Use GLAMOR acceleration.
[   366.696] (II) AMDGPU(0): Acceleration enabled
[   366.696] (==) AMDGPU(0): DPMS enabled
[   366.696] (==) AMDGPU(0): Silken mouse enabled
[   366.696] (II) AMDGPU(0): Set up textured video (glamor)

```

wulf7, 2021-09-08:
Committed to ports tree as graphics/drm-devel-lmod

closed 2021-09-08 by wulf7