freebsd/drm-kmod #93: radeonkms no longer loads with drm-devel-kmod on AMD Thames [Radeon HD 7550M/7570M/7650M]

grahamperrin, 2021-07-08:
**Describe the bug**

`5.5.0.g20210707` built and installed with FreeBSD 14.0-CURRENT `n247798-f39dd6a9784` can load radeonkms for me. 

**FreeBSD version**

n247798-f39dd6a9784

<https://cgit.freebsd.org/src/log/?qt=range&q=f39dd6a9784> (2021-07-08)

```text
# bectl list -c creation
BE                    Active Mountpoint Space Created
n246499-097e8701c9f-d -      -          12.4G 2021-05-12 16:33
n247326-2349cda44fe-b -      -          8.17G 2021-06-12 09:36
n247565-b49ba74deeb-f NR     /          65.7G 2021-07-04 10:18
n247798-f39dd6a9784-a -      -          5.27G 2021-07-08 07:54
# 
```

– I reverted from `n247798-f39dd6a9784` to `n247565-b49ba74deeb` (with `drm-devel-kmod-5.4.92.g20210526`) to regain normal use of the computer. 

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```text
hostb0@pci0:0:0:0:      class=0x060000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x0154 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '3rd Gen Core processor DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:       class=0x060400 rev=0x09 hdr=0x01 vendor=0x8086 device=0x0151 subvendor=0x8086 subdevice=0x2010
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e31 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family USB xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:22:0:      class=0x078000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e3a subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family MEI Controller'
    class      = simple comms
uart2@pci0:0:22:3:      class=0x070002 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e3d subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family KT Controller'
    class      = simple comms
    subclass   = UART
em0@pci0:0:25:0:        class=0x020000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1502 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '82579LM Gigabit Network Connection (Lewisville)'
    class      = network
    subclass   = ethernet
ehci0@pci0:0:26:0:      class=0x0c0320 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e2d subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family USB Enhanced Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:0:27:0:      class=0x040300 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e20 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib2@pci0:0:28:0:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e10 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family PCI Express Root Port 1'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:1:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e12 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family PCI Express Root Port 2'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:28:2:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e14 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family PCI Express Root Port 3'
    class      = bridge
    subclass   = PCI-PCI
pcib5@pci0:0:28:3:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e16 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family PCI Express Root Port 4'
    class      = bridge
    subclass   = PCI-PCI
ehci1@pci0:0:29:0:      class=0x0c0320 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e26 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family USB Enhanced Host Controller'
    class      = serial bus
    subclass   = USB
isab0@pci0:0:31:0:      class=0x060100 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e55 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = 'QM77 Express Chipset LPC Controller'
    class      = bridge
    subclass   = PCI-ISA
ahci0@pci0:0:31:2:      class=0x010601 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e03 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series Chipset Family 6-port SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
vgapci0@pci0:1:0:0:     class=0x030000 rev=0x00 hdr=0x00 vendor=0x1002 device=0x6841 subvendor=0x103c subdevice=0x17a9
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Thames [Radeon HD 7550M/7570M/7650M]'
    class      = display
    subclass   = VGA
hdac0@pci0:1:0:1:       class=0x040300 rev=0x00 hdr=0x00 vendor=0x1002 device=0xaa90 subvendor=0x103c subdevice=0x17a9
    vendor     = 'Advanced Micro Devices, Inc. [AMD/ATI]'
    device     = 'Turks HDMI Audio [Radeon HD 6500/6600 / 6700M Series]'
    class      = multimedia
    subclass   = HDA
none1@pci0:36:0:0:      class=0x0c0010 rev=0x30 hdr=0x00 vendor=0x197b device=0x2380 subvendor=0x103c subdevice=0x17a7
    vendor     = 'JMicron Technology Corp.'
    device     = 'IEEE 1394 Host Controller'
    class      = serial bus
    subclass   = FireWire
none2@pci0:36:0:1:      class=0x088000 rev=0x30 hdr=0x00 vendor=0x197b device=0x2392 subvendor=0x103c subdevice=0x17a7
    vendor     = 'JMicron Technology Corp.'
    device     = 'SD/MMC Host Controller'
    class      = base peripheral
sdhci_pci0@pci0:36:0:2: class=0x080501 rev=0x30 hdr=0x00 vendor=0x197b device=0x2391 subvendor=0x103c subdevice=0x17a7
    vendor     = 'JMicron Technology Corp.'
    device     = 'Standard SD Host Controller'
    class      = base peripheral
    subclass   = SD host controller
iwn0@pci0:37:0:0:       class=0x028000 rev=0x34 hdr=0x00 vendor=0x8086 device=0x0082 subvendor=0x8086 subdevice=0x1301
    vendor     = 'Intel Corporation'
    device     = 'Centrino Advanced-N 6205 [Taylor Peak]'
    class      = network
```

</details>

**DRM KMOD version**

```text
# bectl mount n247798-f39dd6a9784-a /tmp/drm
Successfully mounted n247798-f39dd6a9784-a at /tmp/drm
# chroot /tmp/drm
root@mowa219-gjp4-8570p:/ # pkg query -x '%n %v' 'drm.*kmod'
drm-devel-kmod 5.5.0.g20210707
root@mowa219-gjp4-8570p:/ # exit
exit
# bectl umount n247798-f39dd6a9784-a
# 
``` 

**To Reproduce**

First boot of the system failed to present SDDM, there appeared a black screen with an immovable pointer. No response to keyboard input e.g. <kbd>Control</kbd>-<kbd>Alt</kbd>-<kbd>F2</kbd>, it was necessary to force off the computer.

No better with the OS safe mode, again it was necessary to force off the computer. 

With SDDM set to not start automatically, I booted, signed in as root and attempted: 

`kldload radeonkms`

– again, the computer became non-responsive, I forced it off. 

**Screenshots**

Not possible.

**Additional context**

Nothing at this time.

grahamperrin, 2021-07-08:
Comparision: `n247798-f39dd6a9784` is OK with drm-_current_-kmod built and installed from ports. 

```text
% pkg query '%o %v %R' drm-kmod drm-current-kmod drm-devel-kmod gpu-firmware-kmod
graphics/drm-kmod g20190710_1 unknown-repository
graphics/drm-current-kmod 5.4.92.g20210526 unknown-repository
graphics/gpu-firmware-kmod g20210330 unknown-repository
% uname -a
FreeBSD mowa219-gjp4-8570p 14.0-CURRENT FreeBSD 14.0-CURRENT #100 main-n247798-f39dd6a9784: Thu Jul  8 07:38:23 BST 2021     root@mowa219-gjp4-8570p:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64
% bectl list -c creation
BE                    Active Mountpoint Space Created
n246499-097e8701c9f-d -      -          12.4G 2021-05-12 16:33
n247326-2349cda44fe-b -      -          8.17G 2021-06-12 09:36
n247565-b49ba74deeb-f -      -          65.7G 2021-07-04 10:18
n247798-f39dd6a9784-a -      -          12.9M 2021-07-08 07:54
n247798-f39dd6a9784-b NR     /          5.32G 2021-07-08 12:12
% 
```

grahamperrin, 2021-07-10:
### `5.5.0.g20210707`

https://github.com/freebsd/drm-kmod/issues/93#issue-939634896

> … No response to keyboard input …

In retrospect, maybe not entirely true. Probably truer: no _visible_ response. 

### `5.5.19.g20210709`

Installed to boot environment `n247798-f39dd6a9784-e`. In context: 

```text
% bectl list -c creation | grep n247798-f39dd6a9784
n247798-f39dd6a9784-a -      -          14.5M 2021-07-08 07:54
n247798-f39dd6a9784-b -      -          6.58M 2021-07-08 12:12
n247798-f39dd6a9784-c -      -          722M  2021-07-09 19:25
n247798-f39dd6a9784-d NR     /          5.83G 2021-07-09 21:06
n247798-f39dd6a9784-e -      -          15.9M 2021-07-10 04:31
% 
````

I could load `drm` but not `radeonkms`. 

After entering `kldload radeonmks`, the command remained on screen – with a white pointer, which I did not attempt to move (IIRC with `5.5.0.g20210707` the pointer flickered slightly). Other than the pointer, there was no visible change to the screen of the notebook (HP EliteBook 8570p); and nothing on the screen of the peripheral display (Philips, DisplayPort via HP dock). 

A short press on the power button successfully suspended the computer. (Note to self: devices on USB; before attempting this, remember to `zpool offline copperbowl gpt/cache-copperbowl` and `zpool export Transcend`.) 

The next short press woke the computer; I tried <kbd>Alt</kbd>-<kbd>F1</kbd> whilst waking but nothing was visibly 'caught'; the display remained dark, blank. 

<kbd>Control</kbd>-<kbd>Alt</kbd>-<kbd>Delete</kbd> successfully but invisibly rebooted. 

I booted good environment `n247798-f39dd6a9784-d` (with drm-current-kmod) then ran `dmesg` and copied these lines: 

```text
[drm] radeon kernel modesetting enabled.
drmn0: <drmn> on vgapci0
VT: Replacing driver "efifb" with new "dummy".
vgapci0: child drmn0 requested pci_enable_io
vgapci0: child drmn0 requested pci_enable_io
sysctl_warn_reuse: can't re-use a leaf (hw.dri.debug)!
[drm] initializing kernel modesetting (TURKS 0x1002:0x6841 0x103C:0x17A9 0x00).
[drm ERROR :radeon_atombios_init] Unable to find PCI I/O BAR; using MMIO for ATOM IIO
ATOM BIOS: HP/Foxconn
drmn0: VRAM: 1024M 0x0000000000000000 - 0x000000003FFFFFFF (1024M used)
drmn0: GTT: 1024M 0x0000000040000000 - 0x000000007FFFFFFF
Successfully added WC MTRR for [0xc0000000-0xcfffffff]: 0; 
[drm] Detected VRAM RAM=1024M, BAR=256M
[drm] RAM width 64bits DDR
WARNING vma_manager == ((void *)0) failed at /usr/ports/graphics/drm-devel-kmod/work/drm-kmod-drm_v5.5.19_0/drivers/gpu/drm/ttm/ttm_bo.c:1711
#0 0xffffffff80e683b3 at linux_dump_stack+0x23
#1 0xffffffff83d84919 at ttm_bo_device_init+0x2c9
#2 0xffffffff83ce3011 at radeon_ttm_init+0x31
#3 0xffffffff83c6db63 at evergreen_init+0x103
#4 0xffffffff83cc35bd at radeon_device_init+0xd9d
#5 0xffffffff83cd3da9 at radeon_driver_load_kms+0x139
#6 0xffffffff83bc05c6 at drm_dev_register+0xc6
#7 0xffffffff83cca60e at radeon_pci_probe+0x20e
#8 0xffffffff80e6c665 at linux_pci_attach_device+0x305
#9 0xffffffff80c57f24 at device_attach+0x3d4
#10 0xffffffff80c59b86 at bus_generic_driver_added+0xb6
#11 0xffffffff80c55859 at devclass_driver_added+0x39
#12 0xffffffff80c557dd at devclass_add_driver+0x13d
#13 0xffffffff80e6ce8f at _linux_pci_register_driver+0xdf
#14 0xffffffff80be8633 at linker_load_module+0xc43
#15 0xffffffff80bea10b at kern_kldload+0x14b
#16 0xffffffff80bea22b at sys_kldload+0x5b
#17 0xffffffff810bb69c at amd64_syscall+0x10c
[drm ERROR :radeon_ttm_init] failed initializing buffer object driver(-22).
drmn0: Fatal error during GPU init
[drm] radeon: finishing device.
[TTM] Memory type 2 has not been initialized
Warning: can't remove non-dynamic nodes (dri)!
device_attach: drmn0 attach returned 22
```

grahamperrin, 2021-07-24:
Noted at <https://www.freebsd.org/status/report-2021-04-2021-06/#_graphics_driver_update_from_linux_5_7>: 

> … radeonkms bring-up …

grahamperrin, 2021-10-12:
Note to self: if/when this issue #93 is fixed, it might become easier for me to trigger symptoms of https://github.com/freebsd/drm-kmod/issues/108 (not necessarily whilst using drm-devel-kmod; maybe after a switch away from drm-devel-kmod to drm-current-kmod) …

evadot, 2021-10-14:
Might be fixed by acfc459

grahamperrin, 2021-10-17:
Thanks, when might be it ported? 

(Not to be pushy. Just curious.)

<https://www.freshports.org/graphics/drm-devel-kmod/#history>

grahamperrin, 2021-10-17:
OK, I'll happily test but (sorry) I need my hand held beyond cloning the repo. What's the routine?

jrtc27, 2021-11-06:
Yeah this is exactly the bug I fixed, didn’t realise there was already an issue open about it

grahamperrin, 2021-11-07:
I don't know whether I'm testing properly. 

```text
% sysrc kld_list
kld_list: fusefs /boot/modules/radeonkms
% 
```

– however I found it necessary to load the module manually; <https://old.reddit.com/r/freebsd/comments/qnvyr4/-/hjmglsz/?context=1>

grahamperrin, 2021-11-07:
```text
% uname -aKU
FreeBSD mowa219-gjp4-8570p-freebsd 14.0-CURRENT FreeBSD 14.0-CURRENT #114 main-n250511-5f73b3338ee: Sat Nov  6 21:15:23 GMT 2021     root@mowa219-gjp4-8570p-freebsd:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64 1400040 1400040
% sysrc kld_list
kld_list: fusefs /boot/modules/radeonkms.ko
% 
```

Loads successfully. 

```text
% pwd
/usr/home/grahamperrin/dev/drm-kmod
% git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
% 
```

Was `master` the correct branch with which to test?

----

> What's the routine?

Partial notes to self: 

```text
% cd /usr/home/grahamperrin/dev/drm-kmod
% git pull
% make
…
% sudo make install
grahamperrin's password:
===> linuxkpi (install)
install -T release -o root -g wheel -m 555   linuxkpi_gplv2.ko /boot/modules/
===> ttm (install)
install -T release -o root -g wheel -m 555   ttm.ko /boot/modules/
===> drm (install)
install -T release -o root -g wheel -m 555   drm.ko /boot/modules/
===> amd (install)
===> amd/amdgpu (install)
install -T release -o root -g wheel -m 555   amdgpu.ko /boot/modules/
===> radeon (install)
install -T release -o root -g wheel -m 555   radeonkms.ko /boot/modules/
===> i915 (install)
install -T release -o root -g wheel -m 555   i915kms.ko /boot/modules/
kldxref /boot/modules
% 
```

grahamperrin, 2022-01-12:
I can't pretend to understand <https://cgit.freebsd.org/ports/commit/?id=4519b9e81473444a7a2ac2bffe2158bd2da05b36> (2022-01-10), but I'm **very** pleased to confirm that `5.5.19.g20211230` works here with FreeBSD 14.0-CURRENT `main-n252255-78beb051a26`. 

Thanks!

```text
% date ; uptime
Wed 12 Jan 2022 02:47:22 GMT
 2:47a.m.  up  8:15, 6 users, load averages: 6.46, 5.17, 3.90
% pkg info -x drm-devel-kmod gpu-firmware-kmod
drm-devel-kmod-5.5.19.g20211230
gpu-firmware-kmod-g20210330
% uname -aKU
FreeBSD mowa219-gjp4-8570p-freebsd 14.0-CURRENT FreeBSD 14.0-CURRENT #120 main-n252255-78beb051a26: Fri Jan  7 07:09:23 GMT 2022     root@mowa219-gjp4-8570p-freebsd:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64 1400046 1400046
% 
````

closed 2022-01-12 by grahamperrin