freebsd/drm-kmod #98: [drm ERROR :amdgpu_ttm_training_reserve_vram_init] alloc c2p_bo failed(-12)!

darkain, 2021-07-23:
**Describe the bug**
Error while initialization drm-kmod. This is on a Dell G5-SE 5505 laptop that has a Ryzen 4000 series APU paired with a Radeon RX5000 series GPU. The error MIGHT be related to the fact that there are two separate "display" devices. The APU acts as the primary driving the laptop display. The GPU is secondary, used for high-performance tasks and/or driving external displays.

[drm ERROR :amdgpu_ttm_training_reserve_vram_init] alloc c2p_bo failed(-12)!

**FreeBSD version**
FreeBSD g5-se 14.0-CURRENT FreeBSD 14.0-CURRENT #1 main-67a51854e8: Thu Jul 22 21:37:13 UTC 2021     root@g5-se:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**
http://bsd-hardware.info/?probe=9933a09c4f

<details>
  <summary>pciconf -lv</summary>
hostb0@pci0:0:0:0:  class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x1028 subdevice=0x09f5
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:   class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x1028 subdevice=0x09f5
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:  class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1633 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:1:2:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:  class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:2:1:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:2:2:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib7@pci0:0:2:4:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:  class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib8@pci0:0:8:1:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
pcib9@pci0:0:8:2:   class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
[... 117 lines cut ...]
hdac2@pci0:8:0:6:   class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x1028 subdevice=0x09f5
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Family 17h (Models 10h-1fh) HD Audio Controller'
    class      = multimedia
    subclass   = HDA
ahci0@pci0:9:0:0:   class=0x010601 rev=0x81 hdr=0x00 vendor=0x1022 device=0x7901 subvendor=0x1028 subdevice=0x09f5
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
ahci1@pci0:9:0:1:   class=0x010601 rev=0x81 hdr=0x00 vendor=0x1022 device=0x7901 subvendor=0x1028 subdevice=0x09f5
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA
</details>

**DRM KMOD version**
drm-devel-kmod 5.5.19.g20210709

**To Reproduce**
Build the driver from ports, and attempt to load it at boot via rc.conf

**Screenshots**
[image]

draumaz, 2022-03-11:
Still facing this issue as of 2022.

draumaz, 2022-03-13:
Hey there, I've got the same rig as you and am facing this exact problem. Every branch and build of drm-{,current-,devel-}kmod I've tried thus far always seems to give me this error. Has there been any updates regarding this since the initial posting?

[image]

draumaz, 2022-04-26:
Today, I was finally able to clone master and install drm-kmod without an immediate kernel panic. However, Xorg doesn't see any screens, and checking dmesg shows that, while the Renoir APU successfully initializes, NAVI10 is still broken. 

[image]

evadot, 2022-06-09:
xf86-video-scfb is always needed, that's probably your problem.

draumaz, 2022-06-09:
[image] drm-kmod-20220501, FreeBSD 14.0-CURRENT

evadot, 2022-06-10:
what do you mean "stuck in framebuffer mode" ? You can't interact with the console after ? The backtrack is "normal" (meaning it's a small bug that I have to fix but it's not a problem)

evadot, 2022-09-14:
No news, closing.
If this is still happening please re-open and answer the questions.