freebsd/drm-kmod #99: i915kms: symbol lkpi_kernel_fpu_begin undefined

vishwin, 2021-08-01:
After freebsd/freebsd-src@469884cf04a9b92677c7c83e229ca6b8814f8b0a and freebsd/freebsd-src@792b602a337ddc5efaa5e5326d9433fe3da7f303, builds successfully but fails to kldload:
```
link_elf_obj: symbol lkpi_kernel_fpu_begin undefined
linker_load_file: /boot/modules/i915kms.ko - unsupported file type
```
Not sure if this is something to fix here or a regression in base.

For the pedantic, I am currently running freebsd/freebsd-src@ff01d6343f1; port was built against the same.

shiorid, 2021-08-01:
While loading `i915kms` I receive this additional error:
```
Warning: memory type debugfsint leaked memory on destroy (2 allocations, 80 bytes leaked).
```

vishwin, 2021-08-02:
Should be fixed by freebsd/freebsd-src@9340ebd404efad7068b5a45fd3b3d8bb21187b91. Haven't had a chance to test yet, but as soon as one of us confirms then we can close.

closed 2021-08-02 by vishwin