freebsd/drm-kmod #100: Crash in 13.0-RELEASE when loading amdgpu.ko as part of kld_list when kern.vt.splash_cpu="1" in loader.conf

NorwegianRockCat, 2021-08-08:
**Describe the bug**
Brange 5.4-lts at change  

**FreeBSD version**
Paste the output of `uname -a`

FreeBSD concordia 13.0-RELEASE-p3 FreeBSD 13.0-RELEASE-p3 #12 releng/13.0-n244752-b368bb75bad8: Mon Jul 19 12:39:33 CEST 2021     root@concordia:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG-SG  amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
Paste the output of `pciconf -vl`

hostb0@pci0:0:0:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1480 subvendor=0x1043 subdevice=0x87c0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse Root Complex'
    class      = bridge
    subclass   = HOST-PCI
none0@pci0:0:0:2:       class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1481 subvendor=0x1043 subdevice=0x87c0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x87c0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x87c0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb3@pci0:0:3:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib9@pci0:0:3:2:       class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1483 subvendor=0x1043 subdevice=0x87c0
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb4@pci0:0:4:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb5@pci0:0:5:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
hostb6@pci0:0:7:0:      class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1482 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Starship/Matisse PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
[... 275 lines cut ...]
#15 0xffffffff80c2711d in softclock_call_cc (
    c=0xffffffff818e7398 <vt_consdev+272>, 
    cc=cc@entry=0xffffffff81ca9200 <cc_cpu>, direct=direct@entry=0)
    at /usr/src/sys/kern/kern_timeout.c:696
#16 0xffffffff80c275a9 in softclock (arg=0xffffffff81ca9200 <cc_cpu>)
    at /usr/src/sys/kern/kern_timeout.c:816
#17 0xffffffff80bcc5ed in intr_event_execute_handlers (p=<optimized out>, 
    ie=0xfffff80003bd5b00) at /usr/src/sys/kern/kern_intr.c:1168
#18 ithread_execute_handlers (p=<optimized out>, ie=0xfffff80003bd5b00)
    at /usr/src/sys/kern/kern_intr.c:1181
#19 ithread_loop (arg=arg@entry=0xfffff80003c1e840)
    at /usr/src/sys/kern/kern_intr.c:1269
#20 0xffffffff80bc93ee in fork_exit (
    callout=0xffffffff80bcc3a0 <ithread_loop>, arg=0xfffff80003c1e840, 
    frame=0xfffffe015cf82bc0) at /usr/src/sys/kern/kern_fork.c:1069
#21 <signal handler called>
(kgdb) 

```

Disabling kern.vt.splash_cpu will let the module load.

**Additional context**

I suspect that the dummy vt issue is causing an issue with vt_draw_2_vga16_px, maybe it gets called later?

evadot, 2022-01-27:
That should have been fixed by c14d65c11f15646b, I'll update the ports soon-ish

NorwegianRockCat, 2022-01-28:
@evadot Great! Thank you. I'll build from the repo and give it a shot this weekend and let you know if there are still problems.

NorwegianRockCat, 2022-01-29:
Tried it out. Yes, it is fixed. Thank you!

closed 2022-01-29 by NorwegianRockCat