freebsd/drm-kmod #101: amdgpu: Kernel panic on loading: malloc: called with spinlock or critical section held

neelchauhan, 2021-08-09:
**Describe the bug**

When I attempt to load `amdgpu` on an HP Envy x360 with AMD Ryzen 7 4700U, it goes to a blank screen and then get an `panic: malloc: called with spinlock or critical section held`.

This happens on the Ports 5.5 and the 5.6-wip branch where Renior should be supported (unlike my previous now-sold TigerLake HP Spectre).

**FreeBSD version**

    FreeBSD envy.local 14.0-CURRENT FreeBSD 14.0-CURRENT #0 main-n248478-f3a3b061216: Thu Aug  5 06:53:04 UTC 2021     root@releng1.nyi.freebsd.org:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

**PCI Info**

    hostb0@pci0:0:0:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Root Complex'
        class      = bridge
        subclass   = HOST-PCI
    none0@pci0:0:0:2:    class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir IOMMU'
        class      = base peripheral
        subclass   = IOMMU
    hostb1@pci0:0:1:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib1@pci0:0:1:3:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb2@pci0:0:2:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib2@pci0:0:2:1:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    pcib3@pci0:0:2:3:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb3@pci0:0:8:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib4@pci0:0:8:1:    class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Internal PCIe GPP Bridge to Bus'
        class      = bridge
        subclass   = PCI-PCI
    intsmb0@pci0:0:20:0:    class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    none1@pci0:0:20:3:    class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH LPC Bridge'
        class      = bridge
        subclass   = PCI-ISA
    hostb4@pci0:0:24:0:    class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1448 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Device 24: Function 0'
[... 75 lines cut ...]
        device     = 'Raven/Raven2/FireFlight/Renoir Audio Processor'
        class      = multimedia
    hdac1@pci0:4:0:6:    class=0x040300 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e3 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Family 17h (Models 10h-1fh) HD Audio Controller'
        class      = multimedia
        subclass   = HDA
    none5@pci0:4:0:7:    class=0x118000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e4 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Raven/Raven2/Renoir Sensor Fusion Hub'
        class      = dasp

**DRM KMOD version**

Happens on Ports `5.5.19.g20210709` and `5.6-wip`

**To Reproduce**
Take an HP Envy x360 with AMD Ryzen 4700U (or any DCN21 GPU), and load `amdgpu`. It should go to a blank screen and panic.

**Screenshots**
N/A - blank screen

**Additional context**

Stacktrace: [crash.log](https://github.com/freebsd/drm-kmod/files/6951746/crash.log)

unrelentingtech, 2021-08-09:
The critical section in bandwidth validation must be the FPU one… as of https://reviews.freebsd.org/D29921

This FPU stuff was a complete mess upstream and the malloc issue is only fixed in https://github.com/torvalds/linux/commit/680174cfd1e1cea70a8f30ccb44d8fbdf996018e / https://github.com/torvalds/linux/commit/f41ed88cbd6f025f7a683a11a74f901555fba11c

But the thing is, `dcn21_validate_bandwidth` in 5.5/5.6 really shouldn't malloc inside an FPU block (only dcn20 should), so maybe something left an FPU block still open…??

Try logging all that stuff (and also cleaning up leftover direct ifdef'd `kernel_fpu` calls), this is for v5.6-wip:

```diff
diff --git i/drivers/gpu/drm/amd/display/dc/dcn20/dcn20_resource.c w/drivers/gpu/drm/amd/display/dc/dcn20/dcn20_resource.c
index 443255335b..b441a20109 100644
--- i/drivers/gpu/drm/amd/display/dc/dcn20/dcn20_resource.c
+++ w/drivers/gpu/drm/amd/display/dc/dcn20/dcn20_resource.c
@@ -3047,9 +3047,6 @@ bool dcn20_validate_bandwidth(struct dc *dc, struct dc_state *context,
 	bool dummy_pstate_supported = false;
 	double p_state_latency_us;
 
-#ifdef __FreeBSD__
-	kernel_fpu_begin();
-#endif
 	DC_FP_START();
 	p_state_latency_us = context->bw_ctx.dml.soc.dram_clock_change_latency_us;
 	context->bw_ctx.dml.soc.disable_dram_clock_change_vactive_support =
@@ -3057,9 +3054,6 @@ bool dcn20_validate_bandwidth(struct dc *dc, struct dc_state *context,
 
 	if (fast_validate) {
 		voltage_supported = dcn20_validate_bandwidth_internal(dc, context, true);
-#ifdef __FreeBSD__
-		kernel_fpu_end();
-#endif
 		DC_FP_END();
 		return voltage_supported;
 	}
@@ -3614,9 +3608,6 @@ static bool dcn20_resource_construct(
 	enum dml_project dml_project_version =
 			get_dml_project_version(ctx->asic_id.hw_internal_rev);
 
-#ifdef __FreeBSD__
-	kernel_fpu_begin();
-#endif
 	DC_FP_START();
 
 	ctx->dc_bios->regs = &bios_regs;
@@ -3921,9 +3912,6 @@ static bool dcn20_resource_construct(
 
 create_fail:
 
-#ifdef __FreeBSD__
-	kernel_fpu_end();
-#endif
 	DC_FP_END();
 	dcn20_resource_destruct(pool);
 
diff --git i/drivers/gpu/drm/amd/display/dc/os_types.h w/drivers/gpu/drm/amd/display/dc/os_types.h
index c34eba1986..b781e17789 100644
--- i/drivers/gpu/drm/amd/display/dc/os_types.h
+++ w/drivers/gpu/drm/amd/display/dc/os_types.h
@@ -53,8 +53,10 @@
 #if defined(CONFIG_DRM_AMD_DC_DCN)
 #if defined(CONFIG_X86)
 #include <asm/fpu/api.h>
-#define DC_FP_START() kernel_fpu_begin()
-#define DC_FP_END() kernel_fpu_end()
+#define DC_FP_START() { DRM_WARN("FPU start @ %s:%d (%s)", __FILE__, __LINE__, __func__); \
+	kernel_fpu_begin(); }
+#define DC_FP_END() { DRM_WARN("FPU end @ %s:%d (%s)", __FILE__, __LINE__, __func__); \
+	kernel_fpu_end(); }
 #elif defined(CONFIG_PPC64)
 #include <asm/switch_to.h>
 #include <asm/cputable.h>
```

unrelentingtech, 2021-08-09:
Yeah, so as @wulf7 said in the chat, even correctly using nowait mallocs, the `KASSERT` trips anyway and here's your panic.

Having to run a non-`INVARIANTS` kernel on new GPUs just for this is not acceptable, so we have to pick an option:

- fix the damn assertion to be conditional on `M_WAITOK` (or non-`M_NOWAIT`)
  - I'm not an expert, I have no idea if that's safe, or why the assertion is non-conditional currently (i.e. why the code treats critical sections differently from interrupts)
- abandon `NOCTX`, use a stored context again — but now that we're thread-safe, the context would have to be stored in LinuxKPI `current` as well, so another version bump… oh well
  - doing this as a workaround for *an assertion* feels wrong!
- add local `ifdef` patches to move the alloc out of the FPU section
  - also sucks obviously but the least intrusive option (it's literally just two allocation calls…)

I guess I'll go with the last one but I'd like to hear some opinions about these options.

neelchauhan, 2021-08-09:
I think the last option is a good option also, we shouldn't complicate things too much...

wulf7, 2021-08-14:
May be we can add `DC_FP_END` right before malloc() and `DC_FP_START` right after it?

unrelentingtech, 2021-08-14:
@wulf7 yeah that's kinda what I meant by local patches. I'll go with that then.

wulf7, 2021-08-14:
following free() requires to be guarded too. Or converted to async version.

wulf7, 2021-08-17:
@neelchauhan Could you test both https://github.com/freebsd/drm-kmod/pull/103 and https://github.com/freebsd/drm-kmod/tree/5.6-wip for this INVARIANTS panic and for https://github.com/freebsd/drm-kmod/issues/105 panic?

neelchauhan, 2021-08-24:
Sorry for the long delay. The patch works well for me.

Thanks for your work!

wulf7, 2021-08-24:
> The patch works well for me.
Thanks!

Which one works?  #103 or https://github.com/freebsd/drm-kmod/tree/5.6-wip ?

neelchauhan, 2021-08-24:
5.6-wip works.

wulf7, 2021-08-24:
Awesome! Do you have opportunity to test 5.5-stable (#103) as well?

neelchauhan, 2021-09-04:
Sorry for the extreme delay (again), 5.5-stable works as well.

wulf7, 2021-09-09:
Thank you for testing and @unrelentingtech for fix

PR #103 (2021-09-08): [5.5-stable] Fix malloc-in-FPU-section
This should fix #101 on 5.5-stable hopefully?

Since https://reviews.freebsd.org/D29921 our FPU sections are critical sections (`NOCTX`) as they are on Linux, and our `INVARIANTS` caught the waiting-malloc-under-critical-section issue that was only found by the Linux developers *several* releases later. (dcn20 in 5.9, and then dcn21 in 5.12 — actually dcn21 has just been [left *without* an FPU block (!) until then](https://github.com/torvalds/linux/commit/41401ac67791810dd880345962339aa1bedd3c0d))

ARGH WHY IS LINUX SO SLOPPY WITH THE FPU STUFF
closed 2021-09-09 by wulf7