freebsd/drm-kmod #105: amdgpu: Panic when connecting or disconnecting AC Adapter

neelchauhan, 2021-08-10:
**Describe the bug**

When I use an amdgpu-based laptop and connect or disconnect the power, it panics. This happens on a HP Envy x360.

**FreeBSD version**
    FreeBSD envy.local 14.0-CURRENT FreeBSD 14.0-CURRENT #0: Mon Aug  9 15:53:45 PDT 2021     root@envy.local:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64

**PCI Info**

    hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Root Complex'
        class      = bridge
        subclass   = HOST-PCI
    none0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir IOMMU'
        class      = base peripheral
        subclass   = IOMMU
    hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib1@pci0:0:1:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib2@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    pcib3@pci0:0:2:3:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe GPP Bridge'
        class      = bridge
        subclass   = PCI-PCI
    hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir PCIe Dummy Host Bridge'
        class      = bridge
        subclass   = HOST-PCI
    pcib4@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Internal PCIe GPP Bridge to Bus'
        class      = bridge
        subclass   = PCI-PCI
    intsmb0@pci0:0:20:0:	class=0x0c0500 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790b subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH SMBus Controller'
        class      = serial bus
        subclass   = SMBus
    none1@pci0:0:20:3:	class=0x060100 rev=0x51 hdr=0x00 vendor=0x1022 device=0x790e subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'FCH LPC Bridge'
        class      = bridge
        subclass   = PCI-ISA
    hostb4@pci0:0:24:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1448 subvendor=0x0000 subdevice=0x0000
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Renoir Device 24: Function 0'
        class      = bridge
        subclass   = HOST-PCI
    hostb5@pci0:0:24:1:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1449 subvendor=0x0000 subdevice=0x0000
[... 77 lines cut ...]
        class      = multimedia
        subclass   = HDA
    none5@pci0:4:0:7:	class=0x118000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x15e4 subvendor=0x103c subdevice=0x876e
        vendor     = 'Advanced Micro Devices, Inc. [AMD]'
        device     = 'Raven/Raven2/Renoir Sensor Fusion Hub'
        class      = dasp

**DRM KMOD version**

    drm-devel-kmod 5.5.19.g20210709

**To Reproduce**
Steps to reproduce the behavior:

Take an amdgpu-based laptop, and plug in or unplug the laptop while it's powered on.

**Screenshots**

N/A

**Additional context**

Another FreeBSD user using a Ryzen-based HP ProBook also had this issue: https://lists.freebsd.org/archives/freebsd-current/2021-August/000412.html

Stack trace is [here](https://github.com/freebsd/drm-kmod/files/6958302/ac-crash.txt).

unrelentingtech, 2021-08-14:
> amdgpu_pm_acpi_event_handler

Duplicate of #74! committers @wulf7 @evadot please cherry-pick 3ed2827 already

wulf7, 2021-08-14:
Cherry-picked it in to 5.5-stable branch.

wulf7, 2021-08-14:
5.6 will get it in 5.6.12.

wulf7, 2021-08-14:
@neelchauhan could you test 5.5-stable from github?

neelchauhan, 2021-08-23:
Sorry for the delay, I was busy.

I can confirm that it works! Thanks!

closed 2021-08-23 by neelchauhan