freebsd/drm-kmod #106: AMDGPU memory leak over time

hselasky, 2021-08-15:
**Describe the bug**
AMDGPU memory leak over time

**FreeBSD version**
FreeBSD hpstv2020.selasky.org 13.0-STABLE FreeBSD 13.0-STABLE #11 stable/13-n246667-64a0116cd00-dirty: Thu Aug  5 02:11:14 CEST 2021     hans_other@hpstv2020.selasky.org:/usr/obj/usr/img/src/amd64.amd64/sys/GENERIC  amd64

**DRM KMOD version**
drm-fbsd13-kmod 5.4.92.g20210720_1

**To Reproduce**
N/A

**Screenshots**

**Additional context**

vmstat -m | grep linux

reports over 2GByte over time. I suspect it is related to one of the issues mentioned here:

https://lists.freedesktop.org/archives/amd-gfx/2021-May/063137.html

Does anyone have any time to see if any of those patches apply for 5.4 ?

--HPS

wulf7, 2021-08-15:
Some of these patches were included into 5.4 LTS branch. I'll try to advance it to 5.4.140 next week

wulf7, 2021-08-16:
I place 5.4.92 -> 5.4.141 update here for now: https://github.com/wulf7/drm-kmod/tree/5.4-lts

I don't know if it fixes memory leak or not, but it worth a try.

hselasky, 2021-08-17:
I'll try to give it a spin. Thank you!

hselasky, 2021-08-21:
The amdgpu driver works with your 5.4-lts. Now checking if it leaks memory.

hselasky, 2021-09-17:
Still no memory leak. Can you please merge these changes?

wulf7, 2021-09-17:
> Still no memory leak. Can you please merge these changes?

Already did it few days ago. Just update your ports/packages.

closed 2021-09-17 by hselasky