freebsd/drm-kmod #108: The system may fail to start properly – a blackout – if kld_list includes radeonkms (workaround: instead, load drm)

grahamperrin, 2021-09-02:
**Description**

AMD _Thames [Radeon HD 7550M/7570M/7650M]_ in a circa 2013 HP EliteBook 8570p.

Last year I learnt, for myself: 

`sysrc -f /etc/rc.conf kld_list-=radeonkms && sysrc -f /etc/rc.conf kld_list+=drm`

Without this substitution – `drm` in lieu of `radeonkms` – there is (for me) a high likelihood of SDDM not appearing. More specifically: 

* an endless blackout, maybe with a movable pointer
* the OS might become generally non-responsive i.e. no response to <kbd>Control</kbd>-<kbd>Alt</kbd>-<kbd>F1</kbd> or <kbd>Control</kbd>-<kbd>Alt</kbd>-<kbd>F2</kbd>.

**FreeBSD version**

`FreeBSD mowa219-gjp4-8570p-freebsd 14.0-CURRENT FreeBSD 14.0-CURRENT #105 main-n248685-c9f833abf1d: Fri Aug 13 20:24:43 BST 2021     root@mowa219-gjp4-zbook-freebsd:/usr/obj/usr/src/amd64.amd64/sys/GENERIC-NODEBUG  amd64`

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
```text
hostb0@pci0:0:0:0:      class=0x060000 rev=0x09 hdr=0x00 vendor=0x8086 device=0x0154 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '3rd Gen Core processor DRAM Controller'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:0:       class=0x060400 rev=0x09 hdr=0x01 vendor=0x8086 device=0x0151 subvendor=0x8086 subdevice=0x2010
    vendor     = 'Intel Corporation'
    device     = 'Xeon E3-1200 v2/3rd Gen Core processor PCI Express Root Port'
    class      = bridge
    subclass   = PCI-PCI
xhci0@pci0:0:20:0:      class=0x0c0330 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e31 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family USB xHCI Host Controller'
    class      = serial bus
    subclass   = USB
none0@pci0:0:22:0:      class=0x078000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e3a subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family MEI Controller'
    class      = simple comms
uart2@pci0:0:22:3:      class=0x070002 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e3d subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family KT Controller'
    class      = simple comms
    subclass   = UART
em0@pci0:0:25:0:        class=0x020000 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1502 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '82579LM Gigabit Network Connection (Lewisville)'
    class      = network
    subclass   = ethernet
ehci0@pci0:0:26:0:      class=0x0c0320 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e2d subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family USB Enhanced Host Controller'
    class      = serial bus
    subclass   = USB
hdac1@pci0:0:27:0:      class=0x040300 rev=0x04 hdr=0x00 vendor=0x8086 device=0x1e20 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family High Definition Audio Controller'
    class      = multimedia
    subclass   = HDA
pcib2@pci0:0:28:0:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e10 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C216 Chipset Family PCI Express Root Port 1'
    class      = bridge
    subclass   = PCI-PCI
pcib3@pci0:0:28:2:      class=0x060400 rev=0xc4 hdr=0x01 vendor=0x8086 device=0x1e14 subvendor=0x103c subdevice=0x17a7
    vendor     = 'Intel Corporation'
    device     = '7 Series/C210 Series Chipset Family PCI Express Root Port 3'
[... 617 lines cut ...]

The tail: 

```text
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: drmn0: GPU reset succeeded, trying to resume
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] enabling PCIE gen 2 link speeds, disable with radeon.pcie_gen2=0
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] PCIE GART of 1024M enabled (table at 0x0000000000162000).
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: drmn0: WB enabled
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: drmn0: fence driver on ring 0 use gpu addr 0x0000000040000c00 and cpu addr 0x0xfffff800219ebc00
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: drmn0: fence driver on ring 3 use gpu addr 0x0000000040000c0c and cpu addr 0x0xfffff800219ebc0c
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: drmn0: fence driver on ring 5 use gpu addr 0x0000000000072118 and cpu addr 0x0xfffff800c0072118
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] ring test on 0 succeeded in 1 usecs
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] ring test on 3 succeeded in 4 usecs
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] ring test on 5 succeeded in 2 usecs
Aug 29 04:56:10 mowa219-gjp4-8570p-freebsd kernel: [drm] UVD initialized successfully.
Aug 29 04:56:11 mowa219-gjp4-8570p-freebsd kernel: [drm ERROR :r600_ib_test] radeon: fence wait timed out.
Aug 29 04:56:11 mowa219-gjp4-8570p-freebsd kernel: [drm ERROR :radeon_ib_ring_tests] radeon: failed testing IB on GFX ring (-60).
Aug 29 04:56:35 mowa219-gjp4-8570p-freebsd login[2241]: ROOT LOGIN (root) ON ttyv1
```

### Retrospective

Notes, from before I learnt the `drm` workaround, suggest that the issue was observable as far back as September 2018 (FreeBSD 12.0-ALPHA7 at the time). I never bothered to report the issue because I assumed a peculiarity with my setup. 

Now I'm better prepared to make the issue reproducible, if I can …

grahamperrin, 2021-09-02:
### Scope

Reportedly (I can not confirm this), another case where a workaround involved substitution – `drm` in lieu of a specific module: 

<https://lists.freebsd.org/pipermail/freebsd-questions/2021-May/294001.html>

– in that case, it was `i915kms`. 

----

For now: with a focus on reproducibility, I'll stick to `radeonkms` (and not edit the title of the issue).

zeising, 2021-09-02:
It should be enough to load the more specific graphics driver, i915kms, radeonkms or amdgpu, you don't need to manually load drm.ko, as that will be pulled in automatically.

If there are issues, ensure that the correct drm module has been loaded, it should be from /boot/modules, not /boot/kernel. Older FreeBSD systems still have drm drivers in /boot/kernel as well, but they might conflict.

grahamperrin, 2021-09-02:
Thank you. 

> … Older FreeBSD systems still have drm drivers in /boot/kernel as well, but they might conflict. …

At a glance, I see nothing in conflict at this path. (Recently-built 14.0-CURRENT here.) 

> … you don't need to manually load drm.ko, as that will be pulled in automatically. …

Understood, thanks. 

My sense of things is that timely auto-loading of `drm` (i.e. `kld_list`) with _later_ auto-loading of `radeonkms` is less likely to trigger symptoms of this issue.

PaddyMac, 2021-12-11:
I'm using drm-fbsd13-kmod-5.4.144.g20211013 on 13.0-RELEASE-p4, and this issue is still present for me. I might get a successful boot maybe once out of a dozen system reboots if I'm lucky. But I just realized it seems to be working okay if I just leave my kld_list entry empty (since radeonkms was the only thing I had in there) and let radeonkms and drm be pulled in automatically because I also have SDDM starting up at boot. This is probably not an option on 12, but it seems to work on 13.

grahamperrin, 2022-01-13:
https://github.com/freebsd/drm-kmod/issues/108#issue-985940817

> … I reckon that there's somehow an issue with the order in which things are loaded/started.
> 
> I'll aim to make the issue reproducible reasonably soon, hopefully some time this month. …

I did have, at the back of my mind, a fairly clear test plan, however I chose not to share it in GitHub. Didn't want to make noise here. As things turned out: test situations (according to the plan) rarely arose. In a nutshell: not a type of situation that could be created on demand. 

----

Some months have passed without me encountering symptoms of this issue. 

Closing thoughts: 

* <https://github.com/freebsd/drm-kmod/issues/108#issuecomment-943092377> let's assume that the mid-October 2021 commit was a positive fix
* if that was _not_ the fix, then my best guess would be that in recent months, I have somehow masked the issue by making use of removable persistent L2ARC (two old USB flash drives) – with startup so much faster, less likelihood of triggering an issue that involved _timing/order_.

closed 2022-01-13 by grahamperrin