freebsd/drm-kmod #111: Missing support for AMD Lucienne APU?

bluedudexoo, 2021-09-12:
Hi all! I've recently grabbed a laptop with a Ryzen 7 5700u, with the integrated Lucienne Radeon Vega 8 GPU chip. From my research, it looks like it's a slightly tweaked Renoir Radeon Vega 8 chip. FreeBSD 14.0-CURRENT with drm-devel-kmod compiled and installed yesterday can see it, knows what it's called, but I don't believe the amdgpu driver supports it yet. I've tried: 
-adding it with the kldlist rc.conf option
-using the xf86-video-amdgpu driver
-hand-configuring xorg.conf (using info from pciconf)

I'm wracking my brain the see if there's anything that I've missed. All my users are added to the video group, I've tried letting X configure itself, and I still end up with some variant of "no screens found."

I can attach logs later on tonight, as I'm currently helping remodel my kitchen. Thanks for your time!

neelchauhan, 2021-09-13:
I don't think the 5700u is supported in `drm-kmod` yet. `drm-kmod` development is very slow.

bluedudexoo, 2021-09-13:
That's what I thought- is there any info or anything I could provide to help speed things up? I'm totally inexperienced at driver programming, but I can provide hardware info and a testing bed.

neelchauhan, 2021-09-13:
I once attempted to update drm-kmod to 5.7, but it wasn't stable. I copied over the code/effort I have done to 5.6-wip.

I know Renoir got support in 5.5. I'm using it right now (after selling an Intel TigerLake PC for it).

drm-kmod goes step-by-step in Linux major versions. It's slow I'd prefer to jump versions, say each LTS, or maybe just use RHEL `drm` and jump with each new RHEL, but it's the way it's done.

If you want to speed it up, let's get 5.6 stable, but you may need an older PC using `amdgpu`.

For development, there is a `scripts/` folder.

bluedudexoo, 2021-09-24:
Sorry for the late reply- got incredibly busy with everything. I'll definitely check that out, since I've been meaning to learn a bit of C and Assembly. Unfortunately, I don't currently have an older amdgpu laptop, but my current one will definitely be available for any testing/feedback anyone needs for now. I try to update it from source every few days, though it takes a little while (about an hour for FreeBSD, 2~ hours for ports).

Once I get a bit more stable in terms of time, I'll be able to start checking things out. Is there an IRC channel or something I can join to keep updated?

bluedudexoo, 2021-10-10:
I'll give that a shot!

arvedarved, 2021-12-30:
I have also a ```0x164c``` device and adding the PCI ID to the ```5.7-wip``` branch made things working.

headwhacker, 2022-04-07:
Has anyone managed to successfully start X on this GPU or we are still waiting for 5.11 or later to get the official support for this GPU?

headwhacker, 2022-04-11:
Actually FreeBSD 13.1 RC2 plus drm-devel-kmod which is based on 5.7 is working now my Thinkpad E14 Gen3 (Ryzen 5 5500U)

opsec, 2022-05-18:
> Actually FreeBSD 13.1 RC2 plus drm-devel-kmod which is based on 5.7 is working now my Thinkpad E14 Gen3 (Ryzen 5 5500U)

Can you describe the steps necessary to build drm-devel-kmod ? It's no longer in the tree ?

unrelentingtech, 2022-05-18:
There was a switch to numbered version ports: `graphics/drm-510-kmod`

opsec, 2022-06-28:
- how to build:
  mkdir ~/git
  cd ~/git
  git clone https://github.com/freebsd/drm-kmod
  cd drm-kmod
  git checkout 5.7-stable
- this step needs /usr/src populated, so populate it:
  cd /usr
  git clone -o freebsd --config \
        remote.freebsd.fetch='+refs/notes/*:refs/notes/*' \
        https://git.freebsd.org/src.git src
  cd src
  git checkout releng/13.1
  cd ~/git/drm-kmod
  make
  make install
- then: reboot

evadot, 2024-01-05:
Closing as we got support for this a long time ago.

closed 2024-01-05 by evadot