freebsd/drm-kmod #113: Seemingly random kernel panics with RENOIR

0xSteeW, 2021-09-17:
**Describe the bug**
A clear and concise description of what the bug is.
I am getting random kernel panics with amd renoir integrated graphics.

As a side note, since I reinstalled freebsd, I am not able to open firefox due to a SIGSEGV, and I'm considering this might be the cause.

**FreeBSD version**
Paste the output of `uname -a`

FreeBSD zerserk 13.0-STABLE FreeBSD 13.0-STABLE #1: Wed Sep 15 22:09:18 CEST 2021     root@zerserk:/usr/obj/usr/src/amd64.amd64/sys/STEEW  amd64

**PCI Info**

<details>
  <summary>pciconf -lv</summary>
  
Paste the output of `pciconf -vl`
hostb0@pci0:0:0:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1630 subvendor=0x103c subdevice=0x87b1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne Root Complex'
    class      = bridge
    subclass   = HOST-PCI
amdiommu0@pci0:0:0:2:	class=0x080600 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1631 subvendor=0x103c subdevice=0x87b1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne IOMMU'
    class      = base peripheral
    subclass   = IOMMU
hostb1@pci0:0:1:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib1@pci0:0:1:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1633 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib2@pci0:0:1:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb2@pci0:0:2:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib3@pci0:0:2:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
pcib4@pci0:0:2:4:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1634 subvendor=0x1022 subdevice=0x1453
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir/Cezanne PCIe GPP Bridge'
    class      = bridge
    subclass   = PCI-PCI
hostb3@pci0:0:8:0:	class=0x060000 rev=0x00 hdr=0x00 vendor=0x1022 device=0x1632 subvendor=0x0000 subdevice=0x0000
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir PCIe Dummy Host Bridge'
    class      = bridge
    subclass   = HOST-PCI
pcib5@pci0:0:8:1:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
    class      = bridge
    subclass   = PCI-PCI
pcib6@pci0:0:8:2:	class=0x060400 rev=0x00 hdr=0x01 vendor=0x1022 device=0x1635 subvendor=0x1022 subdevice=0x1635
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'Renoir Internal PCIe GPP Bridge to Bus'
[... 105 lines cut ...]
    class      = mass storage
    subclass   = SATA
ahci1@pci0:6:0:1:	class=0x010601 rev=0x81 hdr=0x00 vendor=0x1022 device=0x7901 subvendor=0x103c subdevice=0x87b1
    vendor     = 'Advanced Micro Devices, Inc. [AMD]'
    device     = 'FCH SATA Controller [AHCI mode]'
    class      = mass storage
    subclass   = SATA

</details>

**DRM KMOD version**
Paste the output of `pkg query -x '%n %v' 'drm.*kmod'`

**To Reproduce**
Steps to reproduce the behavior:

Random usage, though seems to happen sometimes when opening mozilla firefox.
**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Additional context**
Add any other context about the problem here.
Display server: wayland
Compositor: wayfire

evadot, 2021-10-08:
Do you have a reliable way to reproduce ?

evadot, 2022-03-30:
Closing as OP said it's not drm related.

closed 2022-03-30 by evadot